FooDeas/raspberrypi-ua-netinst #2: Testing on Zero

Phiber2000, 2016-07-27:
Does someone have a RaspberriPi Zero to test the netinstaller on it?

Phiber2000, 2016-07-27:
Please post if it boots the kernel (no rainbow screen) and the network chipset you tried.

thijstriemstra, 2016-08-16:
Where you located, I can sponsor one for you.

FooDeas, 2016-08-16:
Phiber2000 was a university classmate and transferred me the ownership of this repository because I always used his GitHub account. :smile:
I'm from Germany and wouldn't say no.

thijstriemstra, 2016-08-17:
cool, I sent you an email at your github commit email address.

FooDeas, 2016-08-18:
No email received yet - typo?

thijstriemstra, 2016-08-18:
It's your username ending with aeberlein.de correct?

FooDeas, 2016-08-18:
Absolutely: foodeas@aeberlein.de

thijstriemstra, 2016-08-18:
I've resent it, weird..

FooDeas, 2016-08-18:
I see why:
`reject: RCPT from unknown[166.78.137.222]: 450 4.7.1 Client host rejected: cannot find your reverse hostname, [166.78.137.222]; from=<info@collab.nl> to=<foodeas@...> proto=ESMTP helo=<mail.collab.nl>`

Weird too: The mailserver IP is located in "San Antonio, Texas, United States" :D
Could you use another mail account? The server you sent from is not configured correctly.

thijstriemstra, 2016-08-18:
damn, i have to fix that. But yea the servers in the us. I'll use gmail sigh.

FooDeas, 2016-08-22:
Did you receive my answer?

FooDeas, 2016-09-13:
Thanks to @thijstriemstra for the donation!
Running the Zero with an USB ethernet adapter, I found a bug belonging to the kernel modules and here is the result:
```
==================================================
raspberrypi-ua-netinst
==================================================
Revision 93f302b
Built on Tue Sep 13 20:36:21 CEST 2016
Running on Raspberry Pi version Zero
==================================================
```
:+1::+1::+1::+1::+1::+1:

There are two possibilities to get the Zero working:

**Variant 1:**
- You need a USB ethernet adapter, which is supported by the current kernel.
- Connect the USB ethernet adapter, you want to use with the installer to a running Debian linux and note the appropriate kernel module via `lsmod`. 
- Add `drivers_to_load=<kernel module>` to the `installer-config.txt`.

**Variant 2:**
- Use any other Raspberry Pi with µSD slot for netinstall.
- Just insert the µSD card after the installation.

closed 2016-09-13 by FooDeas