FooDeas/raspberrypi-ua-netinst #3: Additional (new) config.txt variables for camera

Phiber2000, 2016-08-03:
Should `gpu_mem` be available in `installer-config.txt`?
Other variables: `disable_camera_led` and `start_x`

@thijstriemstra
The variable `start_x` is not listed in https://www.raspberrypi.org/documentation/configuration/config-txt.md. Could you check if it's still needed?

thijstriemstra, 2016-08-03:
Yea good question. Turned out my first camera (or cable) was broken but by the time I figured that out I already configured start_x and left it as is and continued with my second camera that worked. I'll test without `start_x` and get back to you (as soon as current installer finished ;)).

thijstriemstra, 2016-08-03:
@Phiber2000 just tested and camera didn't work without `start_x`.

Phiber2000, 2016-08-03:
Thanks a lot for testing! It will be comfortably implemented soon!

Phiber2000, 2016-08-03:
Done with commit 85aeebe055f8dfba975ca05a35a65ea9ba102884.

New release follows in a few minutes. It would be nice, if you could test the camera option!

closed 2016-08-03 by Phiber2000

thijstriemstra, 2016-08-16:
Works great, thanks