FooDeas/raspberrypi-ua-netinst #5: Implement booting from USB storage

FooDeas, 2016-08-05:
The installer should implement the new feature to install and boot from USB storage as described here: [PI 3 BOOTING PART I: USB MASS STORAGE BOOT BETA](https://www.raspberrypi.org/blog/pi-3-booting-part-i-usb-mass-storage-boot/)

FooDeas, 2016-08-08:
Important info from [Raspberry Pi Forum](https://www.raspberrypi.org/forums/viewtopic.php?f=63&t=155167&p=1021091#p1021091):

> ..."program_usb_timeout" is non reversible written to OTP...

> ...
> Yes it's permanent otherwise it couldn't effect the bootrom, it readable in the otp_dump it's bit 24 up near the top of the range...
> ...
> new empty file called TIMEOUT on the sdcard (this triggers the same effect of setting the above extra delay without setting the OTP bit)

FooDeas, 2016-08-12:
Functionality is given with 6d719cee1528b94003271f1cd8a580e29b47356d. This should be working as soon as the firmware package is updated by package maintainers.

Documentation is not completed yet.

thijstriemstra, 2016-09-29:
Here's a tutorial that seems useful: https://www.raspberrypi.org/blog/pi-3-booting-part-i-usb-mass-storage-boot/

FooDeas, 2016-09-29:
Not needed. It already works!
But the firmware is still in `next` branch and not deployed via package.

thijstriemstra, 2016-09-29:
Awesome! Hopefully it'll get merged soon.

thijstriemstra, 2016-11-29:
Is there any documentation on how to do this?

FooDeas, 2016-11-29:
You have to replace the `bootcode.bin` before and after the installation and set `usbboot=1` in `installer-config.txt`.
Have an extra look at the installer output belonging to the msd boot bit you might have to program once!

thijstriemstra, 2016-12-05:
Ordered a [PiDrive](https://www.wdc.com/nl-nl/products/wdlabs/wd-pidrive-foundation-edition.html) and will test this. Also, support for raspberrypi-ua-netinst in https://github.com/maxnet/berryboot would be nice. It uses a Qt front-end, maybe I can hack it a little. [This page](http://www.legitreviews.com/wd-labs-pidrive-314gb-hard-drive-review_180301/5) also mentions a 4GB swap can be created and used as RAM which sounds awesome.

FooDeas, 2016-12-05:
You know, that you can use a simple USB flash drive, too?

thijstriemstra, 2016-12-05:
True, but they're much slower and more expensive (this disk was 30 euros for 250GB). Benchmarks here: http://www.legitreviews.com/wd-labs-pidrive-314gb-hard-drive-review_180301/4

FooDeas, 2016-12-05:
👍

FooDeas, 2016-12-29:
It's still in beta. I'm testing if it is moved to stable from time to time.
I'm pretty sure, it will be posted on the [RPi Blog](https://www.raspberrypi.org/blog/) when the time comes.

thijstriemstra, 2016-12-29:
Alright, too bad.

thijstriemstra, 2017-02-16:
So according to this [post](http://www.makeuseof.com/tag/make-raspberry-pi-3-boot-usb/) the following steps are required:

* replace the default `start.elf` and `bootcode.bin` files with the alternatives from the next branch (using apt-get update but I'd rather wget it?)
* enable usb boot mode and reboot:
```
echo program_usb_boot_mode=1 | sudo tee -a /boot/config.txt
```
* verify that OTP is representative of the address `0x3020000a` (such as `17:3020000a`):
```
vcgencmd otp_dump | grep 17:
```

Which I'll test now.

thijstriemstra, 2017-02-16:
Rebooted and checked the output of OTP but this didn't return the expected result (`17:3020000a`):

```
$ vcgencmd otp_dump | grep 17:
17:1020000a
```

So I did the recommended branch update instead:

```
$ sudo BRANCH=next rpi-update
 *** Updating firmware
 *** Updating kernel modules
 *** depmod 4.9.9-v7+
 *** depmod 4.9.9+
 *** Updating VideoCore libraries
 *** Using HardFP libraries
 *** Updating SDK
 *** Running ldconfig
 *** Storing current firmware revision
 *** Deleting downloaded files
 *** Syncing changes to disk
 *** If no errors appeared, your firmware was successfully updated to 719eb18d040b53903f7083cef2ee9f92300a90ab
 *** A reboot is needed to activate the new firmware

```
And rebooted but still same result.

FooDeas, 2017-02-16:
The first byte (`0x10`) has to contain the 1 on 3rd position (`0b00100000`). So you're right with `0x30...`.

The `config.txt` has to include `program_usb_boot_mode=1` (this is needed only once). At this time, the `bootcode.bin` has to be already replaced from [here](https://github.com/raspberrypi/firmware/tree/next/boot). Do that by yourself.
During the next boot, the OTP will get programmed.

Now `vcgencmd otp_dump | grep 17:` should return the `0x30...`.

thijstriemstra, 2017-02-16:
completed the tutorial and my machine isn't booting from USB; most likely because of this otp_dump issue. Maybe I should hold off until it's officially supported..

FooDeas, 2017-02-16:
It definitely works! Have you get the OTP setup working with my comment from above now?
(I'm unsure, what you've meant with "tutorial"...)

thijstriemstra, 2017-02-16:
I meant [this tutorial](https://github.com/raspberrypi/documentation/blob/master/hardware/raspberrypi/bootmodes/msd.md), I haven't used raspberrypi-ua-netinst for this yet; first I'd like to get it working without but this OTP returns the wrong nr. Will try with a raspberrypi 3 vs rpi2

FooDeas, 2017-02-16:
Forget the RPi 2 - it only works with RPi 3:

> Will it be possible to boot a Pi 1 or Pi 2 using MSD?
> 
> Unfortunately not. The boot code is stored in the BCM2837 device only, so the Pi 1, Pi 2, and Pi Zero will all require SD cards.

thijstriemstra, 2017-02-16:
ah thanks, hopeful.

FooDeas, 2017-02-16:
What I wrote [here](https://github.com/FooDeas/raspberrypi-ua-netinst/issues/5#issuecomment-280444019) has to work definitely! I'm confident that you'll get it working, too

FooDeas, 2017-02-16:
What I wrote [here](https://github.com/FooDeas/raspberrypi-ua-netinst/issues/5#issuecomment-280444019) has to work definitely! I'm confident that you'll get it working, too.

(Damn "Close and comment button"...)

thijstriemstra, 2017-02-16:
Alright this is better:

```
$ vcgencmd otp_dump | grep 17:
17:3020000a
```

thijstriemstra, 2017-02-16:
So raspberrypi-ua-netinst basically needs to:

- reject this feature on RPI <= 2
- program USB boot mode once (wget the `start.elf` and `bootcode.bin` files and put them in place)
- verify OTP has been programmed:
```
$ vcgencmd otp_dump | grep 17:
17:3020000a
```
The final step, [preparing the USB device](https://github.com/raspberrypi/documentation/blob/master/hardware/raspberrypi/bootmodes/msd.md#prepare-the-usb-storage-device) is more of a documentation thing I imagine.

Thoughts?

FooDeas, 2017-02-16:
- [ ] reject this feature on RPI <= 2 **- Great idea!**
- [x] program USB boot mode once **- It's semi automatic at the moment because OTP changes are permanent and I was very careful while developing this. Have a look at the default config.txt of the latest release...**
- [ ] wget the start.elf and bootcode.bin files and put them in place **- The problem is to _keep_ them during a kernel (wheezy)/bootloader (jessy and above) update because they're not delivered by the raspberrypi.org packages yet. So they may be overwritten and your Pi stops booting until you replace them again...**
- [x] verify OTP has been programmed
Everything else is already working... But I paused this topic because the main problem above.

thijstriemstra, 2017-02-16:
> The problem is to keep them [start.elf and bootcode.bin] during a kernel (wheezy)/bootloader (jessy and above) update because they're not delivered by the raspberrypi.org packages yet. So they may be overwritten and your Pi stops booting until you replace them again...

I suppose this is something I can live with; I simply cannot update (the kernel using apt-get)?

thijstriemstra, 2017-02-16:
And what about these [instructions](https://github.com/raspberrypi/documentation/blob/master/hardware/raspberrypi/bootmodes/msd.md#prepare-the-usb-storage-device): raspberrypi-ua-netinst does most of that right?

Create the boot and root file systems:
```
sudo mkfs.vfat -n BOOT -F 32 /dev/sda1
sudo mkfs.ext4 /dev/sda2
```

Mount the target file system and copy the running raspbian system to it:
```
sudo mkdir /mnt/target
sudo mount /dev/sda2 /mnt/target/
sudo mkdir /mnt/target/boot
sudo mount /dev/sda1 /mnt/target/boot/
sudo apt-get update; sudo apt-get install rsync
sudo rsync -ax --progress / /boot /mnt/target
```

FooDeas, 2017-02-16:
Post 1: You can update, but you can't be sure, that a reboot will work if the files got replaced.
Post 2: The system will be installed directly on the target USB device. Nothing special to copy/move/config.

thijstriemstra, 2017-05-20:
Also see https://learn.adafruit.com/external-drive-as-raspberry-pi-root/hooking-up-the-drive-and-copying-slash

FooDeas, 2017-06-16:
Starting with the 2017-04-10 release of Raspbian there ist some progress here!
I need some testing and will possibly release that in very few days.

FooDeas, 2017-06-18:
OK - it's done! Checkout: d00f3c9900a989e0aaf1d6b692fc68bd6f2642a5

thijstriemstra, 2017-06-21:
Awesome, will do!

FooDeas, 2017-07-08:
Results?

thijstriemstra, 2017-09-11:
Testing now with:

```
usbroot=1
usbboot=1
```

thijstriemstra, 2017-09-11:
And it looks like it worked out of the box using a PiDrive. Nice work!

thijstriemstra, 2017-09-11:
```
==================================================
raspberrypi-ua-netinst
==================================================
Revision d01fd04
Built on ma 11 sep 2017 23:47:54 CEST
Running on Raspberry Pi version 3 Model B
==================================================
https://github.com/FooDeas/raspberrypi-ua-netinst/
==================================================
Starting HWRNG... OK
Mounting boot partition... OK
Copying boot files... OK
Executing installer-config.txt...
OK
```
This ``OK`` shouldn't be on newline.

```
Searching for deprecated variables...

Checking if config.txt needs to be modified before starting installation...
  Setting HDMI options... OK
  Checking USB boot flag... OK
OK
```

This ``OK`` doesn't have a description.

```

Unmounting boot partition... OK

Network configuration:
  ifname = eth0
  ip_addr = dhcp
  online_config = 

Waiting for eth0... OK
Configuring eth0 with DHCP... 192.168.2.54
Set time using ntpdate... OK

Installation started at Mon Sep 11 22:39:11 UTC 2017 (UTC).

=============================================================================================
                                  !!! IMPORTANT NOTICE !!!
Because you are installing from SD card and want to boot from USB,
the system will POWERED OFF after installation.
After finishing the installation, you must REMOVE the SD card and reboot the system MANUALLY.

The installation will continue in 15 seconds...
=============================================================================================
```

This notice should be in the documentation as well (if it's not already).

closed 2017-09-30 by FooDeas