FooDeas/raspberrypi-ua-netinst #8: GPIO rules

thijstriemstra, 2016-08-09:
In order to use and run RPi.GPIO python programs without sudo a udev rule needs to be installed. I'm now using the `installation-files.list` feature to install this udev rule, e.g.:

```
foo:foo  775  /etc/udev/rules.d/99-gpio.rules
```
 but would be nice to have a setting for this (requires user to be added to new `gpio` group).

Rule contents:

```
SUBSYSTEM=="bcm2835-gpiomem", KERNEL=="gpiomem", GROUP="gpio", MODE="0660"
SUBSYSTEM=="gpio", KERNEL=="gpiochip*", ACTION=="add", PROGRAM="/bin/sh -c 'chown root:gpio /sys/class/gpio/export /sys/class/gpio/unexport ; chmod 220 /sys/class/gpio/export /sys/class/gpio/unexport'"
SUBSYSTEM=="gpio", KERNEL=="gpio*", ACTION=="add", PROGRAM="/bin/sh -c 'chown root:gpio /sys%p/active_low /sys%p/direction /sys%p/edge /sys%p/value ; chmod 660 /sys%p/active_low /sys%p/direction /sys%p/edge /sys%p/value'"
```

thijstriemstra, 2016-08-09:
I have another rule `92-local-vchiq-permissions.rules` with contents:

```
SUBSYSTEM=="vchiq",GROUP="video",MODE="0660"
```
But maybe this is Qt-ish related and doesn't necessary need a new setting.

FooDeas, 2016-08-12:
Why belongs the vchiq rule to gpio? As I know, it's only needed if a user application needs direct access to the GPU. Please give a short explanation.

thijstriemstra, 2016-08-16:
No the 2nd comment is another udev rule i had to add but it's not related to gpio, no. Should i open a new ticket for that?

thijstriemstra, 2016-08-16:
Im using an eglfs Qt 5.7 app and without this vchiq rule the app wouldn't start.

FooDeas, 2016-08-16:
A second ticket is not needed. I'll keep that in mind and implement that soon.
I think, until then, `installation-files.list` can do the job for you, too.

FooDeas, 2016-08-16:
I did a bit research and I'm pretty sure, that the `vchiq` udev rule is redundant.
Group `video` is set already without the rule:
`crw-rw---- 1 root video 248, 0 Aug  5 15:05 /dev/vchiq`

Try the new commit c7ff9d4, which has the two new installer variables. Feel free to close the issue if it fits to your conception!

thijstriemstra, 2016-08-16:
Here's the problem I ran into without the vchiq rule: https://raspberrypi.stackexchange.com/questions/19436/how-can-i-permanently-fix-dev-vchiq-permission-errors. I'll do a fresh install and see what happens without the rule. Thanks for all the enhancements!

FooDeas, 2016-08-16:
That would be curious because the rule doesn't change anything in result. Permissions and owners keep absolutely the same.
Maybe this was changed during the last 2 years... I'm expecting your report.

FooDeas, 2016-08-16:
I found a c/p-mistake which didn't add the user to the system groups correctly and belongs to this, too.
So please checkout to 39ea9e2.

thijstriemstra, 2016-08-16:
ha @FooDeas I was about to report an error my build ends with "/etc/init.d/rcS: line 885: syntax error: bad substitution". There we go again..

thijstriemstra, 2016-08-16:
@FooDeas it's failing again (on a different line) with `etc/init.d/rcS: line 880: syntax error: bad substitution`. My configuration is:

```
disable_root=1
username=foo
userpw=bar
usersysgroups=lpadmin,input
user_is_admin=1
timezone=Europe/Amsterdam
locales="en_US.UTF-8 nl_NL"
system_default_locale="en_US.UTF-8"
gpu_mem=512
sound_enable=1
camera_enable=1
camera_disable_led=1
usergpio=1
usergpu=1
```

FooDeas, 2016-08-16:
Checkout 7712ae3693d532ee7791c75e00b9594136066a14.

FooDeas, 2016-08-17:
Results?

thijstriemstra, 2016-08-17:
@FooDeas just tried again and everything installed fine. Unfortunately it seems `sudo` is missing and I have no way to install other packages.

```
foo@bar:~$ sudo apt-get install nano
-bash: sudo: command not found
foo@bar:~$ apt-get install nano
E: Could not open lock file /var/lib/dpkg/lock - open (13: Permission denied)
E: Unable to lock the administration directory (/var/lib/dpkg/), are you root?
foo@bar:~$ /sbin/s
setcap             shadowconfig       slattach           sulogin            swapoff            switch_root        
sfdisk             shutdown           start-stop-daemon  swaplabel          swapon             sysctl   
```
See my config in prev comment.

FooDeas, 2016-08-17:
This was broken with 39ea9e2. 0d8d89c should **finally** do the trick.
I'll add some code sanitation because inconsistencies (from base repository) this afternoon and fully test it before merging that to master branch.
Nevertheless, you could use 0d8d89c because the sanitation doesn't change functionality.

thijstriemstra, 2016-08-17:
Thanks, testing it now.

thijstriemstra, 2016-08-17:
Tried again but still no sudo. I saw that during installation the following message was printed: `Can't find package libraspberrypi-bin for inclusion.` Is this maybe the issue?

FooDeas, 2016-08-17:
Not really but I fixed everything and tested it (bf797fa712eb0855ddca760faf4166bb91e3f36f) successfully, using your config.

thijstriemstra, 2016-08-17:
@FooDeas awesome it worked! And you were right, the vchiq isn't needed any longer.

closed 2016-08-17 by thijstriemstra