FooDeas/raspberrypi-ua-netinst #10: Hide startup messages with custom image

thijstriemstra, 2016-08-16:
It would be really nice to be able to specify a custom boot logo and hide the system startup messages.

I would then start my app, which it's already doing using a systemd service, but it shows the login prompt for a couple seconds during the app startup that I'd also like to mask with the boot logo image.

bassemolufda, 2016-08-17:
Hi thijstriemstra,

i found a Script called asplashscreen with google. Maybe this hels you?

```
#! /bin/sh
### BEGIN INIT INFO
# Provides:          asplashscreen
# Required-Start:
# Required-Stop:
# Should-Start:
# Default-Start:     S
# Default-Stop:
# Short-Description: Show custom splashscreen
# Description:       Show custom splashscreen
### END INIT INFO

do_start () {

    /usr/bin/fbi -T 1 -noverbose -a /etc/splash/splash.png
    exit 0
}

case "$1" in
  start|"")
    do_start
    ;;
  restart|reload|force-reload)
    echo "Error: argument '$1' not supported" >&2
    exit 3
    ;;
  stop)
    # No-op
    ;;
  status)
    exit 0
    ;;
  *)
    echo "Usage: asplashscreen [start|stop]" >&2
    exit 3
    ;;
esac

:
```
sincelery Uwe

thijstriemstra, 2016-08-17:
thanks for the suggestion @bassemolufda

FooDeas, 2016-08-17:
Additionally you have to replace `loglevel=3` with `quiet` in `/boot/cmdline.txt`.

thijstriemstra, 2016-08-18:
Thanks for that addition @FooDeas. Is it possible to also [hide the pi logo](http://www.instructables.com/id/Raspberry-Pi-Digital-Signage-Exchange-Rate-Display/step17/Hiding-the-Raspberry-Pi-Logo-on-startup/) as described here with a new option?

thijstriemstra, 2016-08-18:
Looks like video is [also supported](http://blog.sheasilverman.com/2013/09/adding-a-startup-movie-to-your-raspberry-pi/) with the same `asplashscreen` script, very cool.

thijstriemstra, 2016-08-18:
Note that self: the asplashscreen script also needs a `# Required-Start: $local_fs` according to [this comment](http://blog.sheasilverman.com/2013/09/adding-a-startup-movie-to-your-raspberry-pi/#comment-30252).

FooDeas, 2016-08-22:
The `asplashscreen` script adding/customization should be done via `install_files` (see `README.md`).
The `logo.nologo` kernel parameter will be added to next version with the `quiet_boot=1` option.

closed 2016-08-23 by FooDeas

thijstriemstra, 2016-11-30:
The latest raspbian has this nice 'pixel' startup image. Any way we can add an option that allows you to replace/add such an image?

FooDeas, 2016-11-30:
Any info which way they implemented that?