FooDeas/raspberrypi-ua-netinst #11: Can not run Kodi.

bassemolufda, 2016-08-17:
Hi there,

I can install Kodi, but Kodi needs libraspberryi0 to run. After that the Pi boots not more.
But the networkinstaller is good, maybo good for ServerOS, it support the Logitech DiNovo Keyboard.

How can i use Kodi with this Installer?
```
apt-get install libraspberrypi0
Paketlisten werden gelesen... Fertig
Abhängigkeitsbaum wird aufgebaut.       
Statusinformationen werden eingelesen.... Fertig
Die folgenden zusätzlichen Pakete werden installiert:
  raspberrypi-bootloader-nokernel
Die folgenden Pakete werden ENTFERNT:
  raspberrypi-bootloader raspberrypi-kernel

Die folgenden NEUEN Pakete werden installiert:
  libraspberrypi0 raspberrypi-bootloader-nokernel
0 aktualisiert, 2 neu installiert, 2 zu entfernen und 0 nicht aktualisiert.

Es müssen 3.897 kB an Archiven heruntergeladen werden.
Nach dieser Operation werden 104 MB Plattenplatz freigegeben.
Möchten Sie fortfahren? [J/n] 
```

FooDeas, 2016-08-17:
Are you sure, you used v1.1.1? If you come from an older release, just read #6 how to fix it.

bassemolufda, 2016-08-17:
I have used the V1.1.0

I have found a workaround for me. I have downloaded and extract all files from the libraspberrypi0.deb manuall. After that i have run ldconfig. Now Kodi works with the RPI-Display with fbcp.

 Great work with this installer on a PI3. Slim and fast. Sincelery Uwe

FooDeas, 2016-08-17:
Your workaround is a bit dirty but of course works, too.
If you do the fix I wrote in #6, it should clean that up and make it updatable via apt. Just give it a try and report if you want!

closed 2016-08-17 by FooDeas

bassemolufda, 2016-08-17:
yes... it's dirty and hard. Your Solution is clean and it works fine. Now i have used your Fix-Solution. Danke sehr. I am Lucky. Kodi with RPI-Display and Logitech Keyboard and mouse.... Vielen Dank

FooDeas, 2016-08-17:
Gern geschehen! Grüße aus Stuttgart!