FooDeas/raspberrypi-ua-netinst #12: Run ssh on a different port

thijstriemstra, 2016-08-17:
I use the user_ssh_pubkey setting for login but I'd like to do this on a different port. Because this is after first boot I cannot easily change it myself (unless I use installation.files for sshd_config) so it would be nice to have this as a setting. If you think it's overkill I'll look into a different way of changing this. Disabling password logins for SSH would be nice as well (even when my user has a password or is root).

FooDeas, 2016-08-17:
> Disabling password logins for SSH would be nice as well (even when my user has a password or is root)
This was a switch I already wanted to implement.

Changing the SSH port is pretty easy. But I'd prefer to leave that as a function in a `post-install.txt`:
```
sshd_port=22  # new Port
sed -i "s/Port 22/Port "${sshd_port}"/" /rootfs/etc/ssh/sshd_config
```

FooDeas, 2016-08-17:
Disabling root password login for SSH done with edeb182bdb6b72fdad0e1b2d7349217fe9291441.

closed 2016-08-17 by FooDeas