FooDeas/raspberrypi-ua-netinst #15: Network timeout during boot

thijstriemstra, 2016-08-23:
By default the network setup doesn't have a timeout, resulting in a really long boottime when no ethernet cable is plugged in.

I've solved this by setting a timeout of 15 secs (my app doesn't have connectivity most of the time). Can we add an option for this?

You basically need to create (or replace) a `/lib/systemd/system/networking.service.d/network-pre.conf` with permissions `root:root  644 ` with the following contents:

```
[Unit]
After=network-pre.target
	
[Service]
TimeoutStartSec=15
```

FooDeas, 2016-08-24:
There is cleaner solution I implemented with efc6e64a7294090b8ccdbc4baad550968008cc12 which should fit your and the most other users requirements:
In `/etc/network/interfaces` two lines were added for each interface:
```
auto <iface>
allow-hotplug <iface>
```
Just remove the line with `auto` and leave the `allow-hotplug` line.
Only `lo` should have the `auto` line (and no `allow-hotplug`).

This allows to wait for a requested dhcp answer if a network connection is established or boot immediately if there's no cable or other ethernet adapter plugged in.

Is this sufficient for you?

thijstriemstra, 2016-08-25:
I didn't know @FooDeas, I'll test and get back to you.

thijstriemstra, 2016-08-30:
Thanks, this works fine.

closed 2016-08-30 by thijstriemstra