FooDeas/raspberrypi-ua-netinst #16: Add yourself to LICENSE

thijstriemstra, 2016-08-23:
The LICENSE needs to be updated.

FooDeas, 2016-08-24:
I'm not a lawyer and I'm not very familiar with self licensing.
But I know, I have to deploy the original license of this fork and I do that with pleasure!
But also I think you're right, that my modifications are unlicensed at this time and I would like to change this. If you could suggest a source or text, which is appropriate, I'd be thankful!

thijstriemstra, 2016-08-25:
It's the MIT license and you can simply add your name with the copyright year.

FooDeas, 2016-08-25:
Thanks @thijstriemstra! I didn't notice that. :laughing:

closed 2016-08-26 by FooDeas

thijstriemstra, 2016-09-15:
fyi: looks like github detects the license as 'ISC' which is very similar to the MIT license, e.g. https://gist.github.com/indexzero/10602128