FooDeas/raspberrypi-ua-netinst #17: Locale warning

thijstriemstra, 2016-08-29:
When I apt-get upgrade packages on the raspberry (using ssh connection) I keep seeing this annoying warning:

```
perl: warning: Setting locale failed.
perl: warning: Please check that your locale settings:
	LANGUAGE = (unset),
	LC_ALL = (unset),
	LC_TIME = "nl_NL.UTF-8",
	LC_MONETARY = "nl_NL.UTF-8",
	LC_ADDRESS = "nl_NL.UTF-8",
	LC_TELEPHONE = "nl_NL.UTF-8",
	LC_NAME = "nl_NL.UTF-8",
	LC_MEASUREMENT = "nl_NL.UTF-8",
	LC_IDENTIFICATION = "nl_NL.UTF-8",
	LC_NUMERIC = "nl_NL.UTF-8",
	LC_PAPER = "nl_NL.UTF-8",
	LANG = "en_US.UTF-8"
    are supported and installed on your system.
perl: warning: Falling back to a fallback locale ("en_US.UTF-8").
```

My config contains:

```
locales="en_US.UTF-8 nl_NL"
system_default_locale="en_US.UTF-8"
```

Is this something that can be set during installation or is it related to SSH?

FooDeas, 2016-08-29:
If you do `cat /etc/locale.alias | grep nl`, you see that there is no `nl_NL.UTF-8` - only `nl_NL.ISO-8859-1`.
If you use `locales="... nl_NL"`, the installer tries to install nl_NL.UTF-8, which doesn't exist.

To fix this in your existing system, using `dpkg-reconfigure locales` should do that.
For future installations, just use the full, correct alias: `locales="en_US.UTF-8 nl_NL.ISO-8859-1"` (in your case)

Of course, I could add support to automatically choose non-UTF8 locales, too...

thijstriemstra, 2016-08-29:
Thanks, you're right it contains:

```
dutch		nl_NL.ISO-8859-1
```

FooDeas, 2016-08-29:
Hmm - but it exists in `/etc/locale.gen`...
I'll do some research on this issue...

thijstriemstra, 2016-08-29:
@FooDeas I got the faulty locale name for Dutch from the readme:

```
locales=                  # a space delimited list of locales that will be generated during install
                          #   (e.g. "en_US.UTF-8 nl_NL sl_SI.UTF-8")
```

FooDeas, 2016-08-29:
Yes - see above. I have to check, what's wrong there...

FooDeas, 2016-08-29:
Without try to fix that - could you execute `locale-gen` and send me the output?
Does the issue persist after doing that?

thijstriemstra, 2016-08-29:
```
$ locale-gen
-bash: locale-gen: command not found
```

FooDeas, 2016-08-29:
Here we are! Thanks!
The installer chooses `nl_NL.ISO-8859-1` which ist the first matching line in `locale-gen`, your SSH client asks for `nl_NL.UTF-8` - party.
That's it. I'll fix this!

By that time, it's correct:

> To fix this in your existing system, using `dpkg-reconfigure locales` should do that.
> For future installations, just use the full, correct alias: `locales="en_US.UTF-8 nl_NL.UTF-8"` (in your case)

thijstriemstra, 2016-08-29:
@FooDeas I haven't reboot yet but it seems the warning only shows when there's an actual package to install or upgrade so it seems this didn't fix it yet..

```
$ sudo apt-get install libraspberrypi-dev
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following NEW packages will be installed:
  libraspberrypi-dev
0 upgraded, 1 newly installed, 0 to remove and 0 not upgraded.
Need to get 404 kB of archives.
After this operation, 3323 kB of additional disk space will be used.
Get:1 http://archive.raspberrypi.org/debian/ jessie/main libraspberrypi-dev armhf 1.20160620-1 [404 kB]
Fetched 404 kB in 2s (189 kB/s)             
perl: warning: Setting locale failed.
perl: warning: Please check that your locale settings:
	LANGUAGE = (unset),
	LC_ALL = (unset),
	LC_TIME = "nl_NL.UTF-8",
	LC_MONETARY = "nl_NL.UTF-8",
	LC_ADDRESS = "nl_NL.UTF-8",
	LC_TELEPHONE = "nl_NL.UTF-8",
	LC_NAME = "nl_NL.UTF-8",
	LC_MEASUREMENT = "nl_NL.UTF-8",
	LC_IDENTIFICATION = "nl_NL.UTF-8",
	LC_NUMERIC = "nl_NL.UTF-8",
	LC_PAPER = "nl_NL.UTF-8",
	LANG = "en_US.UTF-8"
    are supported and installed on your system.
perl: warning: Falling back to a fallback locale ("en_US.UTF-8").
Selecting previously unselected package libraspberrypi-dev.
(Reading database ... 44526 files and directories currently installed.)
Preparing to unpack .../libraspberrypi-dev_1.20160620-1_armhf.deb ...
Unpacking libraspberrypi-dev (1.20160620-1) ...
Setting up libraspberrypi-dev (1.20160620-1) ...
```

FooDeas, 2016-08-29:
Post updated output of `locale-gen` again.

thijstriemstra, 2016-08-29:
@FooDeas 

```
sudo locale-gen
Generating locales (this might take a while)...
  en_US.UTF-8... done
  nl_NL.ISO-8859-1... done
Generation complete.
```
Seemed generating took longer this time.

FooDeas, 2016-08-29:
Okay here a bit more precise:
To fix that in existing systems - execute `dpkg-reconfigure locales`. In the shown list, you chose `en_US.UTF-8` and `nl_NL.UTF-8`. Unchose `nl_NL.ISO-8859-1`.
Leave all other settings as given.

thijstriemstra, 2016-08-29:
@FooDeas, followed your steps:

```
$ sudo /usr/sbin/dpkg-reconfigure locales
Generating locales (this might take a while)...
  en_US.UTF-8... done
  nl_NL.UTF-8... done
Generation complete.
```
And that seems to have worked:

```
$ sudo apt-get install curl
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following extra packages will be installed:
  libcurl3
The following NEW packages will be installed:
  curl libcurl3
0 upgraded, 2 newly installed, 0 to remove and 0 not upgraded.
Need to get 427 kB of archives.
After this operation, 748 kB of additional disk space will be used.
Do you want to continue? [Y/n] y
Get:1 http://mirrordirector.raspbian.org/raspbian/ jessie/main libcurl3 armhf 7.38.0-4+deb8u4 [231 kB]
Get:2 http://mirrordirector.raspbian.org/raspbian/ jessie/main curl armhf 7.38.0-4+deb8u4 [195 kB]
Fetched 427 kB in 3s (122 kB/s)
Selecting previously unselected package libcurl3:armhf.
(Reading database ... 44897 files and directories currently installed.)
Preparing to unpack .../libcurl3_7.38.0-4+deb8u4_armhf.deb ...
Unpacking libcurl3:armhf (7.38.0-4+deb8u4) ...
Selecting previously unselected package curl.
Preparing to unpack .../curl_7.38.0-4+deb8u4_armhf.deb ...
Unpacking curl (7.38.0-4+deb8u4) ...
Processing triggers for man-db (2.7.0.2-5) ...
Setting up libcurl3:armhf (7.38.0-4+deb8u4) ...
Setting up curl (7.38.0-4+deb8u4) ...
Processing triggers for libc-bin (2.19-18+deb8u4) ...
```
I assume this manual step is not necessary any more once the installer is fixed?

FooDeas, 2016-08-29:
Right. By that time, you can use the full local label `nl_NL.UTF-8` in your installer config.

thijstriemstra, 2016-08-29:
Can confirm it's fixed.

thijstriemstra, 2016-09-14:
It seems the issue is back.. But a different warning (at least the last part of it):

```
perl: warning: Setting locale failed.
perl: warning: Please check that your locale settings:
	LANGUAGE = (unset),
	LC_ALL = (unset),
	LC_TIME = "nl_NL.UTF-8",
	LC_MONETARY = "nl_NL.UTF-8",
	LC_ADDRESS = "nl_NL.UTF-8",
	LC_TELEPHONE = "nl_NL.UTF-8",
	LC_NAME = "nl_NL.UTF-8",
	LC_MEASUREMENT = "nl_NL.UTF-8",
	LC_IDENTIFICATION = "nl_NL.UTF-8",
	LC_NUMERIC = "nl_NL.UTF-8",
	LC_PAPER = "nl_NL.UTF-8",
	LANG = "en_US.UTF-8"
    are supported and installed on your system.
perl: warning: Falling back to the standard locale ("C").
debconf: unable to initialize frontend: Dialog
debconf: (No usable dialog-like program is installed, so the dialog based frontend cannot be used. at /usr/share/perl5/Debconf/FrontEnd/Dialog.pm line 76, <> line 107.)
debconf: falling back to frontend: Readline
```

Using:

```
locales="en_US.UTF-8 nl_NL.UTF-8"
system_default_locale="en_US.UTF-8"
```

FooDeas, 2016-09-14:
Which revision did you use?

thijstriemstra, 2016-09-14:
I used up until https://github.com/FooDeas/raspberrypi-ua-netinst/commit/b40f6ff0238a83bc99719e183b2c18eacb3d85b1

FooDeas, 2016-09-14:
I changed nothing, that could have an impact on this...
But I just released v1.3.0 and tested your locale settings with it on an RPi3 successfully!

If you still have this problem, look into the logfile `/var/log/raspberrypi-ua-netinst.log` and search for this:
```
...
  Enabling locales... en_US.UTF-8... nl_NL.UTF-8... OK
  Generating locales (this might take a while)...
    en_US.UTF-8... done
    nl_NL.UTF-8... done
  Generation complete.
  Setting system default locale... OK
...
```
If there is something other after `Enabling locales...`, please post it!

thijstriemstra, 2016-09-14:
Thanks for your feedback @FooDeas, I'll take a look at the log file. To me it's not a critical issue though, I consider it being cosmetic ;)

thijstriemstra, 2016-09-15:
@FooDeas the error message says the `locales` package isn't installed... Should I add this to my packages or will you add this?

FooDeas, 2016-09-15:
It is already part of the server packages: [rcS](https://github.com/FooDeas/raspberrypi-ua-netinst/blob/master/scripts/etc/init.d/rcS#L575)
And on all tests on 3 different Pis there were no issues while installation.

thijstriemstra, 2016-09-15:
@FooDeas but I'm using the `minimal` package! :)

FooDeas, 2016-09-16:
That explains a lot! :grin:
Note that `minimal` doesn't deliver useful features like memory improvement (raspi-copies-and-fills), too.
If the missing `locales` package doesn't drive apt-get into aborting and only displays a warning, I'd keep it in `minimal` because it's just a feature.

You can easily fix it:
Comment the line `AcceptEnv LANG LC_*` in `/etc/ssh/sshd_config`.

But:

1. I have to update the README.md regarding the flavors - they are outdated.
2. I could add/change the documentation, so you can see what packages are required. (i.e. Camera with `base`?)

FooDeas, 2016-09-16:
I decided to add the package dynamically if the `locales` or `system_default_locale` is used.

thijstriemstra, 2016-09-16:
That's a good idea @FooDeas.

Also found the log and fyi it said:

```
Enabling locales... en_US.UTF-8... NOT found... nl_NL.UTF-8... NOT found... OK
Not generating locales as the 'locales' package isn't installed !
NOT setting system default locale as the 'locale' package isn't installed !
```

Will test again now.

FooDeas, 2016-09-16:
Could you test a35a9f313fc2d073c58c7afaf40d96f723270807 (dev branch)?

thijstriemstra, 2016-09-16:
@FooDeas will do. I switched to the `server` preset btw. I noticed that the `dialog` package will be installed as well but a quick test shows that package doesn't exist?

```
sudo apt-get install dialog
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Package dialog is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Package 'dialog' has no installation candidate
```

thijstriemstra, 2016-09-16:
Apparently the last part of the warning,

```
debconf: unable to initialize frontend: Dialog
debconf: (No usable dialog-like program is installed, so the dialog based frontend cannot be used. at /usr/share/perl5/Debconf/FrontEnd/Dialog.pm line 76, <> line 107.)
debconf: falling back to frontend: Readline
```

can be removed with the `DEBIAN_FRONTEND=noninteractive` env var.

FooDeas, 2016-09-16:
Also notice `./doc/locales.txt` and `README.md` in a504ac6f58197f79840447696cbe195278a4e11b.
`locales` should now be comma separated - but it is backward compatible.

That's the first time I see that dialog package problem. It's available: [PPA](http://mirrordirector.raspbian.org/raspbian/pool/main/d/dialog/)
Do you use a transparent proxy or something that could disturb the connection? I saw you had more general problems while downloading, too...

FooDeas, 2016-09-16:
For testing the locales issue - keep the minimal preset. It should add the package automatically.

thijstriemstra, 2016-09-16:
Looks like the `dialog` package missing was due to a cancelled `sudo apt-get update`, screwing up the installation. Tested with your branch (with `server` preset, I'm not using `minimal` anymore) and output:

```
Enabling locales... en_US.UTF-8,en_US.UTF-8,nl_NL.UTF-8... NOT found... OK
  Generating locales (this might take a while)...
  Generation complete.
  Setting system default locale... FAILED !
```

with config:

```
locales="en_US.UTF-8,nl_NL.UTF-8"
system_default_locale="en_US.UTF-8"
```

FooDeas, 2016-09-16:
I missed a translation line. It should be fixed in dev 0c75d87f37ca4c901d8bfb82df58a67aba455c20.

FooDeas, 2016-09-16:
I'm testing ef716ddf30a7f1f3673a07d4736e96003ee0e378 just now. If it's working, I'll merge it into master branch and finally close here.

thijstriemstra, 2016-09-16:
Also testing with latest dev branch now, should be ready in 30.

FooDeas, 2016-09-16:
Don't waste your time! I pushed 02286b6fbdf52aaaacab0b768581dcab8529b58e for you.

FooDeas, 2016-09-16:
It finally works here! Pretty hard, annoying work. But it improves the usability notably!

thijstriemstra, 2016-09-16:
Looks like it worked here as well:

```
Enabling locales... en_US.UTF-8... en_US.UTF-8... nl_NL.UTF-8... OK
  Generating locales (this might take a while)...
    en_US.UTF-8... done
    nl_NL.UTF-8... done
  Generation complete.
  Setting system default locale... OK
```

Thanks!

closed 2016-09-16 by thijstriemstra

FooDeas, 2016-09-16:
The content check failed because of an c/p mistake, so the `system_default_locale` is redundant. But this is just cosmetic.
This is fixed and __tested__ with c1ef10516be00af55348aba8d8e1dd0439e15d0b:
```
locales="en_US.UTF-8,nl_NL.UTF-8"
system_default_locale="en_US.UTF-8"
```
>   Enabling locales... en_US.UTF-8... nl_NL.UTF-8... OK
>   Generating locales (this might take a while)...
>     en_US.UTF-8... done
>     nl_NL.UTF-8... done
>   Generation complete.
>   Setting system default locale: en_US.UTF-8 UTF-8 OK

Knocking-off time!

thijstriemstra, 2016-10-03:
Tried compiling Python 3.5.2 from source but it failed with a locale error. This command shows errors:

```
$ locale -a
locale: Cannot set LC_CTYPE to default locale: No such file or directory
locale: Cannot set LC_MESSAGES to default locale: No such file or directory
locale: Cannot set LC_COLLATE to default locale: No such file or directory
C
C.UTF-8
POSIX
en_US.utf8
nl_NL.utf8
```

Any idea?

thijstriemstra, 2016-10-03:
The locale command has more info:

```
$ locale
locale: Cannot set LC_CTYPE to default locale: No such file or directory
locale: Cannot set LC_MESSAGES to default locale: No such file or directory
locale: Cannot set LC_ALL to default locale: No such file or directory
LANG=en_US.UTF-8\ UTF-8
LANGUAGE=
LC_CTYPE="en_US.UTF-8 UTF-8"
LC_NUMERIC="en_US.UTF-8 UTF-8"
LC_TIME="en_US.UTF-8 UTF-8"
LC_COLLATE="en_US.UTF-8 UTF-8"
LC_MONETARY="en_US.UTF-8 UTF-8"
LC_MESSAGES="en_US.UTF-8 UTF-8"
LC_PAPER="en_US.UTF-8 UTF-8"
LC_NAME="en_US.UTF-8 UTF-8"
LC_ADDRESS="en_US.UTF-8 UTF-8"
LC_TELEPHONE="en_US.UTF-8 UTF-8"
LC_MEASUREMENT="en_US.UTF-8 UTF-8"
LC_IDENTIFICATION="en_US.UTF-8 UTF-8"
LC_ALL=
```

FooDeas, 2016-10-03:
Confirmed. I'll fix it tomorrow.

thijstriemstra, 2016-10-03:
Thanks!

Contents of `/etc/default/locale`:
```
#  File generated by update-locale
LANG=en_US.UTF-8 UTF-8
```

FooDeas, 2016-10-04:
Workaround/short fix: `update-locale LANG="en_US.UTF-8"` and logout user
Fix is coming soon. Testing first.

FooDeas, 2016-10-04:
Fixed again.

thijstriemstra, 2016-10-05:
New installation:

```
$ locale
LANG=en_US.UTF-8
LANGUAGE=
LC_CTYPE="en_US.UTF-8"
LC_NUMERIC=nl_NL.UTF-8
LC_TIME=nl_NL.UTF-8
LC_COLLATE="en_US.UTF-8"
LC_MONETARY=nl_NL.UTF-8
LC_MESSAGES="en_US.UTF-8"
LC_PAPER=nl_NL.UTF-8
LC_NAME=nl_NL.UTF-8
LC_ADDRESS=nl_NL.UTF-8
LC_TELEPHONE=nl_NL.UTF-8
LC_MEASUREMENT=nl_NL.UTF-8
LC_IDENTIFICATION=nl_NL.UTF-8
LC_ALL=
```

Looks good, thanks.