FooDeas/raspberrypi-ua-netinst #18: Add custom debian repository

thijstriemstra, 2016-08-29:
I have a custom debian repo with my app on a machine in the local network.

Post installation I run something like:

```
# add foo.apt to hosts file
echo "Adding foo.apt to hosts file..."
echo $1"    foo.apt" | sudo tee -a /etc/hosts

# enable deb-src
echo "Enabling deb-src..."
echo "deb-src http://mirrordirector.raspbian.org/raspbian jessie main" | sudo tee -a /etc/apt/sources.list

# add foo apt repository
echo "Adding foo repository..."
wget -O - http://foo.apt/foo.gpg.key| sudo apt-key add -
echo 'deb http://foo.apt/apt/debian jessie main' > foo.list
sudo mv foo.list /etc/apt/sources.list.d/
sudo apt-get update
sudo apt-get upgrade --assume-yes
```
Would be nice to have some sort of setting(s) for a custom repo but if you think it's overkill, please close.

FooDeas, 2016-08-29:
It's very easy:
- New file `config/apt/foo.com.list`  
contains: `deb http://foo.apt/apt/debian jessie main`
- New file `config/apt/foo.gpg.key`  
contains your key file

That's it!

closed 2016-08-29 by FooDeas

thijstriemstra, 2016-08-29:
@FooDeas thanks, that works great.