FooDeas/raspberrypi-ua-netinst #20: Auto pair bluetooth device(s)

thijstriemstra, 2016-08-29:
I have a bluetooth keyboard that I'd like to pair by default. Is this maybe something for a config setting, a list of bluetooth identifiers or something that are trusted by default? I haven't looked into the whole bluetooth support yet but maybe someone tried doing this before.

FooDeas, 2016-08-30:
BT pairing needs communication in both directions. So it isn't done by modifying a config file on one side only.
Scripting in post-install would be possible because I activate the BT on Pi3 during installation process.

thijstriemstra, 2016-09-14:
I'm not able to list the internal bluetooth device for a RPi3, e.g.:

```
hcitool scan
Device is not available: No such device

```
Installed following packages after some searching:

```
bluez,bluetooth,raspberrypi-sys-mods,pi-bluetooth,libbluetooth3,libbluetooth-dev,libdbus-c++-dev,libdbus-1-dev
```
Also added user to bluetooth group as suggested [here](https://www.pi-supply.com/make/fix-raspberry-pi-3-bluetooth-issues/):
```
sudo usermod -G bluetooth -a pi
```
But still not working. Any idea?

FooDeas, 2016-09-14:
I'm working on it at the moment and got it working. Please be patient until the new release in about 12 hours.

FooDeas, 2016-09-14:
To get BT and UART working at the same time was a bit tricky.
It's fixed and tested with b2986a416d1bc19188153ea966721ab83b1d1eae.

Short hint: All packages you need are installed with `apt-get install pi-bluetooth`.

thijstriemstra, 2016-09-14:
Cool, I'll test it now. I saw you removed the GPIO udev rule, I suppose it's still working without that?

thijstriemstra, 2016-09-14:
Awesome it works now:

```
$ hcitool dev
Devices:
	hci0	B8:27:EB:E4:20:75
```

closed 2016-09-14 by thijstriemstra

FooDeas, 2016-09-14:
For your information:
The GPIO rule I added to get #8 working, was removed because the package `raspberrypi-sys-mods` is installed by default now. It delivers another rule which does the same and gets updates if available.