FooDeas/raspberrypi-ua-netinst #21: Enable SPI

thijstriemstra, 2016-09-05:
I need to use SPI to interact with a led matrix. Can we add an option that enables this (instructions [here](https://github.com/rm-hull/max7219#pre-requisites))?

FooDeas, 2016-09-05:
Do we need it? If you look in the well commented ´/boot/config.txt´ you see the following:
```
# Uncomment some or all of these to enable the optional hardware interfaces
#dtparam=i2c_arm=on
#dtparam=i2s=on
#dtparam=spi=on
```
Just uncomment and reboot. There's no need for rpi-config - but I added the basic compatibility, so you can use it, too.

thijstriemstra, 2016-09-05:
ah sorry missed that, thanks!

FooDeas, 2016-09-05:
If you find missing parameters in `config.txt` - like the camera was - I'd definitely add it, too!

closed 2016-09-06 by FooDeas

thijstriemstra, 2016-10-04:
I enabled the following in config.txt:

```
dtparam=i2c_arm=on
dtparam=i2s=on
dtparam=spi=on
```
But when I reboot `dmesg | grep spi` is supposed to show up but the command doesn't return anything. Does this work for you @FooDeas?

FooDeas, 2016-10-04:
It definitely is. Do an apt-get update/upgrade to see, if that changes something. The package "raspberrypi-kernel" is required (latest version is from 2016-09-21).
If this doesn't work, try rpi-update to update these files from latest build repository because this issue is pretty new and the deb package is from 2 weeks ago.

thijstriemstra, 2016-10-04:
```
$ sudo apt-get install  rpi-update
$ sudo rpi-update
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
 *** Performing self-update
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 12022  100 12022    0     0  52627      0 --:--:-- --:--:-- --:--:-- 52960
 *** Relaunching after update
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
 *** We're running for the first time
 *** Backing up files (this will take a few minutes)
 *** Backing up firmware
 *** Backing up modules 4.4.21-v7+
This update bumps to rpi-4.4.y linux tree
Be aware there could be compatibility issues with some drivers
Discussion here:
https://www.raspberrypi.org/forums/viewtopic.php?f=29&t=144087
##############################################################
 *** Downloading specific firmware revision (this will take a few minutes)
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   168    0   168    0     0    316      0 --:--:-- --:--:-- --:--:--   316
100 51.7M  100 51.7M    0     0   281k      0  0:03:08  0:03:08 --:--:--  373k
 *** Updating firmware
 *** Updating kernel modules
 *** depmod 4.4.23+
 *** depmod 4.4.23-v7+
 *** Updating VideoCore libraries
 *** Using HardFP libraries
 *** Updating SDK
 *** Running ldconfig
 *** Storing current firmware revision
 *** Deleting downloaded files
 *** Syncing changes to disk
 *** If no errors appeared, your firmware was successfully updated to 667cfabe63bc663383559ef88317e86f9bd41e45
 *** A reboot is needed to activate the new firmware
```

After reboot still nothing..

FooDeas, 2016-10-04:
Post `lsmod`.

thijstriemstra, 2016-10-04:
@FooDeas 

```
$ lsmod
Module                  Size  Used by
bnep                   10340  2 
hci_uart               17943  1 
btbcm                   5929  1 hci_uart
bluetooth             326105  22 bnep,btbcm,hci_uart
cpufreq_stats           3463  0 
brcmfmac              186339  0 
brcmutil                5661  1 brcmfmac
cfg80211              427855  1 brcmfmac
joydev                  9024  0 
snd_soc_bcm2835_i2s     6354  0 
rfkill                 16037  3 cfg80211,bluetooth
snd_soc_core          125885  1 snd_soc_bcm2835_i2s
snd_bcm2835            20447  0 
snd_pcm_dmaengine       3391  1 snd_soc_core
snd_pcm                75762  3 snd_bcm2835,snd_soc_core,snd_pcm_dmaengine
snd_timer              19288  1 snd_pcm
snd                    51908  4 snd_bcm2835,snd_soc_core,snd_timer,snd_pcm
i2c_bcm2708             4834  0 
bcm2835_gpiomem         3040  0 
spi_bcm2835             6678  0 
bcm2835_wdt             3225  0 
evdev                  11396  0 
uio_pdrv_genirq         3164  0 
uio                     8000  1 uio_pdrv_genirq
ipv6                  347530  34
```

FooDeas, 2016-10-04:
The driver (spi_bcm2835) is loaded and the device is up. Looks like it's already working without a dmesg debug/info output.

FooDeas, 2016-10-04:
I just tried it without rpi-update. It works as well.

thijstriemstra, 2016-10-04:
Hmm, alright. Does the loopback test listed [here](https://www.raspberrypi.org/documentation/hardware/raspberrypi/spi/README.md#troubleshooting) work for you?

Only output I see is:

```
$ ./spidev_test -D /dev/spidev0.0
spi mode: 0
bits per word: 8
max speed: 500000 Hz (500 KHz)

00 00 00 00 00 00 
00 00 00 00 00 00 
00 00 00 00 00 00 
00 00 00 00 00 00 
00 00 00 00 00 00 
00 00 00 00 00 00 
00 00 
```

Fuck what a time waster :(

thijstriemstra, 2016-10-04:
I've also tested this on a Raspberrypi model B + v1.2 with Raspbian wheezy:

```
$ uname -a
Linux raspberrypi 4.1.19+ #858 Tue Mar 15 15:52:03 GMT 2016 armv6l GNU/Linux
```
and the loopback works fine there after adding `dtparam=spi=on`:

```
spi mode: 0
bits per word: 8
max speed: 500000 Hz (500 KHz)

FF FF FF FF FF FF 
40 00 00 00 00 95 
FF FF FF FF FF FF 
FF FF FF FF FF FF 
FF FF FF FF FF FF 
DE AD BE EF BA AD 
F0 0D 
```

And the max7219 example now also works fine. Sigh. I suppose I'll also test it on a RPi2 with the latest and greatest to see if it's not a hardware issue with the Pi3 I was using before. I also haven't done a clean raspberrypi-ua-netinst installation yet with spi so maybe it'll work after a fresh start there.

FooDeas, 2016-10-04:
On RPi 3:
`root@pi:~# dmesg | grep spi
(no output)
root@pi:~# ./spidev_test -D /dev/spidev0.0
spi mode: 0
bits per word: 8
max speed: 500000 Hz (500 KHz)
FF FF FF FF FF FF
40 00 00 00 00 95
FF FF FF FF FF FF
FF FF FF FF FF FF
FF FF FF FF FF FF
DE AD BE EF BA AD
F0 0D
`
Works as intended...

thijstriemstra, 2016-10-04:
On that Raspberry B+:

```
$dmesg | grep spi
[ 5.817699] spi spi0.0: setting up native-CS0 as GPIO 8
[ 5.818605] spi spi0.1: setting up native-CS1 as GPIO 7
```
I used the same GPIOs on Rpi3 and Rpi1B+, but only Rpi1B+ succeeded the loopback test.

FooDeas, 2016-10-04:
RPi 3 and RPi B+ have the same Pin layout regarding SPI MISO/MOSI. Because the device exists on your RPi, it has to work.
I did nothing special: Fresh installed, uncommented the SPI in config.txt, rebooted, installed gcc and typed in the three lines for the loopback test.

FooDeas, 2016-10-04:
Many people abused their Pis with 5V from Arduino hardware/shields. I hope, you accidentally didn't...

thijstriemstra, 2016-10-04:
Yea fuck!! hehe we'll see.

FooDeas, 2016-10-04:
I already noticed it above. But it doesn't seem to be a software issue.

thijstriemstra, 2016-10-05:
With a fresh system installation the RaspberryPi 3 SPI still doesn't work, but its not fried, the GPIO channels still work. I'll leave this rest for a week and take a look again later, perhaps try on a RPi2 in the meantime.

thijstriemstra, 2016-10-30:
It eventually started working on that particular machine, it probably was a broken package or kernel. Thanks for the suggestions though.