FooDeas/raspberrypi-ua-netinst #22: Enable I2C

thijstriemstra, 2016-09-05:
I want to experiment with an OLED screen using the [SSD1306 library](https://github.com/rm-hull/ssd1306) which requires I2C. Can an option be added for this (instructions [here](https://github.com/rm-hull/ssd1306#pre-requisites))?

FooDeas, 2016-09-05:
See #21.

thijstriemstra, 2016-10-23:
@FooDeas thanks for adding a i2c [option](https://github.com/FooDeas/raspberrypi-ua-netinst/blob/master/doc/INSTALL_CUSTOM.md#device--peripheral). After reboot I see `i2c_bm2708` enabled:

```
$ lsmod | grep i2c
i2c_bcm2708             4834  0
```
But the acual devices in /dev/ are not created. After some searching it seems i2c_dev also needs to be enabled. To accomplish this I had to add the following lines, compared to an otherwise empty file apart from the comments, to `/etc/modules`:

```
# /etc/modules: kernel modules to load at boot time.
#
# This file contains the names of kernel modules that should be loaded
# at boot time, one per line. Lines beginning with "#" are ignored.

i2c-dev
i2c-bm2708
```
After reboot the devices are available:

```
$ lsmod | grep i2c
i2c_bcm2708             4834  0 
i2c_dev                 5859  0
```

Is there something we can do about this? Shouldn't the dev devices always be available?

FooDeas, 2016-10-24:
👍
Please put this in a new issue. Thanks!

thijstriemstra, 2016-10-24:
Opened #43

closed 2016-10-24 by thijstriemstra