FooDeas/raspberrypi-ua-netinst #23: GUI

thijstriemstra, 2016-09-14:
Would be nice to have a cross-platform configuration tool for this project. Something with Qt would make sense to me. I'm not sure if it can or should be backwards compatible with the original version of this fork, but I'd say it shouldn't.

Thoughts on where (the repository) it should be placed?

thijstriemstra, 2016-09-14:
I can probably whip up something pretty easily with Qt.

FooDeas, 2016-09-14:
Sounds nice! What can I imagine under?
A kind of ´installer-config.txt´ maker? If you'd make it open source - GitHub is a nice place to be...

thijstriemstra, 2016-09-14:
Yeah exactly. But everytime you add a feature it also needs to be added to the GUI, which should be no problem. I think it'll add maybe 130kb to this repo so mabye its worth consdering to add it here.

FooDeas, 2016-09-14:
With pleasure! Why don't you tell that straight? :grin:
But I'd prefer to begin in a dev branch first.

thijstriemstra, 2016-09-15:
I think a web-based (e.g. Django) thing would be nicer and allows you to store configurations into the database etc.

Jodes81, 2016-10-09:
Funny enough I was thinking of doing this in PHP. Hosting it online somewhere would really make it accessible. And doing it online would make it really easy for windows users so nothing like Python etc would need to be installed. The question is where it could be hosted, since regular 30MB downloads would exceed the quota for any free service I know of.

closed 2023-08-15 by FooDeas