FooDeas/raspberrypi-ua-netinst #24: Missing log file

thijstriemstra, 2016-09-15:
After a succesful installation (that failed twice) I have two log files, one ending with:
```
Waiting 5 seconds.....
Waiting for /dev/mmcblk0... 
  Disk /dev/mmcblk0: 14.5 GiB, 15523119104 bytes, 30318592 sectors
  Disklabel type: dos
  Disk identifier: 0x04130dc6
OK
Applying new partition table... OK
Initializing /boot as vfat... OK
Copying /boot files in... OK
Initializing / as ext4... OK
Mounting new filesystems... OK
Starting install process...
P: Retrieving Release
P: Retrieving Release.gpg
P: Validating Release
I: Good signature from "Mike Thompson (Raspberry Pi Debian armhf ARMv6+VFP) <mpthompson@gmail.com>"
P: Parsing Release
P: Retrieving Packages.gz
P: Validating Packages.gz
E: Couldn't validate Packages!

Oh noes, something went wrong!
You have 10 seconds to hit ENTER to get a shell...
```
And one with:

```
Waiting 5 seconds.....
Waiting for /dev/mmcblk0... 
  Disk /dev/mmcblk0: 14.5 GiB, 15523119104 bytes, 30318592 sectors
  Disklabel type: dos
  Disk identifier: 0xbfa557d0
OK
Applying new partition table... OK
Initializing /boot as vfat... OK
Copying /boot files in... OK
Initializing / as ext4... OK
Mounting new filesystems... OK
Starting install process...
P: Retrieving Release
P: Retrieving Release.gpg
P: Validating Release
I: Good signature from "Mike Thompson (Raspberry Pi Debian armhf ARMv6+VFP) <mpthompson@gmail.com>"
P: Parsing Release
P: Retrieving Packages.gz
P: Validating Packages.gz
P: Parsing Packages
P: Retrieving gcc-4.9-base
P: Validating gcc-4.9-base
E: Couldn't download pool/main/g/gcc-4.9/gcc-4.9-base_4.9.2-10_armhf.deb!

Oh noes, something went wrong!
You have 10 seconds to hit ENTER to get a shell...
```
Unfortunately the last log file wasn't stored? Can you reproduce this?

FooDeas, 2016-09-15:
Sounds like problems with the SD card. Try another!

thijstriemstra, 2016-09-15:
Well, I dont think that's necessarily it. The second log shows it's unable to download. For some reason this can happen multiple times during an install, maybe it's my connection, maybe it's the repo, I don't know. And at the end, the third time, everything installed fine but **no** log file.

FooDeas, 2016-09-16:
> When the installation completes successfully, the logfile is moved to /var/log/raspberrypi-ua-netinst.log on the installed system.
> When an error occurs during install, the logfile is moved to the sd card, which gets normally mounted on /boot/ and will be named raspberrypi-ua-netinst-<datetimestamp>.log

So there will be no logfile in `/boot/` when it was successful. May that be the point?

thijstriemstra, 2016-09-16:
@FooDeas ah damn you're totally right, sorry for the confusion.

closed 2016-09-16 by thijstriemstra