FooDeas/raspberrypi-ua-netinst #27: Cannot install over WiFi

Jodes81, 2016-09-24:
The installation hangs on "Configuring eth0 with DHCP" - I've tried a couple of WiFi dongles, same problem each time.

I also tried installing via wired Ethernet which worked, and then getting WiFi to work, but often it failed to get a lease, and when it managed, it was unable to ping anything. What could be going on here?

Jodes81, 2016-09-24:
I just re-read the readme. Sounds like the installer can only be run over WiFi if the Pi itself has built-in WiFi - so it won't work on the older Pi's with a dongle

FooDeas, 2016-09-24:
It can work if your dongle is supported. First: the devicename has to be wlan0. Try it and report. You should check dmesg if it doesn't work! So we can find out what's missing for your dongle.

Jodes81, 2016-09-24:
@FooDeas thank you for the quick response! I'd love for UA-NetInst to work, very powerful. 

First: I'll say that I did a sanity check - I got WiFi working successfully with the default Raspbian installed by NOOBS, by following [this tutorial](https://www.raspberrypi.org/documentation/configuration/wireless/wireless-cli.md). So in theory it should work.

Second: Trying to get the raspberrypi-ua-netinst to install over WiFi, I used the default install files, with the following modifications:

_**installer-config.txt:**_
```
packages=nano
rootsize=+3800M
cmdline="dwc_otg.lpm_enable=0 console=serial0,115200 elevator=deadline fsck.repair=yes"
wlan_ssid=TheRoutersSSID
wlan_psk=TheRoutersPassword
```

Using the UART and a serial cable to my PC, I logged the following during the install, with the network cable unplugged, and just the WiFi dongle inserted:

```
Uncompressing Linux... done, booting the kernel.
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Initializing cgroup subsys cpuset
[    0.000000] Initializing cgroup subsys cpu
[    0.000000] Initializing cgroup subsys cpuacct
[    0.000000] Linux version 4.4.13+ (dc4@dc4-XPS13-9333) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611) ) #894 Mon Jun 13 12:43:26 BST 2016
[    0.000000] CPU: ARMv6-compatible processor [410fb767] revision 7 (ARMv7), cr=00c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT nonaliasing instruction cache
[    0.000000] Machine model: Raspberry Pi Model B Rev 1
[    0.000000] cma: Reserved 8 MiB at 0x0a400000
[    0.000000] Memory policy: Data cache writeback
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 48720
[    0.000000] Kernel command line: dma.dmachans=0x7f35 bcm2708_fb.fbwidth=1824 bcm2708_fb.fbheight=984 bcm2708.boardrev=0x3 bcm2708.serial=0xda25038e smsc95xx.macaddr=B8:27:EB:25:03:8E bcm2708_fb.fbswap=1 bcm2708.uart_clock=3000000 vc_mem.mem_base=0xec00000 vc_mem.mem_size=0x10000000  dwc_otg.lpm_enable=0 consoleblank=0 console=ttyAMA0,115200 elevator=deadline rootwait
[    0.000000] PID hash table entries: 1024 (order: 0, 4096 bytes)
[    0.000000] Dentry cache hash table entries: 32768 (order: 5, 131072 bytes)
[    0.000000] Inode-cache hash table entries: 16384 (order: 4, 65536 bytes)
[    0.000000] Memory: 160556K/196608K available (6051K kernel code, 436K rwdata, 1840K rodata, 380K init, 726K bss, 27860K reserved, 8192K cma-reserved)
[    0.000000] Virtual kernel memory layout:
[    0.000000]     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
[    0.000000]     fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
[    0.000000]     vmalloc : 0xcc800000 - 0xff800000   ( 816 MB)
[    0.000000]     lowmem  : 0xc0000000 - 0xcc000000   ( 192 MB)
[    0.000000]     modules : 0xbf000000 - 0xc0000000   (  16 MB)
[    0.000000]       .text : 0xc0008000 - 0xc07bce2c   (7892 kB)
[    0.000000]       .init : 0xc07bd000 - 0xc081c000   ( 380 kB)
[    0.000000]       .data : 0xc081c000 - 0xc0889388   ( 437 kB)
[    0.000000]        .bss : 0xc0889388 - 0xc093ef30   ( 727 kB)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] NR_IRQS:16 nr_irqs:16 16
[    0.000029] sched_clock: 32 bits at 1000kHz, resolution 1000ns, wraps every 2147483647500ns
[    0.000072] clocksource: timer: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 1911260446275 ns
[    0.000172] bcm2835: system timer (irq = 27)
[    0.000471] Console: colour dummy device 80x30
[    0.000507] Calibrating delay loop... 697.95 BogoMIPS (lpj=3489792)
[    0.060310] pid_max: default: 32768 minimum: 301
[    0.060673] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.060703] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.061763] Disabling cpuset control group subsystem
[    0.061812] Initializing cgroup subsys io
[    0.061850] Initializing cgroup subsys memory
[    0.061907] Initializing cgroup subsys devices
[    0.061941] Initializing cgroup subsys freezer
[    0.061971] Initializing cgroup subsys net_cls
[    0.062071] CPU: Testing write buffer coherency: ok
[    0.062147] ftrace: allocating 20620 entries in 61 pages
[    0.175468] Setting up static identity map for 0x81c0 - 0x81f8
[    0.177222] devtmpfs: initialized
[    0.185380] VFP support v0.3: implementor 41 architecture 1 part 20 variant b rev 5
[    0.185925] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.187657] pinctrl core: initialized pinctrl subsystem
[    0.188514] NET: Registered protocol family 16
[    0.193773] DMA: preallocated 4096 KiB pool for atomic coherent allocations
[... 2546 lines cut ...]
```
eth0      Link encap:Ethernet  HWaddr b8:27:eb:25:03:8e
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

wlan0     Link encap:Ethernet  HWaddr 00:0f:13:24:03:94
          BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

```

FooDeas, 2016-09-24:
Please post the /etc/networks/interfaces you used at the last step.

Jodes81, 2016-09-24:
**_/etc/networks/interfaces_**
```
# interfaces(5) file used by ifup(8) and ifdown(8)
# Please note that this file is written to be used with dhcpcd
# For static IP, consult /etc/dhcpcd.conf and 'man dhcpcd.conf'
# Include files from /etc/network/interfaces.d:
source-directory /etc/network/interfaces.d
auto lo
iface lo inet loopback
iface eth0 inet manual
allow-hotplug wlan0

iface wlan0 inet manual
    wpa-conf /etc/wpa_supplicant/wpa_supplicant.conf

allow-hotplug wlan1
iface wlan1 inet manual
    wpa-conf /etc/wpa_supplicant/wpa_supplicant.conf
```

**_/etc/wpa_supplicant/wpa_supplicant.conf_**:
```
country=GB
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1
network={
    ssid=TheSSID
    psk=ThePassword
}
```

FooDeas, 2016-09-24:
1. Here you have a working interfaces file when you are using a model without internal wifi:
```
# interfaces(5) file used by ifup(8) and ifdown(8)
# Please note that this file is written to be used with dhcpcd
# For static IP, consult /etc/dhcpcd.conf and 'man dhcpcd.conf'
# Include files from /etc/network/interfaces.d:
source-directory /etc/network/interfaces.d
auto lo
iface lo inet loopback

allow-hotplug eth0
iface eth0 inet dhcp

allow-hotplug wlan0
iface wlan0 inet dhcp
    wpa-conf /etc/wpa_supplicant/wpa_supplicant.conf
```
2. You can skip modifying `/etc/apt/sources.list`. Everything needed is already included.
3. The installer-config.txt has to include these **three** lines:
```
...
ifname=wlan0
wlan_ssid=<yourSSID>
wlan_psk=<yourPass>
```
4. Try again with the missing line and I can see, what's missing when you check console output or dmesg if possible. (Firmware? Modules?)

Jodes81, 2016-09-24:
It all works! Thank you so much!!

closed 2016-09-24 by Jodes81

FooDeas, 2016-09-24:
I don't believe the installer works using a wifi dongle (it was never intended). Using the integrated wifi of model 3 is a unique feature.
But of course, the installed system does without any problems. Glad I could help you.

Jodes81, 2016-09-24:
Well, it worked for my dongle :) I honestly recommend this installer to anyone - it's something the RPi really needed

FooDeas, 2016-09-24:
Okay... 😮
Nice to hear. Thanks for report!