FooDeas/raspberrypi-ua-netinst #28: Ethernet boot

thijstriemstra, 2016-09-29:
Apparently it's possible to boot using ethernet, skipping the fragile SD-card: https://www.raspberrypi.org/blog/pi-3-booting-part-ii-ethernet-all-the-awesome/

Would love to use this with a RaspberryPi 3.

FooDeas, 2016-10-16:
It depends on #5. Queued.

kpfleming, 2020-03-16:
I use netbooting on most of my Pis, but I don't think it would be practical to implement support for this in the installer (I tried). Doing so requires being able to mount the target root and /boot filesystems using NFS in the installer itself, which would increase complexity significantly.

I'm satisfied using an SD card to run the installer and get a working system, then convert that system to netboot/NFS manually after the installation is complete.

jbragdon1, 2020-10-31:
Looking over the code base, and with an admittedly limited understanding of whats going on under the hood, it looks like early on assumptions are made that there is a local file system. Once the base systems are loaded into ram - I suspect up front, checking for SD card and then writting a temp directory with tools might get this going nicely..... I would love to see this as an option linkted with a local proxy copy of the raspian installer. Effectively, plug in a pi and get it going.

closed 2023-08-15 by FooDeas