FooDeas/raspberrypi-ua-netinst #30: Missing newline

thijstriemstra, 2016-10-05:
I have an installer config with some gpu_mem stuff and a custom /boot/config.txt containg:

```
# Enable optional hardware interfaces
dtparam=i2c_arm=on
dtparam=i2s=on
dtparam=spi=on
```

After installation completed `/boot/config.txt` looked like this:

```
# Enable optional hardware interfaces
dtparam=i2c_arm=on
dtparam=i2s=on
dtparam=spi=onstart_x=1
disable_camera_led=1

gpu_mem=512
```

Looks like a newline was missing from my config.txt and when the final config.txt was produced it didn't add a newline, causing the dtparam option to get garbled: `dtparam=spi=onstart_x=1`

FooDeas, 2016-10-05:
Included in v1.4.0 it last minute.

thijstriemstra, 2016-10-11:
Still seeing the issue, refs #36

closed 2016-10-11 by FooDeas

thijstriemstra, 2016-10-12:
Confirmed it's fixed.