FooDeas/raspberrypi-ua-netinst #31: Reinstall fails: depmod: can't change directory to 'lib/modules/4.4.21+': No such file or directory

Jodes81, 2016-10-05:
Following the `README`, I wanted to reinstall, with a few modifications. 

I made some modifications to the files in `/boot/`, eg adding "bring-your-own" files to `my-files.list` and `/boot/config/files`.

Then I run:
```
mv /boot/config-reinstall.txt /boot/config.txt
reboot
```
During the  installation, I get this:

```Waiting for serial device... OK

Searching for boot device...
SD card detected.
Boot files found on SD card.

==================================================
raspberrypi-ua-netinst
==================================================
Revision v1.3.0 (6e71081)
Built on Wed Sep 14 23:34:00 CEST 2016
Running on Raspberry Pi version B
==================================================
https://github.com/FooDeas/raspberrypi-ua-netinst/
==================================================
Starting HWRNG succeeded!
Copying boot files... OK
==================================================
=== Start executing installer-config.txt. ========
=== Finished executing installer-config.txt. =====
==================================================

Network configuration:
  ip_addr = dhcp
  wlan_ssid = <...>
  wlan_psk = <...>
  online_config =

depmod: can't change directory to 'lib/modules/4.4.21+': No such file or directory
Loading additional drivers.
  Loading driver 'brcmfmac'... modprobe: can't change directory to '4.4.21+': No such file or directory

Oh noes, something went wrong!
You have 10 seconds to hit ENTER to get a shell...
```

I noticed that the files in `/boot/`, such as the `bcm*.dtb` files are different to those in the original installer files. Might this be the problem?

FooDeas, 2016-10-05:
There was an issue with the kernel for Raspberry Pi version B.
Backup your installer-config.txt and other changes you made from /boot. Then delete all files and folders there and replace them with the new v1.4.0. I think this will fix it for you.
Please report!

FooDeas, 2016-10-05:
I see the problem: Reinstallation fails because the **kernel** (kernel.img) has been updated by the running system. The installer loads it on reinstall boot and misses the modules in initramfs.

To fix it, read my suggestion above. Thanks for reporting!

closed 2016-10-05 by FooDeas

Jodes81, 2016-10-05:
You're welcome, thanks for sorting it! Excellent tool.