FooDeas/raspberrypi-ua-netinst #33: Dynamic package installation configuration variables

Jodes81, 2016-10-06:
I was reading the latest release notes for new features in v1.4.1, it says there's documentation on this feature, but I can't find it? Are there any examples of this behaviour, and how it differs from previous versions?

FooDeas, 2016-10-06:
See README.md and the new files in folder "./doc/".
The behaviour never changes. There will always be either new customization features added or deprecated old ones removed.

closed 2016-10-06 by FooDeas

FooDeas, 2016-10-07:
Sorry - I misunderstood you. The feature "dynamically add packages if needed by configuration variables" means that if you i.e. chose to set a default keyboard layout with the preset "minimal", the packages that need to be installed to make that work is automatically selected for installation.