FooDeas/raspberrypi-ua-netinst #36: installation-files.list parse errors

thijstriemstra, 2016-10-11:
I guess this is 2 issues, see log below.

 - a http500 error on the raspberrypi.org server causes the jessie/main raspberrypi-bootloader to be ignored (manually installing the package moments later works fine)
- having comments in `installation-files.list` causes parse errors (?)

```
  Get:145 http://archive.raspberrypi.org/debian/ jessie/main libi2c-dev all 3.1.1+svn-2 [10.5 kB]
  Get:146 http://archive.raspberrypi.org/debian/ jessie/main raspberrypi-kernel armhf 1.20160921-1 [30.0 MB]
  Err http://archive.raspberrypi.org/debian/ jessie/main raspberrypi-bootloader armhf 1.20160921-1
    500  Internal Server Error
  Get:147 http://archive.raspberrypi.org/debian/ jessie/main libraspberrypi0 armhf 1.20160921-1 [840 kB]
  Get:148 http://archive.raspberrypi.org/debian/ jessie/main libraspberrypi-bin armhf 1.20160921-1 [328 kB]
  Fetched 75.2 MB in 4min 11s (299 kB/s)
  E: Failed to fetch http://archive.raspberrypi.org/debian/pool/main/r/raspberrypi-firmware/raspberrypi-bootloader_1.20160921-1_armhf.deb  500  Internal Server Error
  E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?
OK
Preserving original config.txt and kernels...
Configuring bootloader to start the installed system...
Creating default cmdline.txt... OK
=================================================
=== Start executing post-install.txt. ===
Adding files & folders listed in /bootfs/config/installation-files.list
awk: cmd. line:1: Unexpected token
awk: cmd. line:1: Unexpected token
awk: cmd. line:1: Unexpected token
    
chmod: invalid mode: ''
Try 'chmod --help' for more information.
chown: cannot access '': No such file or directory
awk: cmd. line:1: Unexpected token
awk: cmd. line:1: Unexpected token
awk: cmd. line:1: Unexpected token
    
chmod: invalid mode: ''
Try 'chmod --help' for more information.
chown: cannot access '': No such file or directory
awk: cmd. line:1: Unexpected token
awk: cmd. line:1: Unexpected token
awk: cmd. line:1: Unexpected token
    
chmod: invalid mode: ''
Try 'chmod --help' for more information.
chown: cannot access '': No such file or directory
awk: cmd. line:1: Unexpected token
awk: cmd. line:1: Unexpected token
awk: cmd. line:1: Unexpected token
    
chmod: invalid mode: ''
Try 'chmod --help' for more information.
chown: cannot access '': No such file or directory

=== Finished executing post-install.txt. ===
=================================================
Removing cdebootstrap-helper-rc.d... OK
Saving current time for fake-hwclock...

Installation finished at Tue Oct 11 18:06:42 UTC 2016 and took 17 min 33 sec (1053 seconds)
```

Contents of `installation-files.list`:

```
# To add custom files on the target system:
# Add one line per file using this format:
# <ownership> <permission> <path/filename>
#
# <ownership>     OWNER[:GROUP] (see `man chown`)
# <permission>    MODE  (see `man chmod`)
# <path/file>     /path/as/it/will/appear/on/the/target/system.txt
#
# Source files are stored in the directory `config/files` in the same path as
# where they will be stored on the target system.
#
# EXAMPLE:
# A file placed on the SD-card in /config/files/home/pi/.bashrc is listed as follows:
# pi:pi  755  /home/pi/.bashrc
#
# Don't forget to call the function:
# install_files my-files.list
# from your `post-install.txt` where "my-files.list" is this file.
# The function `install_files` will assume the file is located in the `config` directory.

me:me  775  /home/me/install.sh
me:me  775  /home/me/install-printer.sh
root:root  600  /etc/lsyncd.conf
```

thijstriemstra, 2016-10-11:
Testing again without comments now. Note that I copy/pasted the comment text from `my-files.list`..

FooDeas, 2016-10-11:
Does the 500 persist in your network?
Please attach your installation-files.list as file so I can do some experiments.

thijstriemstra, 2016-10-11:
Tested again without comments (I think, I removed comments in installation-files.list in boot and moved config-reinstall.txt) and similar errors, different package for the 500 this time:

```
Fetched 76.6 MB in 3min 45s (340 kB/s)
  E: Failed to fetch http://archive.raspberrypi.org/debian/pool/main/f/firmware-nonfree/firmware-libertas_0.43+rpi5_all.deb  500  Internal Server Error
  E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?
OK
Preserving original config.txt and kernels...
Configuring bootloader to start the installed system...
Creating default cmdline.txt... OK
=================================================
=== Start executing post-install.txt. ===
Adding files & folders listed in /bootfs/config/installation-files.list
awk: cmd. line:1: Unexpected token
awk: cmd. line:1: Unexpected token
awk: cmd. line:1: Unexpected token
    
chmod: invalid mode: ''
Try 'chmod --help' for more information.
chown: cannot access '': No such file or directory
awk: cmd. line:1: Unexpected token
awk: cmd. line:1: Unexpected token
awk: cmd. line:1: Unexpected token
    
chmod: invalid mode: ''
Try 'chmod --help' for more information.
chown: cannot access '': No such file or directory
awk: cmd. line:1: Unexpected token
awk: cmd. line:1: Unexpected token
awk: cmd. line:1: Unexpected token
    
chmod: invalid mode: ''
Try 'chmod --help' for more information.
chown: cannot access '': No such file or directory
awk: cmd. line:1: Unexpected token
awk: cmd. line:1: Unexpected token
awk: cmd. line:1: Unexpected token
    
chmod: invalid mode: ''
Try 'chmod --help' for more information.
chown: cannot access '': No such file or directory

=== Finished executing post-install.txt. ===
=================================================
Removing cdebootstrap-helper-rc.d... OK
Saving current time for fake-hwclock...

Installation finished at Tue Oct 11 18:44:24 UTC 2016 and took 17 min 23 sec (1043 seconds)
```

Attached `installation-files.list` (as txt cause of github restrictions).
[installation-files.list.txt](https://github.com/FooDeas/raspberrypi-ua-netinst/files/522677/installation-files.list.txt)

.

FooDeas, 2016-10-11:
Thanks!

Change proposal:
- try updating package list up to 3 times (instead once), otherwise fail the installation
- allow comments in install_files method

Opinions?

thijstriemstra, 2016-10-11:
Yes sounds good :+1: 

install_files should basically work with https://github.com/FooDeas/raspberrypi-ua-netinst/blob/master/config/my-files.list which has the same comment.

FooDeas, 2016-10-11:
Please attach your post-install.txt too.

FooDeas, 2016-10-11:
Could you test and report 420cd5ba3c7ba1a0c53be5a0893557a7a1159b39 if it's working for you?

thijstriemstra, 2016-10-11:
@FooDeas yep, testing now.

thijstriemstra, 2016-10-11:
I'm also hoping that this somehow fixes the SPI problems I was having (maybe a dependency wasn't installed, causing my SPI to malfucntion.

FooDeas, 2016-10-11:
OT:
Regarding your SPI issue you could send me your complete installer settings (via E-Mail if they shouldn't be public) - so I can test it here and check it on my hardware.

thijstriemstra, 2016-10-11:
@FooDeas looks like that fixed it, thanks! And I might get back to you on that SPI issue.

closed 2016-10-11 by FooDeas

thijstriemstra, 2016-10-11:
@FooDeas after installation I started my app and it looks like the camera isn't enabled any longer even though it's enabled in the config..

```
  File ":/picamera/camera.py", line 415, in __init__
  File ":/picamera/camera.py", line 444, in _init_camera
Uncaught exception <class 'picamera.exc.PiCameraError'>: Camera is not enabled. Try running 'sudo raspi-config' and ensure that the camera has been enabled.
```

FooDeas, 2016-10-11:
Check `raspistill` to see if it's a low level problem.

FooDeas, 2016-10-11:
The output of `vcgencmd get_camera` should always be checked, too.

thijstriemstra, 2016-10-11:
```
$ sudo vcgencmd get_camera
supported=0 detected=0
```

thijstriemstra, 2016-10-11:
```
$ raspistill -fp
mmal: mmal_vc_component_create: failed to create component 'vc.ril.camera' (1:ENOMEM)
mmal: mmal_component_create_core: could not create component 'vc.ril.camera' (1)
mmal: Failed to create camera component
mmal: main: Failed to create camera component
mmal: Camera is not enabled in this build. Try running "sudo raspi-config" and ensure that "camera" has been enabled
```

FooDeas, 2016-10-11:
If `vcgencmd get_camera` shows `detected=0` the problem is the firmware or the "config.txt".
Check the installer log, if the firmware was installed properly and check (or post) the "config.txt".

FooDeas, 2016-10-11:
Checking the cable connection might be useful, too.

thijstriemstra, 2016-10-11:
The config has (according to the log):

```
  gpu_mem = 512
  quiet_boot = 1
  sound_enable = 1
  camera_enable = 1
  camera_disable_led = 1
```

There's nothing in the log that sticks out. 

`config.txt` contains:

```
# Enable optional hardware interfaces
dtparam=i2c_arm=on
dtparam=i2s=on
dtparam=spi=onstart_x=1
disable_camera_led=1

gpu_mem=512
```

hmm looks like the newline issue is still there. Fixing manually and rebooting now.

thijstriemstra, 2016-10-11:
Yea that was the issue, ugh. Reopening #30

thijstriemstra, 2016-10-12:
> Would be useful to print something in the installation log when enabling the camera.

@FooDeas can you print a message in the log when the camera (and led) is enabled for the installer?

FooDeas, 2016-10-12:
The installer variable normally do what they should do and they are already shown by installer and and in the log file.
Because the config.txt changes aren't active while installing, checking the results isn't possible at this time.
I thought about a config.txt plausibility test - but it wouldn't be useful while installing. So I think it would be the best to find and fix issues like this and cover conditions that haven't been foreseen by the code authors.

pmontrasio, 2016-11-01:
I run into the same http 500 error right now but without parse errors. I didn't turn on my Raspberry since mid August and I was upgrading many packages. Only the firmware ones had problems.

```
E: Failed to fetch http://archive.raspberrypi.org/debian/pool/main/r/raspberrypi-firmware/libraspberrypi-bin_1.20161020-1_armhf.deb  500  Internal Server Error [IP: 2a00:1098:0:82:1000:13:0:7 80]

E: Failed to fetch http://archive.raspberrypi.org/debian/pool/main/r/raspberrypi-firmware/libraspberrypi0_1.20161020-1_armhf.deb  500  Internal Server Error [IP: 2a00:1098:0:82:1000:13:0:7 80]

E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?
```

I added an empty line at the end of ```/boot/config.txt``` and rebooted. ```apt-get upgrade``` worked this time. I didn't do another ```apt-get update```.

```
Get:1 http://archive.raspberrypi.org/debian/ jessie/main libraspberrypi-bin armhf 1.20161020-1 [328 kB]
Get:2 http://archive.raspberrypi.org/debian/ jessie/main libraspberrypi0 armhf 1.20161020-1 [841 kB]
Fetched 1,169 kB in 0s (2,376 kB/s)      
```

FooDeas, 2016-11-02:
Your added empty line to `config.txt` definitively has nothing to do with this. The corresponding issue only occurred only if you provided your own special `config.txt` to the installer.

Your issue is a `Internal Server Error`. I recognized some problems with the raspberrypi.org server the last ~5 days. They are rare - but present for some minutes.

I'm sure, the issues above are solved.