FooDeas/raspberrypi-ua-netinst #41: Ask for / generate password

thijstriemstra, 2016-10-20:
The user password is currently hardcoded in the config file (and therefore also in my git repo). Would be nice to be able ask for a password when generating the installer (?) or generate one itself (and print that out). Perhaps this is not in the scope of this project but just thought I'd bring it up.

FooDeas, 2016-10-20:
To generate and show the password wouldn't be a working solution for a headless installation.

Asking for a password results in saving it to a installer-config.txt - so we would have to code a installer-config generator. That's what it's intended by #23.
I think, your question belongs to #40!? This would be a solution. I think, this will work.

closed 2016-10-20 by FooDeas

thijstriemstra, 2016-10-20:
I guess I was more thinking of an interactive installer config generator thing. E.g.

```
$ raspi-install init
Username []:
Password [empty to generate one]:
etc..
```

FooDeas, 2016-10-20:
I won't write a generator. I confine myself to the docs because the usability of a complete, interactive generator would be bad. A menu based generator would be a possibility.
If really necessary I'd implement it from #23 when I have a kind of template.