FooDeas/raspberrypi-ua-netinst #42: installer: cleanup /boot directory

FooDeas, 2016-10-20:
The goal is, to have a clean /boot directory.
At the moment installer files and system files are mixed and it's not easy to find out, what can be safely deleted, if the installer isn't needed anymore.
So all installer related files should be placed in a specific subdir.

Challenge: provide backward compatibility to the old and convert it to the new structure.
Question: what to do, if both structures are present? Which has the priority?

thijstriemstra, 2016-10-23:
I wouldn't worry about backwards compatibility.

thijstriemstra, 2016-10-31:
It's pretty unclear (to me) where all files go now. The documentation consists of long paragraphs with loads of information. Having a simple step-by-step explanation would help, which I'll think about.

My question: I used to do this:

```
export CARD_PATH=/media/5FGT-123F
cp -v debian/app/shared/installer-config-common.txt $CARD_PATH
cp -v debian/app/${PROJECT_NAME}/installer-config.txt $CARD_PATH
cp -v debian/app/shared/post-install.txt $CARD_PATH
cp -v debian/app/shared/installation-files.list $CARD_PATH
```

What should I use now with this new structure? This?

```
export CARD_PATH=/media/5FGT-123F/raspberrypi-ua-netinst
cp -v debian/app/shared/installer-config-common.txt $CARD_PATH/config/files/
cp -v debian/app/${PROJECT_NAME}/installer-config.txt $CARD_PATH/config/files/
cp -v debian/app/shared/post-install.txt $CARD_PATH/config/
cp -v debian/app/shared/installation-files.list $CARD_PATH/config/files/
```

thijstriemstra, 2016-10-31:
The first build (with no custom config because I'm not sure where to place them, see prev comment) failed with the following:

```
Updating package lists... OK

Downloading packages...
  Reading package lists...
  Building dependency tree...
  Reading state information...
  The following extra packages will be installed:
    device-tree-compiler libraspberrypi0
  The following NEW packages will be installed:
    device-tree-compiler libraspberrypi-bin libraspberrypi0
    raspberrypi-bootloader raspberrypi-kernel raspberrypi-sys-mods
    raspi-copies-and-fills
  0 upgraded, 7 newly installed, 0 to remove and 0 not upgraded.
  Need to get 34.7 MB of archives.
  After this operation, 125 MB of additional disk space will be used.
  WARNING: The following packages cannot be authenticated!
    raspberrypi-sys-mods raspi-copies-and-fills device-tree-compiler
    raspberrypi-bootloader libraspberrypi0 libraspberrypi-bin raspberrypi-kernel
  E: There are problems and -y was used without --force-yes
ERROR: 100, trying again (2/3)...   Reading package lists...
  Building dependency tree...
  Reading state information...
  The following extra packages will be installed:
    device-tree-compiler libraspberrypi0
  The following NEW packages will be installed:
    device-tree-compiler libraspberrypi-bin libraspberrypi0
    raspberrypi-bootloader raspberrypi-kernel raspberrypi-sys-mods
    raspi-copies-and-fills
  0 upgraded, 7 newly installed, 0 to remove and 0 not upgraded.
  Need to get 34.7 MB of archives.
  After this operation, 125 MB of additional disk space will be used.
  WARNING: The following packages cannot be authenticated!
    raspberrypi-sys-mods raspi-copies-and-fills device-tree-compiler
    raspberrypi-bootloader libraspberrypi0 libraspberrypi-bin raspberrypi-kernel
  E: There are problems and -y was used without --force-yes
ERROR: 100, trying again (3/3)...   Reading package lists...
  Building dependency tree...
  Reading state information...
  The following extra packages will be installed:
    device-tree-compiler libraspberrypi0
  The following NEW packages will be installed:
    device-tree-compiler libraspberrypi-bin libraspberrypi0
    raspberrypi-bootloader raspberrypi-kernel raspberrypi-sys-mods
    raspi-copies-and-fills
  0 upgraded, 7 newly installed, 0 to remove and 0 not upgraded.
  Need to get 34.7 MB of archives.
  After this operation, 125 MB of additional disk space will be used.
  WARNING: The following packages cannot be authenticated!
    raspberrypi-sys-mods raspi-copies-and-fills device-tree-compiler
    raspberrypi-bootloader libraspberrypi0 libraspberrypi-bin raspberrypi-kernel
  E: There are problems and -y was used without --force-yes
ERROR: 100, FAILED !

Oh noes, something went wrong!
You have 10 seconds to hit ENTER to get a shell...
```

FooDeas, 2016-10-31:
> WARNING: The following packages cannot be authenticated!
That seems to be a problem with the repository. I had this a couple of times two days ago with a normal Raspbian image when I tried to install a package. It wasn't happening some hours later.

FooDeas, 2016-10-31:
Belonging to your question above:
```
export CARD_PATH=/media/5FGT-123F/raspberrypi-ua-netinst
cp -v debian/app/shared/installer-config-common.txt $CARD_PATH/config/
cp -v debian/app/${PROJECT_NAME}/installer-config.txt $CARD_PATH/config/
cp -v debian/app/shared/post-install.txt $CARD_PATH/config/
cp -v debian/app/shared/installation-files.list $CARD_PATH/config/files/
```
```
sanitize_inputfile /bootfs/raspberrypi-ua-netinst/config/installer-config-common.txt
source /bootfs/raspberrypi-ua-netinst/config/installer-config-common.txt
```

FooDeas, 2016-10-31:
It's pretty easy to understand: All configuration and customization was moved from `/` to `/raspberrypi-ua-netinst/config/`.

FooDeas, 2016-10-31:
Be careful with existing config files on sd card. The installer tries to move the old files to the new locations. But I missed to replace the paths **in** these files. This would be hard to implement and decide what to replace and what not.

I think you were right to ignore the old files...

thijstriemstra, 2016-10-31:
Thanks for the comments, I'm about to test with the new setup.

>Please check the log for errors before this!! (Do you use custom repositories?)
>Really no customizing done? This happens if using https repository sources.

Well basically, the config files were placed in the root, like before, so the config looked really empty ;)
And no, I didn't see any errors prior to that. And yes, it did install a custom apt repository I'm using, but the config files were simply in the wrong place.

thijstriemstra, 2016-10-31:
@FooDeas it installed everything correctly but my custom files were not installed. There's nothing in the log about these files either. The content of post-install.txt is:

```
install_files installation-files.list
```

Structure of dir on sd-card:

```
$ ls -l /boot/raspberrypi-ua-netinst/config/
total 12
drwxr-xr-x 2 root root 2048 okt 31 23:21 apt
drwxr-xr-x 2 root root 2048 okt 31 23:21 boot
drwxr-xr-x 4 root root 2048 okt 31 23:21 files
-rwxr-xr-x 1 root root  590 okt 31 23:21 installer-config-common.txt
-rwxr-xr-x 1 root root 1008 okt 31 23:21 installer-config.txt
-rwxr-xr-x 1 root root   37 okt 31 23:21 post-install.txt
```

FooDeas, 2016-11-02:
Thanks for your input! 👍
I'll have a look at it in some hours.

thijstriemstra, 2016-11-02:
@FooDeas I'd say config/files/.. (or both?)

FooDeas, 2016-11-02:
I have a additional idea:
The file list(s) must have the extension ".list" and must be placed in "config/files". There will be a subfolder "root" which represents the root of the sd card.
The list files will be read automatically and the post-installation script won't have to be touched for this...

What do you think about it?

thijstriemstra, 2016-11-02:
sounds good!

FooDeas, 2016-11-02:
I implemented and pushed it untested. Updated doc is in "config/files/custom_files.txt".

thijstriemstra, 2016-11-02:
I just did an `./update.sh` and the output contained a warning, not sure if it was there before, or whether its going to cause problems or not.

```
Downloading package: "http://archive.raspbian.org/raspbian/pool/main/z/zlib/zlib1g_1.2.8.dfsg-2_armhf.deb"
######################################################################## 100,0%
Verifying downloaded packages... OK
mkdir: cannot create directory ‘boot’: File exists

Downloading ./config.txt...
######################################################################## 100,0%
```

FooDeas, 2016-11-02:
You can safely ignore this. It doesn't affect anything. But I'll fix this warning.

thijstriemstra, 2016-11-02:
Did a reinstall but my files aren't there. Guess I shouldve read `config/files/custom_files.txt` and I'm missing something. Trying again..

FooDeas, 2016-11-02:
It should not be complicated:

- create a file list with extension .list in "files"
- put the directories and files in the folder "root"

Nothing else.

thijstriemstra, 2016-11-02:
@FooDeas worked great! Setup's a little more complicated but definitely cleaner.

FooDeas, 2016-11-02:
Good news! Thanks for testing! 👍

I have to revise the docs. They are much too big and complicated...
I'm just missing a short, simple option: the wifi region setting
When that's done - v1.5.0 can be tagged.

thijstriemstra, 2016-11-02:
Cool. And I'll test the #43 i2c before that.

FooDeas, 2016-11-02:
That's already included in devel.

thijstriemstra, 2016-11-02:
I don't use the wifi region setting, can't help you with that ;)

FooDeas, 2016-11-02:
No problem. It's just a text file with 2 letters.

closed 2016-11-03 by FooDeas