FooDeas/raspberrypi-ua-netinst #43: i2c-dev not enabled

thijstriemstra, 2016-10-24:
@FooDeas thanks for adding a i2c [option](https://github.com/FooDeas/raspberrypi-ua-netinst/blob/master/doc/INSTALL_CUSTOM.md#device--peripheral). After reboot I see `i2c_bm2708` enabled:

```
$ lsmod | grep i2c
i2c_bcm2708             4834  0
```
But the acual devices in /dev/ are not created. After some searching it seems i2c_dev also needs to be enabled. To accomplish this I had to add the following lines (not sure if i2c-bm2708 is also needed there), compared to an otherwise empty file apart from the comments, to `/etc/modules`:

```
# /etc/modules: kernel modules to load at boot time.
#
# This file contains the names of kernel modules that should be loaded
# at boot time, one per line. Lines beginning with "#" are ignored.

i2c-dev
i2c-bm2708
```
After reboot the devices are available:

```
$ lsmod | grep i2c
i2c_bcm2708             4834  0 
i2c_dev                 5859  0
```

```
$ sudo i2cdetect -l
i2c-1	i2c       	3f804000.i2c                    	I2C adapter
```

Is there something we can do about this? Shouldn't the dev devices always be available?

ps. apparently I should also change this option to increase performance (for [this device](https://github.com/rm-hull/ssd1306#pre-requisites)):

```
dtparam=i2c_arm=on,i2c_baudrate=400000
```

FooDeas, 2016-10-24:
Thanks for reporting this!
Adding "i2c-bm2708" to /etc/modules isn't needed. But i'll add "i2c-dev". I have to check permissions/user group, too.
The I²C baudrate will get a separate option.

thijstriemstra, 2016-10-24:
awesome, thanks.

FooDeas, 2016-10-30:
It would be nice if you could test a4bd728485cba4754d9a5e6ba32f98a340cd91f9.

Note that the directory structure for customizing has changed, too.
Please do some changes to the devel branch if you think the documentation for this is not meaningful.

thijstriemstra, 2016-10-30:
Will test it asap, probably tomorrow.

thijstriemstra, 2016-11-03:
Works fine now (with upcoming 1.5.0)!

closed 2016-11-03 by FooDeas