FooDeas/raspberrypi-ua-netinst #44: GPS and NTP

thijstriemstra, 2016-11-02:
I recently managed to hook up a [Ublox NEO-6M GPS Module](http://www.ebay.com/itm/-/322149202365) serial device, use `gpsmon` to monitor GPS activity and also use it with QtPositioning.

The device exists as `/dev/ttyS0` on a RaspberryPi 3 (`/dev/ttyAMA0` on older machines). All tutorials I followed suggested to remove the `serial0=11500` line from `/boot/cmdline.txt`. My question, is this necessary? And if it is, how would I remove this using rpi-ua-netinst?

There has been some attention for the NTP/time functionality and I wonder if I'd need to make some drastic changes in NTP in order to use GPS timestamps?

I managed to get the lang, lat, timestamp in my Qt app but when I do that, other parts of the OS cannot access the serial port for the GPS anymore (restricted to one user). So rather than using it on an app-level I'd like to use it on a system-wide level where `date` returns a GPS time. I've read a couple of tutorials and I think I know how to setup the NTP stuff but I just wonder if you see any implications using rpi-ua-netinst.

FooDeas, 2016-11-03:
First: use serial0! This is dynamically symlinked always to the serial port on gpio header.
Using ttyS0 and ttyAMA0 is **deprecated** because the vary from model to model!

Normally you can let the cmdline.txt as set by default. Test it!

FooDeas, 2016-11-03:
The changes to cmdline.txt can be done via the option "cmdline" in installer-config.txt.
The default value is documented.

thijstriemstra, 2016-11-03:
So should I use `/dev/serial0` instead but still remove it from `cmdline`?

FooDeas, 2016-11-03:
Exact.

closed 2016-11-03 by FooDeas

thijstriemstra, 2016-11-03:
Thanks, I see now that /dev/serial0 is a symlink to /dev/ttyS0, so that's ok. The reason I'm hesitant to use the cmdline option, is the fact it might change and I'll have to double check the default value and compare everytime there's a new version (or kernel?). Would be cool to specify an exclude pattern perhaps..

FooDeas, 2016-11-04:
That won't change. It became a fixed part (udev rule) of Raspbian because of the bluetooth module in model 3b.
I think, this was a good idea: For example, if you disable the bluetooth module ttyS0 will disappear and you have to use ttyAMA0.
So using ttyS0 or ttyAMA0 isn't stable.

Of course you can use the devices itself (ttyS0 in your case) if you are careful what you do - but I think you misunderstand the functionality of the cmdline option:
It just provides terminal access via UART which blocks the device. If you want to use the device to connect peripherals, the terminal access option must be removed. It has nothing to do with the devices itself!

thijstriemstra, 2016-11-04:
I ended up with something like this:

```rst
GPS
===

This document describes how to enable to UBlox NEO-6M GPS device on a
RaspberryPi 3.

Installation
------------

Install some packages::

   sudo apt-get install gpsd-clients ntpdate

Compile and install jq::

  wget https://github.com/stedolan/jq/releases/download/jq-1.5/jq-1.5.tar.gz
  tar xf jq-1.5.tar.gz
  cd jq-1.5
  ./configure
  make -j4
  sudo make install
  cd ..
  rm -rf jq*
  jq --version

Remove ``console=serial0,115200`` from ``/boot/cmdline.txt``.

Disable services::

  sudo systemctl stop serial-getty@ttyS0.service
  sudo systemctl disable serial-getty@ttyS0.service

Change ``/etc/default/gpsd`` and change the ``GPSD_OPTIONS`` setting to::

  GPSD_OPTIONS="/dev/serial0"

Make sure it's started during boot::

  sudo ln -s /lib/systemd/system/gpsd.service /etc/systemd/system/multi-user.target.wants/

Now shutdown the device.

Wire up device
--------------

========== ================
**uBlox**  **Raspberry Pi**
GND	       GND         
TX	       RX (GPIO 15)
RX	       TX (GPIO 14)
VCC	       3V3
========== ================

Test
----

Identify the device::

   gpsctl /dev/serial0

Output should be something like::

   /dev/serial0 identified as a u-blox 1.00 (59842) at 9600 baud.

Run gpsmon (Ctrl+c to exit)::

  gpsmon /dev/serial0

Show time packet. ``"mode": 1`` means the GPS sensor has not achieved satellite
lock yet::

  gpspipe -w -n 30 | grep -m 1 time | jq '.'

Show GPS packet::

  gpspipe -w -n 30 | grep -m 1 lat | jq '.'

Show device info::

  gpspipe -w -n 30 | grep -m 1 DEVICE | jq '.'

Using the GPS as an NTP Clock Source
------------------------------------

Add the following to ``/etc/ntp.conf``::

  # GPS Serial data reference
  server 127.127.28.0
  fudge 127.127.28.0 stratum 10 refid GPS

Restart NTP::

  sudo service ntp restart

Verify the current date/time::

  ntpq -p

Compare local time to a timer server::

  sudo ntpdate -d time.nist.gov
```