FooDeas/raspberrypi-ua-netinst #47: password_ssh_allow=0

thijstriemstra, 2016-11-18:
It would be nice to have a setting that disallows SSH password login. I already disabled root ssh access (`root_ssh_allow=0`) but was also under the wrong impression that this would disable password SSH logins.

FooDeas, 2016-11-18:
I'll add `user_ssh_pwlogin` and change `root_ssh_allow` to `root_ssh_pwlogin`.

FooDeas, 2016-11-19:
Implemented by c86885b6f257247ec15c169a6b680c06278cf4a1 and 52f540f9e6fbd26c5b177ed71893f85764240c33.

FooDeas, 2016-11-20:
Replaced 'user_ssh_pwlogin' with 'ssh_pwlogin' in d7b476a0240abcf39e0330f7538e34ab840fb0ee.

thijstriemstra, 2016-11-23:
Going to test this now. It was hard to find the various ssh settings, what about adding a section called SSH that has all SSH related settings? Than `ssh_pwlogin` and `root_ssh_pwlogin` can maybe become `ssh_pw_login` and `ssh_root_login`?

FooDeas, 2016-11-24:
If I do that, the sectioning becomes more inconsistent.
I'd have to change/group all ssh settings if I separate the root SSH from the root users customization section. I think, that would be more confusing...

But I simply placed the `ssh_pwlogin` option in the "Advanced" section. Maybe it's better placed anywhere else?

thijstriemstra, 2016-11-24:
I thought these were the only 2 ssh settings right now. I would place the `ssh_pwlogin` close to the other `root_ssh_pwlogin`.

FooDeas, 2016-11-24:
We have:

```
Package
Device / peripheral
Root
User
Network
Localization
Partitioning / Filesystem
Advanced
```

In "network" as last entry?

FooDeas, 2016-11-24:
(Other SSH settings are `root_ssh_pubkey` and `user_ssh_pubkey`.)

thijstriemstra, 2016-11-24:
I don't know.. They're both security related options. I find the section 'Root' odd. If you call it Security and place the options there...

thijstriemstra, 2016-11-24:
Or just call it SSH ;)

FooDeas, 2016-11-24:
Other SSH settings are `root_ssh_pubkey` and `user_ssh_pubkey` - I wouldn't like to move them.

thijstriemstra, 2016-11-24:
Ok. But in general: section Root has weird name (if you dont look at its contents) and having SSH-related options in different sections **felt** weird, I had to search, but I'll leave it up to you.

FooDeas, 2016-11-24:
Hm - I have to accept your confusion!
I'd like to see your proposal in a devel-* branch. I'm sure, we will find a pretty solution!

thijstriemstra, 2016-11-24:
Ok cool.

closed 2016-11-24 by FooDeas

thijstriemstra, 2016-11-24:
I forgot about the:

> [change] ssh_pwlogin and root_ssh_pwlogin to ssh_pw_login and ssh_root_login

now that there's a SSH section. The page index also still references `Root`.

FooDeas, 2016-11-24:
Oups. I'll fix it.

FooDeas, 2016-11-24:
I'd like to leave `ssh_pwlogin`. The underscore will be inconsistent with the other `*_pwlogin`s.
Same consistency with `root_ssh_pwlogin` <> `user_*`.

FooDeas, 2016-11-30:
@thijstriemstra wrote:
> Just tested it and I'm not able to get into the pi using ssh with following settings:
> 
> ```
> user_is_admin=1
> root_ssh_pwlogin=0
> ssh_pwlogin=0
> root_ssh_pubkey="[real key here]"
> ```
> 
> Is it just me or can you reproduce this @FooDeas?

I tested the following options using a simple 1024 Bit testkey:
(Bigger keyfiles are no problem but too huge to post them here.)
```
user_is_admin=1
root_ssh_pwlogin=0
ssh_pwlogin=0
root_ssh_pubkey="ssh-rsa AAAAB3NzaC1yc2EAAAADAQABAAAAgQC2wDUDjML/SG5BonLAwZM3iAQ6ddEVT/0XhiIuzQ7Bx7doyGk4zQVPtOItWfmw8kLmCiSITHHtNkRT+LdkakvtGX9OWHOL1UMJ5xTtHUCi56Istui2wUhPA4pHS6BuH+DwFiDJhBsLIU8msJJ6UmtCUsVILrRdE7o9yERVP3YDDw== testkey_insecure"
```
Private SSH key on local machine (saved as testkey_insecure):
```
-----BEGIN RSA PRIVATE KEY-----
MIICXAIBAAKBgQC2wDUDjML/SG5BonLAwZM3iAQ6ddEVT/0XhiIuzQ7Bx7doyGk4
zQVPtOItWfmw8kLmCiSITHHtNkRT+LdkakvtGX9OWHOL1UMJ5xTtHUCi56Istui2
wUhPA4pHS6BuH+DwFiDJhBsLIU8msJJ6UmtCUsVILrRdE7o9yERVP3YDDwIDAQAB
AoGAaTxXitrE81JXQ25zXDJcPw8ROETeCMLOAHzaNE8a31XNdQZrKS1OkK6dy/XT
BiULhwJyNRaoSCwGvsArgyR+4Wig7ZDWSj5RM9Gk2RMbIM6KjlMxDFUl4j+WVCKN
CrQCGeiEMWnGY+Sbh9uU+QCzvm3bgRSOLmNf/JMq28yGIykCQQDt1IS1Qcgm75ME
5xD8kCoAxj9fjfxWlBKyfDY1p7Cs47npn6M4iYg2bzbjXLOJj+e031gn9Xahzm7x
votezT+NAkEAxLZycd9+jbxCv0DlZqVAyOFxZWMFUXKmk28H8UyYWak9dH8TB8vk
UiEaHatls8oKsLdEm8vbXuQb7j+E699oCwJBAJkJRjfIU33SbV0bPo/2SEDlUIQx
Pi2L7JgTW2KdPA+UHd5aDdJklVTJrZ0R2qlrAyMdEIHt7l9iEXvXK6FeG0UCQFQi
LGSj0D7SrEGcggEa4NCKC7PvXBDv1mrsbBs6RVbj2ZL/3qXRAMBXxBG4gvEYkuXu
HHv9/r9hFMkbhcBUdu0CQFrHksfrFOHFHRTC+TS9wpLWh34Q4WgpHr27b10grzdH
6NoUx6pxEyZ0ceCfMTnb1Q4ZeFIxEWDJcZZoNCADK1Y=
-----END RSA PRIVATE KEY-----
```

Logon via `ssh -i testkey_insecure root@pi` from local machine works as intended.

thijstriemstra, 2016-11-30:
There's not even an `.ssh` directory in the user's home directory.

I noticed there's a empty line between my config settings, maybe that's it?

```
sanitize_inputfile /bootfs/raspberrypi-ua-netinst/config/installer-config-common.txt
source /bootfs/raspberrypi-ua-netinst/config/installer-config-common.txt

root_ssh_pubkey="foo"

hdmi_type=monitor
hdmi_monitor_res=1280x1024
```

thijstriemstra, 2016-11-30:
Oh I just noticed there's a `user_ssh_pubkey =` setting in `/var/log/raspberrypi-ua-netinst.log`. And because I'm not using the root account it'll get ignored.

@FooDeas, I missed this setting because it's not listed in the SSH section..

FooDeas, 2016-11-30:
The `root_ssh_pubkey` option is for the root user.
Additional line breaks in installer-config.txt don't harm.

FooDeas, 2016-11-30:
Should we move `user_ssh_pubkey` to the SSH section?

thijstriemstra, 2016-11-30:
Yes please!! :dancer: