FooDeas/raspberrypi-ua-netinst #48: Screen/HDMI option

thijstriemstra, 2016-11-23:
My screen requires a different HDMI resolution with the following boot settings:

```
# uncomment to force a specific HDMI mode (this will force VGA)
hdmi_group=2
hdmi_mode=35
```

Something we want to add to the settings? A HDMI or screen section maybe?

FooDeas, 2016-11-24:
Adding customization variables is useless for the installer itself.
It would be helpful for the system only. To get this done, you can already add these lines in `config/boot/config.txt`.

Do you want to have the installers `config.txt` this while building? Then I'm looking forward for suggestions how you'd like to customize the build process in this way - from users/your point of view. It would be a nice improvement!

thijstriemstra, 2016-11-24:
Is there a sort of master template of all the config options available? Perhaps even commented out? If we can extract all options we can decided what user-supplied values are actually valid (and should be included in /boot/config.txt).

FooDeas, 2016-11-24:
Execute `update.sh` and have a look at `res/initramfs/boot/config.txt`. That's the default from Raspbian and the one that will be used, if no custom config.txt is placed in `config/boot/`.

FooDeas, 2016-11-24:
But there are many more not listed on this page or "hidden" in forum threads.

thijstriemstra, 2016-11-24:
Yea that's the problem :( I suppose this document is the most complete when it comes to option but it's not parse-able.. https://www.raspberrypi.org/documentation/configuration/config-txt.md

thijstriemstra, 2016-11-24:
Well in that case, why not write anything I add into the config? Ones not recognized by this lib of course.

FooDeas, 2016-11-24:
For my understanding: Do we talk about the installation process' or the systems config.txt?

thijstriemstra, 2016-11-24:
Anything I add to` installer-config.txt` is interpreted and added to /boot/config.txt based on conditions you added right? What if you detect erroneous values, e.g. `hdmi_mode=35`, do you ignore it, or do you add it to the boot config?

FooDeas, 2016-11-24:
"Erroneous" settings are non considered variables - so they will be completely ignored.

thijstriemstra, 2016-11-24:
Well, let's add a var that will get parsed no matter what, andthat will be directed/appended to the boot config. Otherwise I'll keep bugging you for extra config options.. ;)

FooDeas, 2016-11-24:
That functionality is already implemented... Just create a (empty) `config/boot/config.txt` and add your wishes there.
They will be included in systems `config.txt`.

FooDeas, 2016-11-24:
Of course all other settings from installer-config.txt will be also considered/added to that `config.txt`.

thijstriemstra, 2016-11-24:
Ah ok, than I've been to careful and should see what happen when I add these settings :) I'll have to mark them as such though, and try not to overlap with your settings that have the same or other (side-)effects.

thijstriemstra, 2016-11-24:
Side-note: the options I'm asking for are mentioned in the file format example at the top of the page.. ;)

https://www.raspberrypi.org/documentation/configuration/config-txt.md

FooDeas, 2016-11-24:
You haven't to be too careful: Settings by `installer-config.txt` always overwrite the corresponding settings in the base "template" you provide in `config/boot/config.txt`.

FooDeas, 2016-11-25:
I thought about it and will add hdmi options. I hope to find a solution to get this applied during installation process, too.

FooDeas, 2016-11-25:
Implementation done but untested. I hope, you can assist here.

The installation itself is not affected by the new options. I still have no good idea how the usability could look like...

thijstriemstra, 2016-11-26:
I'll test it early next week, thanks!

thijstriemstra, 2016-11-30:
Just tested it and I'm not able to get into the pi using ssh with following settings:

```
user_is_admin=1
root_ssh_pwlogin=0
ssh_pwlogin=0
root_ssh_pubkey="[real key here]"
```

Is it just me or can you reproduce this @FooDeas?

FooDeas, 2016-11-30:
Interesting! I'll try it and report.

FooDeas, 2016-11-30:
@thijstriemstra
I moved your request belonging SSH to #47.

FooDeas, 2016-11-30:
With 1d6af7239320da4d7b006e674571695566e3c38e, the HDMI options should apply to the installation process, too.
I'm looking forwart for your test results. Please test it without providing a custom `config.txt`, too!

thijstriemstra, 2016-11-30:
Looks good!

closed 2016-11-30 by FooDeas