FooDeas/raspberrypi-ua-netinst #52: Ability to point SSH key settings to files

thijstriemstra, 2016-11-30:
Instead of hard-coding the SSH keys in the config, point it to a file instead, e.g.:

```
user_ssh_pubkey=/home/me/.ssh/foo_rsa.pub
```

Or even fancier, simply specify the key name and it'll find it in your home:

```
user_ssh_pubkey=foo_rsa
```

thijstriemstra, 2016-11-30:
And one could also argue this should be possible for password-related settings. Being able to shield the password in a local file.

thijstriemstra, 2016-11-30:
Or you can add `_file` settings for these particular options..

FooDeas, 2016-11-30:
The **public** key is one or (or multiple strings), placed in authorized keys. No need to be masked/shielded.
I don't know, how you would protect a password. You still need to provide it to set it. And because it's a FAT partition, there is no possibility to protect it at file system level.

The only thing I could do is providing an option to delete it from `installer-config.txt` after successful install. But that will prevent it from reinstalling without providing the original `installer-config.txt` again...

Any suggestions?

thijstriemstra, 2016-11-30:
I forgot that the password is still in /boot/ as well. An option to remove everything from raspberrypi-ua-netinst, inc config/logs etc after a successful installation sounds good, an auto cleanup.

thijstriemstra, 2016-11-30:
And I'm not trying to mask the public key, I'm just trying to prevent duplication. With 10 or more devices it starts to get hairy.. ;)

FooDeas, 2016-11-30:
So you want to add multiple public keys at the same time?

thijstriemstra, 2016-11-30:
I want to generate a new key for each installation and point the installer to that key, e.g. `app/ssh/key.pub`. I can also get it into the config with some other bash-script hack..

FooDeas, 2016-11-30:
Try it! A cleanup option is a nice idea, too.

FooDeas, 2016-11-30:
Example for `files/mykeyfile.xyz`: `user_ssh_pubkey=mykeyfile.xyz`
If the file is not present, the option is interpreted as pubkey string.

thijstriemstra, 2016-12-01:
Thanks, I'll test it tomorrow.

thijstriemstra, 2016-12-02:
@FooDeas I'm seeing the following errors, the rpi keeps re-trying and re-booting:

```
Creating home directory `/home/foo' ...
    Copying files from `/etc/skel' ...
  Setting SSH key for 'foo' from file 'foo_rsa.pub'... 
Oh noes, something went wrong!
You have 10 seconds to hit ENTER to get a shell...
OK
  Setting owner as 'foo' on SSH directory... OK
  Setting permissions on 'foo' SSH authorized_keys... OK
  Setting password for 'foo'... chroot: can't execute '/usr/sbin/chpasswd': No such file or directory

Oh noes, something went wrong!
You have 10 seconds to hit ENTER to get a shell...
```

FooDeas, 2016-12-02:
Thanks for report! I'll have a look at it today.

FooDeas, 2016-12-02:
I see why. Easy fix coming in a few hours.

thijstriemstra, 2016-12-05:
@FooDeas this seems to work ok now. But whenever I login I see this error:

```
-bash: service: command not found
```
And I have no idea where this comes from, it's a clean installation. Any idea?

FooDeas, 2016-12-05:
Check your installer logfile. `/usr/sbin/service` is part of package "sysvinit-utils".

thijstriemstra, 2016-12-05:
`/usr/sbin/service` is there but it's not on the path, only for the sudo user. That `/usr/sbin/` is not on the path is default behavior right, I have to manually add it to .profile or /etc/profile.d/myapp.sh.

closed 2016-12-05 by FooDeas

FooDeas, 2016-12-09:
@thijstriemstra
I fixed the `-bash: service: command not found` problem. It's a Raspbian fault.
I'm testing many things now an then release v1.5.0 today or tomorrow.