FooDeas/raspberrypi-ua-netinst #55: Realtime clock

thijstriemstra, 2016-12-21:
Recently got one of these [DS3231 I2C RTC Real Time Clock](http://www.ebay.com/itm/262123918224) modules and enabling it in Raspbian Pixel was pretty straightforward:

Enable the module by adding it at the bottom of `/etc/modules`:
```
rtc-ds1307
```

Test the module:
```
sudo i2cdetect -y 1
```
Add the following to `/etc/rc.local`:

```
echo ds1307 0x68 > /sys/class/i2c-adapter/i2c-1/new_device
sudo hwclock -s
```

And reboot the Raspberry Pi.

This works on Raspbian but I had issues with raspberrypi-ua-netinst because NTP or fakehwclock kept overriding the time. Any experience with this?  I'll report the exact issue when I find some time. 

Would this also be a good option to have, `rtc=1`?

FooDeas, 2016-12-22:
Nice idea! Will be added. Maybe the full DS1307/3231 initial setup.
In the meantime - remove/purge the package `fake-hwclock`.

But I'd still keep `ntp` in minimal packages because I see no reason to remove that. It doesn't harm if there is no internet connection and benefits if there is a connection.
I'll see how the standard configuration of systemd's timedatectl replaces it. I'm pretty sure, that it can be skipped with jessie and stretch.

FooDeas, 2016-12-22:
I think, I could need your assistance.

Could you test the following:
Purge the `fake-hwclock` package and undo all your changes to `/etc/modules` and `/etc/rc.local`.
Then add this line to your `/boot/config.txt`:
`dtoverlay=i2c-rtc,ds3231`

Write the time to RTC. Then reboot and check if the clock is read at system startup. (I'm pretty sure, it isn't.)
If not, goto `/etc/default/hwclock` and set `HWCLOCKACCESS` to `yes`.
Set the correct time and write it to RTC. Then reboot again and check again.

Please post your results!

thijstriemstra, 2016-12-29:
Thanks for the feedback @FooDeas, I'll test this today.

FooDeas, 2016-12-31:
I did a lot of work and implemented the whole RTC stuff. Feel free to test it!

thijstriemstra, 2017-01-02:
Sounds awesome, will test this asap.

FooDeas, 2017-01-02:
Note, that it's in the devel branch.

The main branch contains the code of the stable releases only. (I moved your latest commit, too.)
A `CONTRIBUTION.md` will be added soon to make this clear.

closed 2017-01-08 by FooDeas

thijstriemstra, 2017-01-17:
Awesome, seems to work just fine!