FooDeas/raspberrypi-ua-netinst #57: Building on OSX

thijstriemstra, 2016-12-29:
Is it important to be able to build the installer on OSX? Because it's currently failing, e.g.:

```
MacBook-Pro:raspberrypi-ua-netinst me$ ./clean.sh 
MacBook-Pro:raspberrypi-ua-netinst me$ ./update.sh 
Setting up gpg... 
gpg: keyring `gnupg/secring.gpg' created
gpg: keyring `gnupg/pubring.gpg' created
gpg: gnupg/trustdb.gpg: trustdb created

Downloading raspbian.public.key.
######################################################################## 100.0%
Checking key file 'raspbian.public.key'... OK
Importing 'raspbian.public.key' into keyring... OK

Downloading raspberrypi.gpg.key.
######################################################################## 100.0%
Checking key file 'raspberrypi.gpg.key'... OK
Importing 'raspberrypi.gpg.key' into keyring...OK

Downloading Release file and its signature...
######################################################################## 100.0%
######################################################################## 100.0%
Verifying Release file... OK

Downloading main package list...
######################################################################## 100.0%
Verifying main package list... ./update.sh: line 205: sha256sum: command not found
ERROR
The checksum of the main/binary-armhf/Packages.gz file doesn't match!
```

thijstriemstra, 2016-12-29:
With patch above `update.sh` is working, but `build.sh` fails with:

```
$ ./build.sh 
./build.sh: line 146: /sbin/depmod: No such file or directory
```

FooDeas, 2016-12-29:
There is no depmod on MacOS. But you need it to calculate and add the dependencies automatically. Maybe you could use `depmod.pl` from the busybox package.

FooDeas, 2016-12-29:
(Because I don't use Macs, I can't help you much.)

closed 2016-12-29 by thijstriemstra