FooDeas/raspberrypi-ua-netinst #58: Doesn't work with Raspberry Pi Model 2 B v1.2 (Revision a22042)

awelters, 2017-01-05:
Using your code, I get passed the rainbow screen and it tells me that it can't run "/etc/init.d/rcS" because permission denied.  So what does this mean? 

With the latest version of RASPBIAN JESSIE LITE it works on my Raspberry Pi Model 2 B v1.2 (Revision a22042):

RASPBIAN JESSIE LITE
Minimal image based on Debian Jessie
Version:November 2016
Release date:2016-11-25
Kernel version:4.4

The linux kernel installed is 4.4.34-v7+

awelters, 2017-01-06:
Nevermind I was able to get it to work, sorry!

FooDeas, 2017-01-06:
Nice, but it would be good to know, what was caused the problem.
I see no potential cause for this issue on that specific model...
The kernel and SoC is identical to the model 3, which is tested successfully.

awelters, 2017-01-06:
Your code works just fine I should say.  What happened is related to making sure that the rcS file has the right permissions on it before you run the build script otherwise it won't be executable on the PI.  So on my build machine I ran the following:  'chmod 751 rcS'.

FooDeas, 2017-01-06:
Weird, that this never happened in my and others test cases...
Thanks! I'll have a look at it.

FooDeas, 2017-01-08:
This can happen if the scripts in the scripts folder have no execution permission before building.

closed 2017-01-08 by FooDeas