FooDeas/raspberrypi-ua-netinst #63: Local mirror for debian packages

thijstriemstra, 2017-01-16:
I currently have limited bandwidth and would like to setup a local mirror for the required debian packages. Should changing the `mirror` setting be enough? I'll look into apt-mirror on how to get these packages in my local mirror.

thijstriemstra, 2017-01-16:
`apt-cacher` is up and running with the following in `/etc/apt/apt.conf.d/01proxy` on the Raspberry Pi:

```
Acquire::http::Proxy "http://mylocal.apt:3142";
```

Now I'll have to figure out if the mirror setting can be replaced by the proxy url above, or that the presence of that `01proxy` file is already enough.

thijstriemstra, 2017-01-17:
Results are in.

Without supplying a custom `/etc/apt/apt.conf.d/01proxy` and just a mirror,

```
mirror=http://192.168.2.50:3142/mirrordirector.raspbian.org/raspbian/
```

it used the apt-cacher server for the first packages. As soon as it started to install the `minimal` or `server` package and my custom packages, it did not use the apt-cacher server.

Before trying to copy a `/etc/apt/apt.conf.d/01proxy` as well, do you think that would work @FooDeas?

FooDeas, 2017-01-17:
I recommend to use [apt-cacher-ng](https://packages.debian.org/jessie/apt-cacher-ng) for this because it doesn't need a separate webserver (Apache).

I hope, this will fix your problem:
- check, if `archive.raspberrypi.org/debian` is configured/allowed by the cacher, too (this might be your problem)
- add custom **list** file `raspberrypi-ua-netinst/config/apt/01proxy.list`:
`Acquire::http::Proxy "http://<CacherIP>:3142/";`

thijstriemstra, 2017-01-18:
- there's a  `/var/cache/apt-cacher-ng/archive.raspberrypi.org` on the cache machine so I'm sure it's being cached without any extra config
 - I'll try that, thanks for the feedback.

thijstriemstra, 2017-01-18:
Your second suggestion resulted in this error:

```
Configuring apt:
  Configuring Raspbian repository... OK
  Checking Raspbian repository entry... OK
  Adding raspberrypi.org GPG key to apt-key... OK
  Configuring RaspberryPi repository... OK
  Configuring RaspberryPi preference... OK
  Copying './01proxy.list' to /etc/apt/sources.list.d/... OK
  Copying './foo.apt.list' to /etc/apt/sources.list.d/... OK
  Adding key './foo.gpg.key' to apt.OK
OK

Updating package lists... ERROR: 100, trying again (2/3)... ERROR: 100, trying again (3/3)... ERROR: 100, FAILED !

Oh noes, something went wrong!
You have 10 seconds to hit ENTER to get a shell...
```

I'll do a manual file copy to `/etc/apt/apt.conf.d/01proxy` now.

FooDeas, 2017-01-18:
Looks like a previous bugfix (8429bd843c17bf02de8d32f847c0c52841e29c78) causes this bug.
I'll have a look at it!

thijstriemstra, 2017-01-18:
This didn't work either, and it's quite obvious why; copying the root files I supplied happens after downloading all packages. I'm now testing with:

```diff
diff --git a/scripts/opt/raspberrypi-ua-netinst/install.sh b/scripts/opt/raspberrypi-ua-netinst/install.sh
index 430d856..97ba936 100644
--- a/scripts/opt/raspberrypi-ua-netinst/install.sh
+++ b/scripts/opt/raspberrypi-ua-netinst/install.sh
@@ -1490,13 +1490,23 @@ done
 # iterate through all the *.pref files and add them to /etc/apt/preferences.d
 for preffile in ./*.pref
 do
-       if [ "${listfile}" != "./archive_raspberrypi_org.pref" ] && [ -e "${preffile}" ]; then
+       if [ "${preffile}" != "./archive_raspberrypi_org.pref" ] && [ -e "${preffile}" ]; then
                echo -n "  Copying '${preffile}' to /etc/apt/preferences.d/... "
                sed "s/__RELEASE__/${release}/g" "${preffile}" > "/rootfs/etc/apt/preferences.d/${preffile}" || fail
                echo "OK"
        fi
 done
 
+# iterate through all the *.conf files and add them to /etc/apt/apt.conf.d
+for conffile in ./*.conf
+do
+    if [ "${conffile}" != "./archive_raspberrypi_org.pref" ] && [ -e "${conffile}" ]; then
+        echo -n "  Copying '${conffile}' to /etc/apt/apt.conf.d/... "
+        sed "s/__RELEASE__/${release}/g" "${conffile}" > "/rootfs/etc/apt/apt.conf.d/${conffile}" || fail
+        echo "OK"
+    fi
+done
+
 # iterate through all the *.key files and add them to apt-key
 for keyfile in ./*.key
 do
```

I also think the first changed line in this diff above is a bugfix @FooDeas.

thijstriemstra, 2017-01-18:
Well, that kinda worked;
```
Configuring apt:
  Configuring Raspbian repository... OK
  Checking Raspbian repository entry... OK
  Adding raspberrypi.org GPG key to apt-key... OK
  Configuring RaspberryPi repository... OK
  Configuring RaspberryPi preference... OK
  Copying './foo.apt.list' to /etc/apt/sources.list.d/... OK
  Copying './01proxy.conf' to /etc/apt/apt.conf.d/... OK
  Adding key './foo.gpg.key' to apt.OK
OK
```

But then;

```
Downloading packages...
  Reading package lists...
  Building dependency tree...
  Reading state information...
  Package xz-utils is not available, but is referred to by another package.
  This may mean that the package is missing, has been obsoleted, or
  is only available from another source
  Package autofs is not available, but is referred to by another package.
  This may mean that the package is missing, has been obsoleted, or
  is only available from another source
  Package libusb-dev is not available, but is referred to by another package.
  This may mean that the package is missing, has been obsoleted, or
  is only available from another source
  However the following packages replace it:
    libusb-0.1-4
```
But I'm not sure if this is related to the cache and just some dependency issue.. meh.

thijstriemstra, 2017-01-18:
After installation my network won't come up though, did something change in relation to that in the `dev` branch? This watchdog stuff? Because I'm also not seeing the whole system launch being logged anymore, it only shows a quick fcdisk check and then shows the prompt.

FooDeas, 2017-01-21:
I tested the last dev build successfully before pushing it. I think your broken installation is caused by missing packages.

Where is your `foo.apt.list` and `foo.gpg.key` from? This could be the reason, that apt can't read the package list!

I setup my own default configured apt-cacher-ng and will test it.

thijstriemstra, 2017-01-21:
Thanks for your feedback. I will go through the logs and see if there's anything related to missing packages. The foo.apt.list and key are copied to the config/apt folder before installation.

FooDeas, 2017-01-21:
Please make a fresh clone of the repo. These files shouldn't be there.

thijstriemstra, 2017-01-21:
No, they're copied by me in order to add a custom repository, suggested by you in some ticket months ago (and it worked fine).

FooDeas, 2017-01-21:
I'm amazed that you didn't change the filename...
Does the apt proxy have access to this custom repository?

thijstriemstra, 2017-01-21:
Of course I did, but for this ticket I used `foo` buddy ;) And yes, custom repo + cacher are on the same machine.

FooDeas, 2017-01-21:
I can confirm your problem, that updating package list when using the proxy fails.
I'm checking out why.

FooDeas, 2017-01-21:
I think, your patch above was pretty perfect and damn near to my solution...
I'm testing. 👍

FooDeas, 2017-01-21:
It's working if you name the file `01proxy.conf` now.
Thanks for your enhancement suggestion!

thijstriemstra, 2017-01-21:
ah great thanks!

FooDeas, 2017-01-21:
I could - but I wanted to fix wheezy installation before. There are some weird things regarding loading kernel modules happening when the old wheezy kernel is used.
I'd invest 1-2 days for it and see, if this can be fixed. If not - I'll release it without full featured wheezy.

thijstriemstra, 2017-01-21:
sounds good.

thijstriemstra, 2017-01-22:
@FooDeas there's still something wrong with the network/whole initialization of the system. When I boot the raspberrypi, after successful installation, it shows 4 lines almost instantly, doing a fsck disk scan and then I see the prompt. What I don't see is network/systemd configuration etc and I don't have network. When I manually login and restart the network it gives me a proper IP. Any idea? Or should I compare to master first? :(

FooDeas, 2017-01-22:
Sounds bad. WiFi or Ethernet?
Please post or better attach `/etc/network/interfaces` and `dpkg --list`.

closed 2017-01-23 by FooDeas

thijstriemstra, 2017-01-23:
Thanks @FooDeas, tested and works fine. Network issue is also gone so it must be something with my DHCP server (which is also running on a different RPI). Saw one thing in the log that is worth pointing out:

```
Configuring installed system:
mount: mounting /run on /rootfs/run failed: No such file or directory
  Setting root password... OK
  Disabling SSH password login for users... OK
```
Or can that mount error be ignored?

FooDeas, 2017-01-24:
The unmount errors are blemish, which I will care about.