FooDeas/raspberrypi-ua-netinst #64: Test devel branch for new release v1.5.1

FooDeas, 2017-01-23:
My tests with all new features were successful.

Please report issues here. Verify against master to check if they originate from new implementations.

Mausy5043, 2017-01-24:
update.sh on the `devel` branch produces undesirable output:
```
pi@rbian ~/builder/raspberrypi-ua-netinst $ ./update.sh
Setting up gpg...
gpg: keyring `gnupg/secring.gpg' created
gpg: keyring `gnupg/pubring.gpg' created
gpg: gnupg/trustdb.gpg: trustdb created

Downloading raspbian.public.key.
raspbian.public.key                        100%[========================================================================================>]   1.73K  --.-KB/s   in 0s
Checking key file 'raspbian.public.key'... OK
Importing 'raspbian.public.key' into keyring... OK

Downloading raspberrypi.gpg.key.
raspberrypi.gpg.key                        100%[========================================================================================>]   1.68K  --.-KB/s   in 0.001s
Checking key file 'raspberrypi.gpg.key'... OK
Importing 'raspberrypi.gpg.key' into keyring...OK

Downloading Release file and its signature...
Verifying Release file... ERROR
Broken GPG signature on Release file!
```

FooDeas, 2017-01-24:
@Mausy5043
Are you using a (caching) proxy?

Mausy5043, 2017-01-24:
Not that I'm aware of

FooDeas, 2017-01-24:
The connection to some files on the repository fails.
The error message is missleading. (And has to be improved...) Maybe they are updating it at the moment. This happened often to me. Please try again in some minutes and report!

Mausy5043, 2017-01-24:
No problem building and installing with the current `devel` branch.

FooDeas, 2017-01-24:
Last changes were minor, non critical.

@thijstriemstra: Any other potential problems?

If there is nothing else, I'd like to issue the release in about 15 hours...

thijstriemstra, 2017-01-25:
cool, checking now...

FooDeas, 2017-01-25:
Thanks a lot!

It's done: [raspberrypi-ua-netinst - Release v1.5.1](https://github.com/FooDeas/raspberrypi-ua-netinst/releases/tag/v1.5.1)

closed 2017-01-25 by FooDeas