FooDeas/raspberrypi-ua-netinst #65: USB audio

thijstriemstra, 2017-02-16:
Maybe this is too generic for a setting but to enable an external usb audio card (from ebay for a few dollars), create `/etc/asound.conf ` with the following content:

```
pcm.!default  {
 type hw card 1
}
ctl.!default {
 type hw card 1
}
```

See https://learn.adafruit.com/usb-audio-cards-with-a-raspberry-pi/updating-alsa-config.

FooDeas, 2017-02-16:
I'd say: Why not?
Can you suggest any default packages?

thijstriemstra, 2017-02-16:
I'm now testing it on a raspberrypi-ua-netinst installation;

```
$ sudo apt-get install mpg123
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following extra packages will be installed:
  libmpg123-0
Suggested packages:
  alsa-utils jackd nas oss-compat oss4-base pulseaudio
The following NEW packages will be installed:
  libmpg123-0 mpg123
0 upgraded, 2 newly installed, 0 to remove and 0 not upgraded.
Need to get 268 kB of archives.
After this operation, 599 kB of additional disk space will be used.
Do you want to continue? [Y/n] y
....

$ mpg123 http://stream2.friskyradio.com/frisky_mp3_hi
```

Gives me this error:

```
$ mpg123 http://stream2.friskyradio.com/frisky_mp3_hi

[module.c:129] error: Failed to open module jack: file not found

[module.c:129] error: Failed to open module portaudio: file not found

[pulse.c:84] error: Failed to open pulse audio output: Connection refused

[module.c:129] error: Failed to open module nas: file not found

[module.c:129] error: Failed to open module openal: file not found

[audio.c:180] error: Unable to find a working output module in this list: alsa,oss,jack,portaudio,pulse,nas,openal

[audio.c:545] error: Failed to open audio output module

[mpg123.c:913] error: Failed to initialize output, goodbye.
```

So yeah there are some packages required (unfortunately). I'll be using it with Qt in the end but some alsa stuff is probably needed).

thijstriemstra, 2017-02-16:
After installing:

```
$ sudo apt-get install jackd oss-compat pulseaudio
```
things are looking better:

```
$ mpg123 http://stream2.friskyradio.com/frisky_mp3_hi
High Performance MPEG 1.0/2.0/2.5 Audio Player for Layers 1, 2 and 3
	version 1.20.1; written and copyright by Michael Hipp and others
	free software (LGPL) without any warranty but with best wishes

Directory: http://stream2.friskyradio.com/
Playing MPEG stream 1 of 1: frisky_mp3_hi ...
ICY-NAME: FRISKY | feelin' frisky?
ICY-URL: http://www.friskyradio.com

MPEG 1.0 layer III, 128 kbit/s, 44100 Hz joint-stereo

ICY-META: StreamTitle='FRISKY | Allure - February 2017 - Niki Sadeki';
```

FooDeas, 2017-02-16:
Are thing hearing better, too? ;)

thijstriemstra, 2017-02-16:
Well, it looks like I (long story coming) plugged it into a USB hub from ebay china that doesn't have enough power (my canon selphy printer also needs to be plugged into the RPi itself, the usb hub is shit) so I haven't heard anything yet, but you're right I'm about to test that next. Just means I have to unscrew quite some..

[image]

FooDeas, 2017-02-16:
👍

thijstriemstra, 2017-04-24:
Apparently using `.asoundrc` instead of `/etc/asound.conf` should also work with this content:
```
pcm.!default {
        type hw
        card 0
}

ctl.!default {
        type hw
        card 0
}
```

thijstriemstra, 2017-04-24:
It's crucial to add the user to the `audio` group. Was trying to figure out why `aplay -l` returned:
```
$ aplay -l
aplay: device_list:268: no soundcards found...
```
Turns out you need to use `sudo` or add the user to the `audio` group:

```
$ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: ALSA [bcm2835 ALSA], device 0: bcm2835 ALSA [bcm2835 ALSA]
  Subdevices: 8/8
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
  Subdevice #7: subdevice #7
card 0: ALSA [bcm2835 ALSA], device 1: bcm2835 ALSA [bcm2835 IEC958/HDMI]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: Device [USB Audio Device], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```

And USB audio playback's now working fine.

So I'd suggest following for usb audio option @FooDeas:

- add user to `audio` system group
- packages: `alsa-utils jackd oss-compat pulseaudio`
- config in `/etc/asound.conf`: 
```
pcm.!default  {
 type hw card 1
}
ctl.!default {
 type hw card 1
}
```

FooDeas, 2017-04-24:
All right! Only one thing: Adding user to audio system group will get an separate option.

Mausy5043, 2017-04-25:
> All right! Only one thing: Adding user to audio system group will get an separate option.

Assuming the additional payload is only added/installed in case that option is used?

FooDeas, 2017-04-25:
@Mausy5043
Of course!

FooDeas, 2017-04-27:
I think, I will do the implementation on the weekend if my own USB audio stick arrives and release on monday.

FooDeas, 2017-04-28:
Tests please!

closed 2017-05-08 by FooDeas

thijstriemstra, 2017-05-08:
It worked fine, thanks.