FooDeas/raspberrypi-ua-netinst #67: Post installer script found no command

kraeml, 2017-04-01:
I created a post-install.txt. After running  raspberrypi-ua-netinst I found this message:

```bash
sudo tail /var/log/raspberrypi-ua-netinst.log
=================================================
=== Start executing post-install.txt. ===
/bootfs/raspberrypi-ua-netinst/config/post-install.txt: line 2: /bin/su: No such file or directory
/bootfs/raspberrypi-ua-netinst/config/post-install.txt: line 3: /bin/su: No such file or directory
=== Finished executing post-install.txt. ===
=================================================
Removing cdebootstrap-helper-rc.d... OK
Saving current time for fake-hwclock... OK

Installation finished at Sat Apr  1 08:41:32 UTC 2017 and took 43 min 37 sec (2617 seconds)
```

The post-install.txt is:

```bash
cat -A /boot/raspberrypi-ua-netinst/config/post-install.txt
#!/bin/bash$
/bin/su -lc "/usr/bin/git clone https://kraeml@gitlab.com/kraeml/raspberry-edu-devops.git /home/pi/.raspberry-edu-devops" pi$
/bin/su -lc "cd /home/pi/.raspberry-edu-devops && ./install_jns_medium.sh" pi$
```

My installer-config.txt
```bash
cat -A /boot/raspberrypi-ua-netinst/config/installer-config.txt
packages="vim,python-minimal,apt-transport-https,debian-archive-keyring,ca-certificates,screen,tmux,git,pandoc,stow,nginx"$

$
firmware=1$
$
timezone="Europe/Berlin"$
keyboard_layout="de"$
system_default_locale="de_DE.UTF-8"$
locales="de_DE.UTF-8, en_US.UTF-8"$
$
username="pi"$
userpw="SECRET"$
user_is_admin=1$
$
usergpu=1$
usergpio=1$
i2c_enable=1$
spi_enable=1$
sound_enable=1$
camera_enable=1$
$
root_ssh_allow=0$
rootpw=""$
root_ssh_pwlogin=0$
$
hostname="pi3-xy"$
$
wlan_country="DE"$
wlan_ssid="gateway"$
wlan_psk="CaHrdf1416"$
$
usbroot=1$
rootfstype="ext4"$
$
enable_watchdog=1$
$
gpu_mem=16$
hdmi_system_only=1$
```

After the reboot I found su with:

```bash
which su
/bin/su
```

The script `sudo /boot/raspberrypi-ua-netinst/config/post-install.txt` runs.

Any suggestions?

Thanks for this cool repro

kraeml

FooDeas, 2017-04-01:
The post install script runs in installer context.
So you have to chroot to do the commands in the installed system: 'chroot /rootfs'

kraeml, 2017-04-05:
Thanks, I never used chroot before. 

I changed my post-install.txt into:

```bash
cat -A /boot/raspberrypi-ua-netinst/config/post-install.txt                                                                                                                                                                                                       
#!/bin/bash$                                                                                                                                                                                                                                                                   
chroot /rootfs /bin/bash -c '/bin/su -lc "/usr/bin/git clone https://kraeml@gitlab.com/kraeml/raspberry-edu-devops.git /home/pi/.raspberry-edu-devops" pi'$ 
```

The log file says:

```bash
=================================================
=== Start executing post-install.txt. ===
fatal: open /dev/null or dup failed: Permission denied
=== Finished executing post-install.txt. ===
=================================================
```

May you have suggestions?

kraeml

Mausy5043, 2017-04-05:
In the `post-install.txt` there is no need to `su` or start a new shell. You already have root permissions.  

Try it like this:
```
chroot /rootfs /usr/bin/git clone https://kraeml@gitlab.com/kraeml/raspberry-edu-devops.git /home/pi/.raspberry-edu-devops
```

then take it from there.

FooDeas, 2017-04-05:
I think @Mausy5043's suggestion will solve your problem.

But I have another useful hint for debugging purposes:
Add `final_action=console` to your installer-config.txt. After installing, the installer will now end in console mode (default is to reboot)! There you can test your commands, which you would like to add to the post-install.txt. If they succeed in this mode, they will succeed in the script, too.

This way, you won't loose as much time as you would loose by running the installer again and again...

kraeml, 2017-04-06:
@Mausy5043 Thanks. The line 

```bash
chroot /rootfs /usr/bin/git clone https://kraeml@gitlab.com/kraeml/raspberry-edu-devops.git /home/pi/.raspberry-edu-devops
```
works.

I use su to switch user to _pi_. The script should not run as root. 

My solution now is your repro [raspboot](https://github.com/Mausy5043/raspboot) thanks for that. It works fine - not the origin script but the idea with cronjob and firstboot ...

This works now the next houres to install my project.

With cat -A I could show that the spaces are ASCII-Spaces not UTF-8 double byte characters. Sorry for confusing.

@FooDeas thanks for this. My script is running (~6h). I will test it next time.

thijstriemstra, 2017-04-24:
Can this be closed @kraeml?

kraeml, 2017-04-24:
Yes I close this issue. Thanks for your help.

closed 2017-04-24 by kraeml