FooDeas/raspberrypi-ua-netinst #68: APT error

TimVNL, 2017-04-07:
Hi,

I have an issue with the installer when I add custom apt sources with (public) GPG keys 

I get the following error:

onfiguring apt:
  Configuring Raspbian repository... OK
  Checking Raspbian repository entry... OK
  Adding raspberrypi.org GPG key to apt-key... OK
  Configuring RaspberryPi repository... OK
  Configuring RaspberryPi preference... OK
  Copying './jessie-backports.list' to /etc/apt/sources.list.d/... OK
  Copying './sury-php.list' to /etc/apt/sources.list.d/... OK
  Copying './debian-archive-jessie.gpg' to /etc/apt/trusted.gpg.d/... OK
  Copying './debian-archive-wheezy.gpg' to /etc/apt/trusted.gpg.d/... OK
  Copying './sury-php.gpg' to /etc/apt/trusted.gpg.d/... OK

Updating package lists... ERROR: 100, trying again (2/3)... ERROR: 100, trying again (3/3)... ERROR: 100, FAILED !

Oh noes, something went wrong!
You have 10 seconds to hit ENTER to get a shell...

Raspberry info:  

==================================================
raspberrypi-ua-netinst
==================================================
Revision v1.5.1 (bd078c1)
Built on Wed Jan 25 12:41:03 CET 2017
Running on Raspberry Pi version 3 Model B

TimVNL, 2017-04-07:
Found the error with apt it missing  

apt-transport-https  

it is needed even tho I am using http URLS, maybe add this before apt sources etc... are added.

FooDeas, 2017-04-07:
The list file has to be innthe same folder as the sample file. Is it?

FooDeas, 2017-04-07:
Thanks for the apt info!

FooDeas, 2017-04-07:
@TimVNL
- Not installing files issue:
The **custom files** list file has to be in the same folder as the sample file (custom_files.txt). Only the source files have to be in the subfolder `root`. Is it?
- Missing apt-transport-https issue:
I'd be happy, if you could test 8f08a9a. It should be fixed there.

TimVNL, 2017-04-09:
@FooDeas 

* Fixed the custom files issue wasn't paying attention.
* Will test this as soon as I have me new rpi ;)
  
Also maybe make a guide for exporting keys to put in the apt folder.

closed 2017-04-23 by FooDeas