FooDeas/raspberrypi-ua-netinst #70: Cannot establish wifi connection on pi zero w

dinosore, 2017-04-24:
Trying to install on a pi zero w using wifi, the install fails after less than 1 minute having failed to establish a connection.

Log (sanitised)
```
==================================================
raspberrypi-ua-netinst
==================================================
Revision v1.5.1 (bd078c1)
Built on Wed Jan 25 12:41:03 CET 2017
Running on Raspberry Pi version unknown (9000c1)
==================================================
https://github.com/FooDeas/raspberrypi-ua-netinst/
==================================================
Starting HWRNG... OK
Mounting boot partition... OK
Copying boot files... OK
Executing installer-config.txt... OK

Checking if config.txt needs to be modified before starting installation...
OK

Unmounting boot partition... OK

Network configuration:
  ifname = wlan0
  ip_addr = dhcp
  wlan_ssid = layby
  wlan_psk = ok, pass the word.
  online_config = 

Waiting for wlan0... 1.. 2.. 3.. 4.. 5.. 6.. 7.. 8.. 9.. FAILED

Oh noes, something went wrong!
You have 10 seconds to hit ENTER to get a shell...
```

installer-config.txt
```
packages=apt-transport-https
release=wheezy
username=david
userpw=password
user_is_admin=1
hostname=pie
domainname=dbr
ifname=wlan0
wlan_country=GB
wlan_ssid="layby"
wlan_psk="ok, pass the word."
timezpne="Europe/London"
keyboard_layout=gb
locales=en_GB.UTF-8
system_default_locale=en_GB.UTF-8
gpu_mem=16
rootfstype=ext4
boot_volume_label=pieboot
packages=apt-transport-https
release=wheezy
username=david
userpw=password
user_is_admin=1
hostname=pie
domainname=dbr
ifname=wlan0
wlan_country=GB
wlan_ssid="layby"
wlan_psk="ok, pass the word."
timezpne="Europe/London"
keyboard_layout=gb
locales=en_GB.UTF-8
system_default_locale=en_GB.UTF-8
gpu_mem=16
rootfstype=ext4
boot_volume_label=pieboot

```
I suspect that wifi hasn't been exercised very much.  Where it is possible, if wifi fails, most people would plug in a cable and sort out wifi later.  This, of course, isn't possible with the 0w.

I notice that my pi zero w isn't recognised by the install script.  The script isn't up to date with the source quoted.  This may well not matter, but it does indicate that I may be breaking new ground.

I've tried a few times with a few variations and I can't think what to try next.  I even tried an scS script with some debugging before I realised that the failure was before the custom script was loaded.

Help would be most appreciated, whether I've found a bug or I'm just being thick.

thijstriemstra, 2017-04-25:
Your config contains every entry twice but I suppose that's a copy/paste error?

dinosore, 2017-04-25:
Good spot. @thijstriemstra .

Uou're right.  Just checked the file to make sure.

FooDeas, 2017-04-25:
@dinosore
Please test v1.5.2 beta and report.

dinosore, 2017-04-25:
@FooDeas Just started the install.

I can ping the address I expect to be allocated by dhcp - so it looks like it's past the problem.

Will post again when it has finished.  I may go to bed, so it might not be until tomorrow.

Thanks for your very prompt response.

dinosore, 2017-04-25:
@FooDeas 
The installation finished, but it doesn't boot.

When I checked the pi, I found the activity led out.

Here is the tailof the log I found in /dev/log
```
  Removing 'diversion of /boot/overlays/w1-gpio-overlay.dtb to /usr/share/rpikernelhack/overlays/w1-gpio-overlay.dtb by rpikernelhack'
  Removing 'diversion of /boot/overlays/w1-gpio-pullup-overlay.dtb to /usr/share/rpikernelhack/overlays/w1-gpio-pullup-overlay.dtb by rpikernelhack'
  Setting up libraspberrypi0 (1.20160315-1) ...
  Setting up libraspberrypi-bin (1.20160315-1) ...
OK
Preserving original config.txt and kernels...
Configuring bootloader to start the installed system...
Creating default cmdline.txt... OK
Fixing non-privileged SSH password warning... not possible!
Removing cdebootstrap-helper-rc.d... OK
Saving current time for fake-hwclock... OK

Installation finished at Tue Apr 25 22:43:09 UTC 2017 and took 16 min 0 sec (960 seconds)

```
I put the sd card back in the pi and powered up.  The led stayed out. (This is expected behaviour for a pi 0 w if it doesn't boot).  I believe that I found the led out after installation because of a failure to boot.

I couldn't ping the pi.

FooDeas, 2017-04-26:
Weird... Could you please attach the full log file?

dinosore, 2017-04-26:
On my laptop, parted -l shows
```
Model: SD ACLCE (sd/mmc)
Disk /dev/mmcblk1: 63.9GB
Sector size (logical/physical): 512B/512B
Partition Table: msdos
Disk Flags: 

Number  Start   End     Size    Type     File system  Flags
 1      4194kB  138MB   134MB   primary  fat16
 2      138MB   63.9GB  63.7GB  primary  ext4
```
lsblk shows
```
mmcblk1     179:0    0  59.5G  0 disk 
├─mmcblk1p1 179:1    0   128M  0 part /media/david/pieboot
└─mmcblk1p2 179:2    0  59.4G  0 part /media/david/3ccdeab0-4f31-48f7-9c8c-b98314e9f000
```
```
ls -lA /media/david/pieboot/
total 19826
-rw-r--r-- 1 david david   14778 Apr 26 15:28 bcm2708-rpi-0-w.dtb
-rw-r--r-- 1 david david   10825 Apr 26 15:47 bcm2708-rpi-b.dtb
-rw-r--r-- 1 david david   11104 Apr 26 15:47 bcm2708-rpi-b-plus.dtb
-rw-r--r-- 1 david david   10855 Apr 26 15:47 bcm2708-rpi-cm.dtb
-rw-r--r-- 1 david david   12092 Apr 26 15:47 bcm2709-rpi-2-b.dtb
-rw-r--r-- 1 david david   12866 Apr 26 15:47 bcm2710-rpi-3-b.dtb
-rw-r--r-- 1 david david   15346 Apr 26 15:28 bcm2710-rpi-cm3.dtb
-rw-r--r-- 1 david david   17920 Apr 26 15:47 bootcode.bin
-rw-r--r-- 1 david david     161 Apr 26 15:48 cmdline.txt
-rw-r--r-- 1 david david    1704 Apr 26 15:48 config.txt
-rw-r--r-- 1 david david   18693 Apr 26 15:47 COPYING.linux
-rw-r--r-- 1 david david    2496 Apr 26 15:47 fixup_cd.dat
-rw-r--r-- 1 david david    6466 Apr 26 15:47 fixup.dat
-rw-r--r-- 1 david david    9709 Apr 26 15:47 fixup_db.dat
-rw-r--r-- 1 david david    9717 Apr 26 15:47 fixup_x.dat
-rw-r--r-- 1 david david 4055172 Apr 26 15:47 kernel7.img
-rw-r--r-- 1 david david 3964404 Apr 26 15:47 kernel.img
-rw-r--r-- 1 david david    1494 Apr 26 15:47 LICENCE.broadcom
drwxr-xr-x 2 david david   14336 Apr 26 15:48 overlays
drwxr-xr-x 4 david david    2048 Apr 26 15:48 raspberrypi-ua-netinst
-rw-r--r-- 1 david david  613624 Apr 26 15:47 start_cd.elf
-rw-r--r-- 1 david david 4889352 Apr 26 15:47 start_db.elf
-rw-r--r-- 1 david david 2740920 Apr 26 15:47 start.elf
-rw-r--r-- 1 david david 3841544 Apr 26 15:47 start_x.elf

 ls -lA /media/david/3ccdeab0-4f31-48f7-9c8c-b98314e9f000/
total 92
drwxrwxr-x  2 root root  4096 Apr 26 16:40 bin
drwxr-xr-x  2 root root  4096 Apr 26 16:28 boot
drwxr-xr-x  3 root root  4096 Apr 26 16:38 dev
drwxr-xr-x 59 root root  4096 Apr 26 16:48 etc
drwxr-xr-x  3 root root  4096 Apr 26 16:43 home
drwxr-xr-x 12 root root  4096 Apr 26 16:48 lib
drwx------  2 root root 16384 Apr 26 16:28 lost+found
drwxr-xr-x  2 root root  4096 Apr 26 16:37 media
drwxr-xr-x  2 root root  4096 Jun 18  2016 mnt
drwxr-xr-x  3 root root  4096 Apr 26 16:47 opt
drwxr-xr-x  2 root root  4096 Jun 18  2016 proc
drwx------  2 root root  4096 Apr 26 16:37 root
drwxr-xr-x  6 root root  4096 Jan  1  1970 run
drwxr-xr-x  2 root root  4096 Apr 26 16:40 sbin
drwxr-xr-x  2 root root  4096 Jun 20  2012 selinux
drwxr-xr-x  2 root root  4096 Apr 26 16:37 srv
drwxr-xr-x  2 root root  4096 Oct 13  2013 sys
drwxrwxrwt  2 root root  4096 Apr 26 16:48 tmp
drwxr-xr-x 10 root root  4096 Apr 26 16:37 usr
drwxr-xr-x 11 root root  4096 Apr 26 16:37 var
ls -lA /media/david/pieboot/
total 19826
-rw-r--r-- 1 david david   14778 Apr 26 15:28 bcm2708-rpi-0-w.dtb
-rw-r--r-- 1 david david   10825 Apr 26 15:47 bcm2708-rpi-b.dtb
-rw-r--r-- 1 david david   11104 Apr 26 15:47 bcm2708-rpi-b-plus.dtb
-rw-r--r-- 1 david david   10855 Apr 26 15:47 bcm2708-rpi-cm.dtb
-rw-r--r-- 1 david david   12092 Apr 26 15:47 bcm2709-rpi-2-b.dtb
-rw-r--r-- 1 david david   12866 Apr 26 15:47 bcm2710-rpi-3-b.dtb
-rw-r--r-- 1 david david   15346 Apr 26 15:28 bcm2710-rpi-cm3.dtb
-rw-r--r-- 1 david david   17920 Apr 26 15:47 bootcode.bin
-rw-r--r-- 1 david david     161 Apr 26 15:48 cmdline.txt
-rw-r--r-- 1 david david    1704 Apr 26 15:48 config.txt
-rw-r--r-- 1 david david   18693 Apr 26 15:47 COPYING.linux
-rw-r--r-- 1 david david    2496 Apr 26 15:47 fixup_cd.dat
-rw-r--r-- 1 david david    6466 Apr 26 15:47 fixup.dat
-rw-r--r-- 1 david david    9709 Apr 26 15:47 fixup_db.dat
-rw-r--r-- 1 david david    9717 Apr 26 15:47 fixup_x.dat
-rw-r--r-- 1 david david 4055172 Apr 26 15:47 kernel7.img
-rw-r--r-- 1 david david 3964404 Apr 26 15:47 kernel.img
-rw-r--r-- 1 david david    1494 Apr 26 15:47 LICENCE.broadcom
drwxr-xr-x 2 david david   14336 Apr 26 15:48 overlays
drwxr-xr-x 4 david david    2048 Apr 26 15:48 raspberrypi-ua-netinst
-rw-r--r-- 1 david david  613624 Apr 26 15:47 start_cd.elf
-rw-r--r-- 1 david david 4889352 Apr 26 15:47 start_db.elf
-rw-r--r-- 1 david david 2740920 Apr 26 15:47 start.elf
-rw-r--r-- 1 david david 3841544 Apr 26 15:47 start_x.elf

 ls -lA /media/david/3ccdeab0-4f31-48f7-9c8c-b98314e9f000/
total 92
drwxrwxr-x  2 root root  4096 Apr 26 16:40 bin
drwxr-xr-x  2 root root  4096 Apr 26 16:28 boot
drwxr-xr-x  3 root root  4096 Apr 26 16:38 dev
drwxr-xr-x 59 root root  4096 Apr 26 16:48 etc
drwxr-xr-x  3 root root  4096 Apr 26 16:43 home
drwxr-xr-x 12 root root  4096 Apr 26 16:48 lib
drwx------  2 root root 16384 Apr 26 16:28 lost+found
drwxr-xr-x  2 root root  4096 Apr 26 16:37 media
drwxr-xr-x  2 root root  4096 Jun 18  2016 mnt
drwxr-xr-x  3 root root  4096 Apr 26 16:47 opt
drwxr-xr-x  2 root root  4096 Jun 18  2016 proc
drwx------  2 root root  4096 Apr 26 16:37 root
drwxr-xr-x  6 root root  4096 Jan  1  1970 run
drwxr-xr-x  2 root root  4096 Apr 26 16:40 sbin
drwxr-xr-x  2 root root  4096 Jun 20  2012 selinux
drwxr-xr-x  2 root root  4096 Apr 26 16:37 srv
drwxr-xr-x  2 root root  4096 Oct 13  2013 sys
drwxrwxrwt  2 root root  4096 Apr 26 16:48 tmp
drwxr-xr-x 10 root root  4096 Apr 26 16:37 usr
drwxr-xr-x 11 root root  4096 Apr 26 16:37 var
```
Everything seems to work, except that the sd card has been rendered unbootable.

To try again, I had to run the xzcat again.  Both partitions were fine in my laptop, it's just that I couldn't boot the pi (even to try the reinstall process).

dinosore, 2017-04-26:
```

==================================================
raspberrypi-ua-netinst
==================================================
Revision __VERSION__
Built on __DATE__
Running on Raspberry Pi version Zero W
==================================================
https://github.com/FooDeas/raspberrypi-ua-netinst/
==================================================
Starting HWRNG... OK
Mounting boot partition... OK
Copying boot files... OK
Executing installer-config.txt... OK

Checking if config.txt needs to be modified before starting installation...
OK

Unmounting boot partition... OK

Network configuration:
  ifname = wlan0
  ip_addr = dhcp
  wlan_ssid = laybay
  wlan_psk = ok, pass the word.
  online_config = 

Waiting for wlan0... 1.. OK
Starting wpa_supplicant... OK
Configuring wlan0 with DHCP... 192.168.1.9
Set time using ntpdate... OK

Installation started at Wed Apr 26 15:28:08 UTC 2017 (UTC).

Installer configuration:
  preset = server
  packages = apt-transport-https
  firmware_packages = 
  mirror = http://mirrordirector.raspbian.org/raspbian/
  mirror_cache = 
  release = wheezy
  hostname = pie
  domainname = dbr
  rootpw = raspbian
  user_ssh_pubkey = 
  root_ssh_pubkey = 
  root_ssh_pwlogin = 1
  ssh_pwlogin = 
  username = david
  userpw = password
  usergpio = 
  usergpu = 
  usergroups = 
  usersysgroups = 
  user_is_admin = 1
  cdebootstrap_cmdline = --flavour=minimal --include=locales,console-setup,cpufrequtils,kmod,raspbian-archive-keyring,rng-tools,ifupdown,net-tools,openssh-server,dosfstools,ntp,fake-hwclock,vim-tiny,iputils-ping,wget,ca-certificates,rsyslog,cron,dialog,locales,less,man-db,logrotate,bash-completion,console-setup,apt-utils,isc-dhcp-client,wpasupplicant,crda,sudo
  packages_postinstall = apt-transport-https,raspberrypi-bootloader,firmware-brcm80211,libraspberrypi-bin,raspi-copies-and-fills
  boot_volume_label = pieboot
  bootsize = +128M
  bootoffset = 8192
  rootsize = 
  timeserver = time.nist.gov
  timezone = Etc/UTC
  rtc = 
  keyboard_layout = gb
  locales = en_GB.UTF-8
  system_default_locale = en_GB.UTF-8
  wlan_country = GB
  cmdline = dwc_otg.lpm_enable=0 console=serial0,115200 console=tty1 elevator=deadline fsck.repair=yes
[... 894 lines cut ...]
  Removing 'diversion of /boot/overlays/smi-dev-overlay.dtb to /usr/share/rpikernelhack/overlays/smi-dev-overlay.dtb by rpikernelhack'
  Removing 'diversion of /boot/overlays/smi-nand-overlay.dtb to /usr/share/rpikernelhack/overlays/smi-nand-overlay.dtb by rpikernelhack'
  Removing 'diversion of /boot/overlays/smi-overlay.dtb to /usr/share/rpikernelhack/overlays/smi-overlay.dtb by rpikernelhack'
  Removing 'diversion of /boot/overlays/spi-bcm2708-overlay.dtb to /usr/share/rpikernelhack/overlays/spi-bcm2708-overlay.dtb by rpikernelhack'
  Removing 'diversion of /boot/overlays/spi-bcm2835-overlay.dtb to /usr/share/rpikernelhack/overlays/spi-bcm2835-overlay.dtb by rpikernelhack'
  Removing 'diversion of /boot/overlays/spi-dma-overlay.dtb to /usr/share/rpikernelhack/overlays/spi-dma-overlay.dtb by rpikernelhack'
  Removing 'diversion of /boot/overlays/tinylcd35-overlay.dtb to /usr/share/rpikernelhack/overlays/tinylcd35-overlay.dtb by rpikernelhack'
  Removing 'diversion of /boot/overlays/uart1-overlay.dtb to /usr/share/rpikernelhack/overlays/uart1-overlay.dtb by rpikernelhack'
  Removing 'diversion of /boot/overlays/vc4-kms-v3d-overlay.dtb to /usr/share/rpikernelhack/overlays/vc4-kms-v3d-overlay.dtb by rpikernelhack'
  Removing 'diversion of /boot/overlays/vga666-overlay.dtb to /usr/share/rpikernelhack/overlays/vga666-overlay.dtb by rpikernelhack'
  Removing 'diversion of /boot/overlays/w1-gpio-overlay.dtb to /usr/share/rpikernelhack/overlays/w1-gpio-overlay.dtb by rpikernelhack'
  Removing 'diversion of /boot/overlays/w1-gpio-pullup-overlay.dtb to /usr/share/rpikernelhack/overlays/w1-gpio-pullup-overlay.dtb by rpikernelhack'
  Setting up libraspberrypi0 (1.20160315-1) ...
  Setting up libraspberrypi-bin (1.20160315-1) ...
OK
Preserving original config.txt and kernels...
Configuring bootloader to start the installed system...
Creating default cmdline.txt... OK
Fixing non-privileged SSH password warning... not possible!
Removing cdebootstrap-helper-rc.d... OK
Saving current time for fake-hwclock... OK

Installation finished at Wed Apr 26 15:48:15 UTC 2017 and took 20 min 7 sec (1207 seconds)

```

FooDeas, 2017-04-26:
Please try the following `installer-config.txt`:
```
packages=apt-transport-https
username=david
userpw=password
user_is_admin=1
hostname=pie
domainname=dbr
ifname=wlan0
wlan_country=GB
wlan_ssid="layby"
wlan_psk="ok, pass the word."
timezone="Europe/London"
keyboard_layout=gb
locales=en_GB.UTF-8
system_default_locale=en_GB.UTF-8
gpu_mem=16
rootfstype=ext4
```
Typo in "timezone" is fixed, "release" is left out to install jessie and "boot_volume_label" is left out because I never tested that.
But I'm pretty sure, that wheezy is the problem here...

FooDeas, 2017-04-26:
Now, I'm absolutely sure, that wheezy is the problem:
The raspberry pi bootloader package is version `1.20160315-1`. You see the date here. The pi zero w needs a much newer bootloader and, important for you too, jessie kernel to get the wifi module working.

dinosore, 2017-04-26:
Started running 3 minutes ago with your config edited to fix passwords etc.

Don't understand timezone typo.

Should I be able to do a dist-upgrade to Wheezy?

FooDeas, 2017-04-26:
- Typo was "timez**p**ne"
- A dist-upgrade from wheezy to jessie would work if you do it on a working RPi (make a backup before). Make sure to have installed `raspberrypi-bootloader` on the upgraded jessie before switching the card into the zero.

I'm looking forward to your report!

dinosore, 2017-04-26:
I've just ssh'd into pie (the zerow) as distinct from pi (the model B.

Looks ok.

```
david@pie:~$ apt search raspberrypi-bootloader
Sorting... Done
Full Text Search... Done
raspberrypi-bootloader/stable,now 1.20170405-1 armhf [installed]
  Raspberry Pi bootloader

raspberrypi-bootloader-nokernel/stable 1.20160523-1~nokernel2 armhf
  Raspberry Pi bootloader

david@pie:~$ apt show raspberrypi-bootloader
Package: raspberrypi-bootloader
Source: raspberrypi-firmware
Version: 1.20170405-1
Maintainer: Serge Schneider <serge@raspberrypi.org>
Installed-Size: 12.5 MB
Recommends: raspberrypi-kernel
Homepage: https://github.com/raspberrypi/firmware
Priority: extra
Section: admin
Download-Size: 3,300 kB
APT-Manual-Installed: yes
APT-Sources: http://archive.raspberrypi.org/debian/ jessie/main armhf Packages
Description: Raspberry Pi bootloader
 This package contains the Raspberry Pi firmware and bootloader.
david@pie:~$ ls /etc/apt/sources.list.d
raspberrypi.org.list

```
On pi I get
```
root@pi:~# apt show raspberrypi-bootloader
Package: raspberrypi-bootloader
State: not a real package (virtual)
N: Can't select candidate version from package raspberrypi-bootloader as it has no candidate
N: Can't select versions from package 'raspberrypi-bootloader' as it is purely virtual
N: No packages found
root@pi:~# ls /etc/apt/sources.list.d
syncthing.list  upm.list

```
pi is alresdy on jessie.  However it doesn't use the raspberrypi.org repository.

Pie is now a working pi! With raspberrypi-nootloader installed.

Can't I just do the dist-upgrade on the pi 0w?

And thanks for getting it working.

FooDeas, 2017-04-26:
> Can't I just do the dist-upgrade on the pi 0w?

Theroretically yes. But to use your old pi would be easier (working network connection, no manual bootloader replacement). In both cases, you have to edit your old installation apt sources. Create it if a file doesn't exist.

Content of `/etc/apt/sources.list`:
```
deb http://mirrordirector.raspbian.org/raspbian jessie main contrib non-free firmware
```
Content of `/etc/apt/sources.list.d/raspberrypi.org.list`:
```
deb http://archive.raspberrypi.org/debian jessie main 
```
Content of `/etc/apt/preferences.d/archive_raspberrypi_org.pref`:
```
Package: *
Pin: origin archive.raspberrypi.org
Pin-Priority: 600
```
Now you can execute `apt-get clean`, `apt-get update` and then `apt-get dist-upgrade` and hope the best! I'd let follow: `apt-get install raspberrypi-bootloader raspberrypi-sys-mods dosfstools `

FooDeas, 2017-04-26:
And before you reboot, execute as su:
`echo "dwc_otg.lpm_enable=0 console=serial0,115200 console=tty1 elevator=deadline fsck.repair=yes root=/dev/mmcblk0p2 rootfstype=ext4 rootwait loglevel=3" > /rootfs/boot/cmdline.txt`
(This is one line.)

dinosore, 2017-04-26:
If I understand you, you suggest I put the sd card from the 0w into the b to do the upgrade mainly to use ethernet rather than wifi.

I'll review your comments tomorrow when I'm more awake.

I've just ordered a faster sd card for the B after noticing how much faster the install this time was than the last one using the installer you forked.  It will let me tidy up things a bit.  It currently runs wheezy.  Will I have to install jessie first, or can I go straight to wheezy?

I realise I'm straying off-topic for an issue.  You can contact me, if you wish, with my github username @dbrsoftware.co.uk

dinosore, 2017-04-26:
One other thing I noticed, I can ping pi.local but not pie.local.

FooDeas, 2017-04-26:
You missunderstood. Fact is, you can't use wheezy on the zero w if you want to use its wifi module at all. It's definitely not working! No kernel modules available for wheezy.

If I understood you right, you want to use your old system from the Pi B in the zero? Then you have to dist-upgrade its wheezy to jessie (steps described above) and now can use the card in the zero without loosing data.

Now I see: Yes, that's off-topic... :)

FooDeas, 2017-04-26:
(ping pie.dbr or its ip - btw. off-topic, too :))

dinosore, 2017-04-27:
Thank you for fixing the bug first time.  Please close.

The follow-on was all my fault - I relied on my poor memory for debian/raspbian release names and got it wrong.  I should have checked.  I was thinking wheezy followed jessie. (If only they used numbers - 7, 8, 9, or even oldstable, stable, testing).

I did a dist-upgrade to jessie on the B a year or two ago and it's been running happily ever since (with cron housekeeping) syncing to a usb drive.

The zero is destined for my son's house to provide some off-site backup.

I apologise for wasting your time and thank you for continuing to provide support to an old man who should know better.

closed 2017-04-27 by FooDeas

FooDeas, 2017-04-27:
Thank you for testing the pi zero w!