FooDeas/raspberrypi-ua-netinst #71: Show failure with led(s) (enhancement)

dinosore, 2017-04-24:
It would be nice if it were easy to detect that the install had failed without waiting until thinking that it must have done so.

When installing blind - probably common for headless servers - it would be good to know of the failure immediately.

The pi 0 w has just 1 led - an activity light.  If it is possible that on failure the processor is halted and the led turned off, then the sd card could be extracted and the log examined.

This would be most useful for failures before ssh can be established.

Mausy5043, 2017-04-25:
It is possible to connect using the serial console which is available during install. 

Normal behaviour for the installer in case of a failure is to offer the possibility to drop into a shell for debugging. If this option times-out the installer reboots.

dinosore, 2017-04-25:
Easy enough to connect a tv with an hdmi cable, but I don't have a usb keyboard. 

As I'm thinking headless, it didn't occur to me to connect a monitor just to see the log.  Doh!

closed 2017-04-25 by dinosore

dinosore, 2017-04-26:
I now realise that it is possible to configure final_action making this "enhancement" unnecessary.