FooDeas/raspberrypi-ua-netinst #72: final_action=halt seems not to work as expected

dinosore, 2017-04-29:
This is based on git-6e67710 on a pi 0w..
zip of relevant files attached.

note final_action=halt and a full stop rather than a comma before ntfs-3g.

On finding the error, I expected the processor to halt and the activity light to go out.

I left it going for a few hours.  It looks like, rather than halting, it rebooted and tried again.  See the multiple error loga.
[
[raspberrypi-ua-netinst.zip](https://github.com/FooDeas/raspberrypi-ua-netinst/files/965776/raspberrypi-ua-netinst.zip)
](url)

FooDeas, 2017-04-29:
You packages setting in the config is wrong.
Use ONE line. Thats all.

dinosore, 2017-04-29:
You have every right to consider this a feature, working as designed.

I would call it an infinite loop. :-) 

 A limit would be good, but probably not important enough to implement.

FooDeas, 2017-04-29:
Thanks for suggestion

FooDeas, 2017-05-10:
Done in 83adf4278abe51ac631fbcb6294756f9eee96c58.

closed 2017-05-10 by FooDeas