FooDeas/raspberrypi-ua-netinst #74: /dev/misc/rtc: No such file or directory

thijstriemstra, 2017-04-30:
I have the rtc option enabled:

```
rtc=ds3231
```

but forgot to hook up the actual rtc device before installation. After everything installed the process died with:

```
=== Finished executing post-install.txt. ===
=================================================
Removing cdebootstrap-helper-rc.d... OK
Saving current time to RTC... hwclock: can't open '/dev/misc/rtc': No such file or directory

Oh noes, something went wrong!
You have 10 seconds to hit ENTER to get a shell...
```

But after that it didn't reboot or anything, I only saw a blank screen with a blinking cursor. After waiting a few mins I decided to take out the sd-card and check the error log where I found this error.

I then hooked up the rtc device and powered up the rpi again. I expected the process to either restart or continue where it left of but instead nothing happened, and there was no new error in the logs either.

thijstriemstra, 2017-04-30:
Perhaps this step can also be executed earlier?

thijstriemstra, 2017-04-30:
Tried again and getting the same error. Only this time it took me into an emergency console after a timeout for mmcblk0p1.device..

thijstriemstra, 2017-04-30:
Seeing this error as well:

```
  Failed to set capabilities on file `/usr/bin/systemd-detect-virt' (Invalid argument)
  The value of the capability argument is not permitted for a file. Or the file is not a regular (non-symlink) file
  Adjusting cmdline.txt...
  Removing old sysctl tweaks, if present...
  Removing old NOPASSWD line, if present...
  sed: can't read /etc/sudoers: No such file or directory
  Failed
  Updating documentation URL in /boot/config.txt...
  Setting up pi-bluetooth (0.1.3) ...
```

thijstriemstra, 2017-04-30:
More errors:
```
Configuring installed system:
mount: mounting /run on /rootfs/run failed: No such file or directory
  Setting root password... OK
  Disabling SSH password login for users... OK
```

thijstriemstra, 2017-04-30:
I also see this twice (note the weird path `/boot/raspberrypi-ua-netinst/config/files/./installation-files.list`):

```
Adding files & folders listed in /boot/raspberrypi-ua-netinst/config/files/./installation-files.list...
  /etc/udev/rules.d/99-escpos.rules
  /etc/lsyncd.conf
  /etc/hosts
  /etc/asound.conf

=================================================
=== Start executing post-install.txt. ===
Adding files & folders listed in /boot/raspberrypi-ua-netinst/config/files/installation-files.list...
  /etc/udev/rules.d/99-escpos.rules
  /etc/lsyncd.conf
  /etc/hosts
  /etc/asound.conf

=== Finished executing post-install.txt. ===
=================================================
```

thijstriemstra, 2017-04-30:
Looks like some of the errors we're caused by using a deprecated parameter, switching to `userperms_admin=1` now. But then the fallback mechanism you created for deprecated params isn't functioning correctly.

FooDeas, 2017-04-30:
- Please check your `post-install.txt`. I don't know what's going on there.
- The deprecated code should work as expected.
- "Can not write log..." error occurs many times but in 9 of 10 it gets filtered. You can safely ignore these.
- "can't read /etc/sudoers..." is totally weired.

Could you please attach/send a full log??

thijstriemstra, 2017-04-30:
> "can't read /etc/sudoers..." is totally weired.

It's because sudo wasn't installed at all (because I was using an old var)

FooDeas, 2017-04-30:
I tested the deprecated code. The old var shouldn't have impact here...

FooDeas, 2017-04-30:
Bug found.

thijstriemstra, 2017-05-01:
@FooDeas what bug(s) exactly?

FooDeas, 2017-05-07:
2. and 4. is a bug.
3. has to be ignored be ignored

FooDeas, 2017-05-07:
Notice that this bug is only present in devel! I'm testing new code with some fixes.

FooDeas, 2017-05-08:
Please test a00f0b1c588040e6e2c9f60e398d0ed8768570eb. Feel free to use deprecated variables!

FooDeas, 2017-05-10:
Bugfix in a6c418dcd9090444b91a170c7618300ed4f1fbe5
Done in a00f0b1c588040e6e2c9f60e398d0ed8768570eb

closed 2017-05-10 by FooDeas