FooDeas/raspberrypi-ua-netinst #76: Expand quiet_boot option

thijstriemstra, 2017-05-08:
The Raspberry Pi ‘color test’ can be disabled by adding the line `disable_splash=1` to `/boot/config.txt`.

thijstriemstra, 2017-05-08:
Disable the various bits of output from the kernel and friends by adding `consoleblank=0 loglevel=1 quiet` to `/boot/cmdline.txt`.

thijstriemstra, 2017-05-08:
Disable the login prompt by running `systemctl disable getty@tty1` as root.

thijstriemstra, 2017-05-08:
I'm taking these steps from https://yingtongli.me/blog/2016/12/21/splash.html and will open a splashscreen option ticket later, but the requests above are all related to "quiet booting" and a splashscreen in particular.

thijstriemstra, 2017-05-08:
Perhaps quiet_boot can have a third value, e.g. `2`, that silences absolutely everything (with steps above)?

FooDeas, 2017-05-08:
Further:
Disable the Raspberry Pi logo in the corner of the screen by adding logo.nologo to /boot/cmdline.txt.

thijstriemstra, 2017-05-08:
> Disable the Raspberry Pi logo in the corner of the screen by adding logo.nologo to /boot/cmdline.txt.

:+1: already have that ;)

thijstriemstra, 2017-05-09:
I also had to remove the black borders: `disable_overscan=1` in `/boot/config.txt`.

FooDeas, 2017-05-10:
Parts of this will be implemented in v1.5.2.

FooDeas, 2017-05-11:
The following parameters will be implemented soon:
```
disable_splash=1
logo.nologo
disable_overscan=1
```

BTW: Set `installer_retries="0"` in the `installer-retries.txt` and see what happens! :)

dinosore, 2017-05-13:
> BTW: Set installer_retries="0" in the installer-retries.txt and see what happens! :)

I suspect the 0 gets decremented and you get a very large number of retries.

So, do you need to specify 1 to get 0 retries as the first try is incorrectly treated as a retry?

I would, defensively set to 1 instead of 0 just in case my suspicion were justified.

FooDeas, 2017-05-13:
Every installation attempt is a "retry". So `installer_retries="0"` means: no attempt left. Default is `3`.

dinosore, 2017-05-13:
> Every installation attempt is a "(re)try". So installer_retries="0" means: no attempt left. Default is 3.
> So you can simulate what happens in case of too many tries if you set it to 0.

You're going to confuse English speakers with this view.  The prefix "re" indicates repetition.  The verb "to retry" means "to try again".  The first attempt cannot be a retry.

Unless you take the same view as Humpty Dumpty, your interpretation is just plain wrong.
(url)https://en.wikipedia.org/wiki/Humpty_Dumpty#In_Through_the_Looking-Glass

Using "(re)try" is understood to indicate that the prefix is only applied where appropriate.  However, the brackets don't appear in the variable's name.  Most people won't read any explanation, but expect it to take its usual meaning.  If they wanted to check after the first attempt if it was worth retrying, they would just set it to zero.

A few dictionary references...
retry https://en.oxforddictionaries.com/definition/retry
re-enter https://en.oxforddictionaries.com/definition/us/re-enter
attempt https://en.oxforddictionaries.com/definition/attempt
re- http://dictionary.cambridge.org/dictionary/english/re

I hope you enjoy the Humpty Dumpty link :-)

FooDeas, 2017-05-13:
No reproof needed! :D
I first implemented this on another location in code and hadn't changed it afterwards.
If you worry about it, make a meaningful suggestion with "installer_" as prefix!

Mausy5043, 2017-05-13:
'installer_attemps'

closed 2017-05-13 by FooDeas

FooDeas, 2017-05-13:
Oups... Forget about it!
The wording "retries" was correct! 😆
The setting only gets important if and only if there gets something wrong. It will be decreased and the installer restarts. If the retry countdown is zero, there will be no further attempt/retry. That means, setting it to zero does nothing if no errors occur.

Sorry for the confusion and thanks for the attention!