FooDeas/raspberrypi-ua-netinst #80: Building the image fails on debian 8.5

Mohitsharma44, 2017-05-19:
Hi,
I am trying to build a custom image using build instructions mentione in the Build.md file but it doesn't seem to work in Debian 8.5 and Debian 8.6 (w/ priority, essential and required packages + packages mentioned in build.md installed)

#### Versions tried:
1.5.1 (code) and Master

#### Commands ran (as normal user):
./clean.sh (runs fine)
./update.sh (runs fine)
./build.sh (crashes with following errors):
> tar: ./sbin/btrfsck: Cannot hard link to ‘./sbin/btrfs’: Operation not permitted
> tar: ./usr/share/man/man8/btrfsck.8.gz: Cannot utime: Operation not permitted
> tar: ./usr/lib/arm-linux-gnueabihf/libbtrfs.so: Cannot utime: Operation not permitted
> tar: ./usr/lib/arm-linux-gnueabihf/libbtrfs.so.0: Cannot utime: Operation not permitted
> tar: Exiting with failure status due to previous errors

#### Commands ran (as root user):
./clean.sh (runs fine)
./update.sh (runs fine)
./build.sh (crashes with following errors):
> tar: ./bin/bash-static: Cannot change ownership to uid 0, gid 0: Operation not permitted
> tar: ./bin: Cannot change ownership to uid 0, gid 0: Operation not permitted
> tar: ./usr/share/man/man1/bash-static.1.gz: Cannot change ownership to uid 0, gid 0: Operation not permitted
> tar: ./usr/share/man/man1: Cannot change ownership to uid 0, gid 0: Operation not permitted
> tar: ./usr/share/man: Cannot change ownership to uid 0, gid 0: Operation not permitted
> tar: ./usr/share/doc/bash-static/copyright: Cannot change ownership to uid 0, gid 0: Operation not permitted
> tar: ./usr/share/doc/bash-static/changelog.Debian.gz: Cannot change ownership to uid 0, gid 0: Operation not permitted
> tar: ./usr/share/doc/bash-static: Cannot change ownership to uid 0, gid 0: Operation not permitted
> tar: ./usr/share/doc: Cannot change ownership to uid 0, gid 0: Operation not permitted
> tar: ./usr/share/lintian/overrides/bash-static: Cannot change ownership to uid 0, gid 0: Operation not permitted
> tar: ./usr/share/lintian/overrides: Cannot change ownership to uid 0, gid 0: Operation not permitted
> tar: ./usr/share/lintian: Cannot change ownership to uid 0, gid 0: Operation not permitted
> tar: ./usr/share: Cannot change ownership to uid 0, gid 0: Operation not permitted
> tar: ./usr: Cannot change ownership to uid 0, gid 0: Operation not permitted
> tar: .: Cannot change ownership to uid 0, gid 0: Operation not permitted
> tar: Exiting with failure status due to previous errors

Any suggestions?

FooDeas, 2017-05-19:
Maybe you build it from a FAT partition or SMB mount?

Mohitsharma44, 2017-05-19:
@FooDeas Hmm... I am running it in a virtual environment. Think that could be related?

FooDeas, 2017-05-19:
Make sure, that you copy the repository on the virtual hard disk and execute the scripts from there.
The virtualization itself or your debian version is definitely not the problem.

FooDeas, 2017-05-19:
Further make sure, you are using a Linux file system like ext4 there.
(btw. "build.sh" definitely needs no root permissions, if you have user permissions on the repository clone folder)

Mohitsharma44, 2017-05-19:
@FooDeas Thanks for changing the title to more appropriate. 
And yeah you are right, I have mounted my development directory on virtual machines and containers. So for this case too, I cloned the repo on a shared folder on my mac! Closing it now.

Maybe for future people who face same issue, clone the repo on the virtual drive itself.

closed 2017-05-19 by Mohitsharma44