FooDeas/raspberrypi-ua-netinst #81: Testing for upcoming release

FooDeas, 2017-05-28:
I'd release the v1.5.2 tomorrow evening. Features are fixed and can be found in the changelog text file.
If anyone discovers a bug, please post it here!

Note that the deprecated commits before july 2016 will get sqashed then, so local repositories will require a rebase.

thijstriemstra, 2017-05-31:
Haven't had time lately (was focused on cryptocurrencies ;) but I'll try to test this asap.

thijstriemstra, 2017-06-01:
@FooDeas I enabled the following:

```
hdmi_monitor_res=640x480
hdmi_disable_overscan=1
console_blank=0
quiet_boot=1
disable_raspberries=1
disable_splash=1
cleanup=1
cleanup_logfiles=1
```

And something went wrong because the resulting config.txt contains:

```
[all]
initramfs raspberrypi-ua-netinst/raspberrypi-ua-netinst.cpio.gz
[pi3]
enable_uart=1

# Only for RPi model 3: The following line enables the ability to boot from USB.
# Notice: This flag will be written to OTP and is permanent.
#program_usb_boot_mode=1
disable_overscan=1
disable_overscan=1
disable_overscan=1
disable_overscan=1
disable_overscan=1
disable_overscan=1
disable_overscan=1
disable_overscan=1
disable_overscan=1
disable_overscan=1
disable_overscan=1
disable_overscan=1
disable_overscan=1
disable_overscan=1
disable_overscan=1
disable_overscan=1
disable_overscan=1
disable_overscan=1
disable_overscan=1
disable_overscan=1
disable_overscan=1
disable_overscan=1
```

And I couldn't find any error log..

thijstriemstra, 2017-06-06:
any idea @FooDeas?

FooDeas, 2017-06-06:
Yep. Just a bit busy atm.

FooDeas, 2017-06-11:
Finally! The `disable_overscan` bug is fixed.

Fya: `hdmi_monitor_res=640x480` doesn't make sense and gets ignored if `hdmi_type` isn't set to `monitor`.

thijstriemstra, 2017-06-12:
It's a 640x480 VGA monitor with a VGA to HDMI converter..

thijstriemstra, 2017-06-12:
Tested, and everything looks fine. Files get cleaned up. Logs as well. Please release these goodies, cheers!

thijstriemstra, 2017-06-12:
One thing though;

I have a master config and slave configs that extend this master file. You suggested to add this at the top of these slave configs:

```
sanitize_inputfile /bootfs/raspberrypi-ua-netinst/config/installer-config-common.txt
source /bootfs/raspberrypi-ua-netinst/config/installer-config-common.txt
```

But I'm now seeing an error about `sanitize_inputfile` not found. Should I remove that line or is there a replacement?

FooDeas, 2017-06-12:
Replacement: inputfile_sanitize

thijstriemstra, 2017-06-12:
Ok thanks. And about disabling the raspberries; I still see them during installation (but this could be related to `inputfile_sanitize` and my master config not being read). Will test again.

FooDeas, 2017-06-12:
Boot and kernel config (like the logo) except some device settings will be set to the system only.

FooDeas, 2017-06-15:
Done.

closed 2017-06-15 by FooDeas