FooDeas/raspberrypi-ua-netinst #84: Does the "server" preset have components needed for GPIO program access?

ornamentist, 2017-07-13:
This is more of a question than a problem:

I'd like to use the "server" preset for a headless Raspberry Pi 3 used with a peripheral that is accessed through the GPIO pins and the R-Pi Python GPIO library.

I can post-install Python for the Python libraries but does the "server" preset install all the necessary kernel modules or other low-level software components that are needed for GPIO access?

ornamentist, 2017-07-13:
I realize there's a setting for non-privileged access to the GPIO pins, I'm more uncertain about system-level components needed. (Ignorance showing here I know).

FooDeas, 2017-07-27:
The server (and minimal) preset comes with all you need for GPIO access!

closed 2017-07-27 by FooDeas

ornamentist, 2017-07-28:
Excellent and thank you so much for creating and maintaining this project.