FooDeas/raspberrypi-ua-netinst #86: Building in the future & TIMEOUT created

Mausy5043, 2017-07-25:
After the building of the `devel` branch this evening; looking at the `tree` listing (see below) two odd things caught my eye. 
First is the fact that some files were created in "The Future" and the second is presence of the file `TIMEOUT` that's in the root. 

Note: This is after the **build** phase (clean, update, build) and the `usbboot` parameter is *not* set (as far as I know).

WTF?

NB: [This is how I build:](https://gist.github.com/Mausy5043/d1aa29af76fab960fa1e1fc0c53baf7b.js)

```
./raspberrypi-ua-netinst/build_dir/bootfs
├── [-rw-r--r--       15660 May 15 21:09]  bcm2708-rpi-0-w.dtb
├── [-rw-r--r--       15197 May 15 21:09]  bcm2708-rpi-b.dtb
├── [-rw-r--r--       15456 May 15 21:09]  bcm2708-rpi-b-plus.dtb
├── [-rw-r--r--       14916 May 15 21:09]  bcm2708-rpi-cm.dtb
├── [-rw-r--r--       16523 May 15 21:09]  bcm2709-rpi-2-b.dtb
├── [-rw-r--r--       17624 May 15 21:09]  bcm2710-rpi-3-b.dtb
├── [-rw-r--r--       16380 May 15 21:09]  bcm2710-rpi-cm3.dtb
├── [-rw-r--r--       50248 Jul  3 12:07]  bootcode.bin
├── [-rw-r--r--         100  The Future ]  cmdline.txt
├── [-rw-r--r--          94  The Future ]  config.txt
├── [-rw-r--r--       18693 Aug 21  2015]  COPYING.linux
├── [-rw-r--r--        2583 Jul  3 16:07]  fixup_cd.dat
├── [-rw-r--r--        6674 Jul  3 16:07]  fixup.dat
├── [-rw-r--r--        9813 Jul  3 16:07]  fixup_db.dat
├── [-rw-r--r--        9813 Jul  3 16:07]  fixup_x.dat
├── [-rw-r--r--     4579248 Jul  3 12:07]  kernel7.img
├── [-rw-r--r--     4379032 Jul  3 12:07]  kernel.img
├── [-rw-r--r--        1494 Nov 18  2015]  LICENCE.broadcom
├── [drwxr-sr-x        4096 Jul  3 12:07]  overlays
│   ├── [-rw-r--r--        1027 Apr 28  9:18]  adau1977-adc.dtbo
│   ├── [-rw-r--r--        1587 Apr 28  9:18]  adau7002-simple.dtbo
│   ├── [-rw-r--r--        2425 Jun 22  2016]  ads1015.dtbo
│   ├── [-rw-r--r--        2425 Apr 28  9:18]  ads1115.dtbo
│   ├── [-rw-r--r--        2426 May  4  2016]  ads7846.dtbo
│   ├── [-rw-r--r--        1387 Apr 28  9:18]  akkordion-iqdacplus.dtbo
│   ├── [-rw-r--r--        1473 Apr 28  9:18]  allo-boss-dac-pcm512x-audio.dtbo
│   ├── [-rw-r--r--        1208 Jul  3 12:07]  allo-digione.dtbo
│   ├── [-rw-r--r--        1011 Oct 24  2016]  allo-piano-dac-pcm512x-audio.dtbo
│   ├── [-rw-r--r--        1585 Apr 28  9:18]  allo-piano-dac-plus-pcm512x-audio.dtbo
│   ├── [-rw-r--r--        1668 May  4  2016]  at86rf233.dtbo
│   ├── [-rw-r--r--        1972 Apr 28  9:18]  audioinjector-addons.dtbo
│   ├── [-rw-r--r--         787 May 27  2016]  audioinjector-wm8731-audio.dtbo
│   ├── [-rw-r--r--         575 Sep 22  2016]  audremap.dtbo
│   ├── [-rw-r--r--         449 May  4  2016]  bmp085_i2c-sensor.dtbo
│   ├── [-rw-r--r--         983 May  4  2016]  dht11.dtbo
│   ├── [-rw-r--r--         663 Sep 22  2016]  dionaudio-loco.dtbo
│   ├── [-rw-r--r--        1027 Apr 28  9:18]  dionaudio-loco-v2.dtbo
│   ├── [-rw-r--r--         765 Sep 22  2016]  dpi18.dtbo
│   ├── [-rw-r--r--         789 May  4  2016]  dpi24.dtbo
│   ├── [-rw-r--r--         852 Apr 28  9:18]  dwc2.dtbo
│   ├── [-rw-r--r--         423 May  4  2016]  dwc-otg.dtbo
│   ├── [-rw-r--r--        1403 May  4  2016]  enc28j60.dtbo
│   ├── [-rw-r--r--        1279 Dec 19  2016]  enc28j60-spi2.dtbo
│   ├── [-rw-r--r--        1932 Apr 28  9:18]  fe-pi-audio.dtbo
│   ├── [-rw-r--r--        1437 Jul  3 12:07]  goodix.dtbo
│   ├── [-rw-r--r--        1259 Apr 28  9:18]  googlevoicehat-soundcard.dtbo
│   ├── [-rw-r--r--        1036 Jun 22  2016]  gpio-ir.dtbo
│   ├── [-rw-r--r--         901 May  4  2016]  gpio-poweroff.dtbo
│   ├── [-rw-r--r--         779 May  4  2016]  hifiberry-amp.dtbo
│   ├── [-rw-r--r--         655 May  4  2016]  hifiberry-dac.dtbo
│   ├── [-rw-r--r--        1693 Apr 28  9:18]  hifiberry-dacplus.dtbo
│   ├── [-rw-r--r--         959 Apr 28  9:18]  hifiberry-digi.dtbo
│   ├── [-rw-r--r--        1125 Apr 28  9:18]  hifiberry-digi-pro.dtbo
│   ├── [-rw-r--r--        2468 May  4  2016]  hy28a.dtbo
│   ├── [-rw-r--r--        2939 May  4  2016]  hy28b.dtbo
│   ├── [-rw-r--r--        1182 May  6  2016]  i2c0-bcm2708.dtbo
│   ├── [-rw-r--r--         854 May  4  2016]  i2c1-bcm2708.dtbo
│   ├── [-rw-r--r--         270 Apr 28  9:18]  i2c-bcm2708.dtbo
│   ├── [-rw-r--r--         991 Jun 22  2016]  i2c-gpio.dtbo
[... 91 lines cut ...]
│   │   │       │   ├── [drwxr-sr-x        4096 Jul 25 21:30]  init.d
│   │   │       │   │   └── [-rw-r--r--         687 Jul 25 21:30]  switch_cpu_governor
│   │   │       │   ├── [drwxr-sr-x        4096 Jul 25 21:30]  logrotate.d
│   │   │       │   │   └── [-rw-r--r--         320 Jul 25 21:30]  raspmod
│   │   │       │   └── [drwxr-sr-x        4096 Jul 25 21:30]  rsyslog.d
│   │   │       │       └── [-rw-r--r--         433 Jul 25 21:30]  raspmod.conf
│   │   │       └── [drwxr-sr-x        4096 Jul 25 21:30]  home
│   │   │           └── [drwxr-sr-x        4096 Jul 25 21:30]  pi
│   │   │               ├── [-rw-r--r--        3508 Jul 25 21:30]  .bashrc
│   │   │               ├── [-rw-r--r--        4400 Jul 25 21:30]  .dircolors
│   │   │               ├── [-rw-r--r--         170 Jul 25 21:30]  .lftprc
│   │   │               ├── [-rw-r--r--         735 Jul 25 21:30]  .profile
│   │   │               └── [-rw-r--r--         177 Jul 25 21:30]  .screenrc
│   │   ├── [-rw-r--r--        2138  The Future ]  installer-config.txt
│   │   ├── [-rw-r--r--          21  The Future ]  installer-retries.txt
│   │   └── [-rw-r--r--        3327  The Future ]  post-install.txt
│   └── [-rw-r--r--    16103257  The Future ]  raspberrypi-ua-netinst.cpio.gz
├── [-rw-r--r--      659492 Jul  3 16:07]  start_cd.elf
├── [-rw-r--r--     4993604 Jul  3 16:07]  start_db.elf
├── [-rw-r--r--     2855556 Jul  3 16:07]  start.elf
├── [-rw-r--r--     3939492 Jul  3 16:07]  start_x.elf
└── [-rw-r--r--           0  The Future ]  TIMEOUT

14 directories, 158 files
```

FooDeas, 2017-07-27:
Interesting! Only the timestamps of these files, which are created while building are affected. Maybe building on a VM?

The empty "TIMEOUT" file belongs to the MSD boot feature. It tells the bootloader to increase the enumeration timeout limit from two to five seconds without the need to permanently change the OTP bit.

Mausy5043, 2017-07-28:
Built on a Raspberry Pi B+

FooDeas, 2017-07-29:
You are using a remote FS. Eigher the time on that system is in future compared to the local or it uses another time zone. Check it by using the 'stat <file>' command and compare it to the output of 'date'.

Mausy5043, 2017-07-30:
@FooDeas You're right. The remote system's time wasn't getting update by `ntp`

Thanks!

closed 2017-07-30 by Mausy5043