FooDeas/raspberrypi-ua-netinst #88: Raspbian stretch supported?

thijstriemstra, 2017-09-04:
Is https://www.raspberrypi.org/blog/raspbian-stretch/ supported?

thijstriemstra, 2017-09-04:
When running `update.sh` it just hangs:

```
$ ./update.sh 
Setting up gpg... 
gpg: keyring `gnupg/secring.gpg' created
gpg: keyring `gnupg/pubring.gpg' created
gpg: gnupg/trustdb.gpg: trustdb created

Downloading raspbian.public.key.
raspbian.public.key            100%[====================================================>]   1,73K  --.-KB/s    in 0s      
Checking key file 'raspbian.public.key'... OK
Importing 'raspbian.public.key' into keyring... OK

Downloading raspberrypi.gpg.key.
raspberrypi.gpg.key            100%[====================================================>]   1,68K  --.-KB/s    in 0,003s  
Checking key file 'raspberrypi.gpg.key'... OK
Importing 'raspberrypi.gpg.key' into keyring...OK

Downloading Release file and its signature...
```

thijstriemstra, 2017-09-05:
`jessie` is also hardcoded: https://github.com/FooDeas/raspberrypi-ua-netinst/blob/devel/update.sh#L19

thijstriemstra, 2017-09-05:
> When running update.sh it just hangs

This was an error on my part, my build.conf contained a non-existing cache server, fixed now.

My build.conf looks like:

```
mirror_raspbian_cache=192.168.178.27:3142
mirror_raspberrypi_cache=192.168.178.27:3142
release=wheezy
```

And update.sh now dies with this error:

```
Adding required packages...
  firmware-brcm80211               firmware-brcm80211_0.43+rpi5_all.deb
  libgcc1                          libgcc1_4.8.2-21~rpi3rpi1_armhf.deb
  libpcre3                         libpcre3_8.31-2rpi2_armhf.deb
  libraspberrypi-bin               libraspberrypi-bin_1.20160315-1_armhf.deb
  libraspberrypi0                  libraspberrypi0_1.20160315-1_armhf.deb
  raspberrypi-bootloader           raspberrypi-bootloader_1.20160315-1_armhf.deb

Adding required packages...
  bash-static                      bash-static_4.2+dfsg-0.1+deb7u4_armhf.deb
  btrfs-tools                      btrfs-tools_0.19+20120328-7.1_armhf.deb
  busybox                          busybox_1.20.0-7_armhf.deb
  cdebootstrap-static              cdebootstrap-static_0.5.10+rpi1_armhf.deb
  coreutils                        coreutils_8.13-3.5_armhf.deb
  diffutils                        diffutils_3.2-6_armhf.deb
  dosfstools                       dosfstools_3.0.13-1+rpi1_armhf.deb
  dpkg                             dpkg_1.16.18+rpi1_armhf.deb
  e2fslibs                         e2fslibs_1.42.5-1.1+deb7u1_armhf.deb
  e2fsprogs                        e2fsprogs_1.42.5-1.1+deb7u1_armhf.deb
  gpgv                             gpgv_1.4.12-7+deb7u9_armhf.deb
  ifupdown                         ifupdown_0.7.8_armhf.deb
  libacl1                          libacl1_2.2.51-8_armhf.deb
  libatm1                          libatm1_2.5.1-1.5_armhf.deb
  libattr1                         libattr1_2.4.46-8_armhf.deb
  libblkid1                        libblkid1_2.20.1-5.3_armhf.deb
  libbz2-1.0                       libbz2-1.0_1.0.6-4_armhf.deb
  libc-bin                         libc-bin_2.13-38+rpi2+deb7u12_armhf.deb
  libc6                            libc6_2.13-38+rpi2+deb7u12_armhf.deb
  libcap2                          libcap2_2.22-1.2_armhf.deb
  libcomerr2                       libcomerr2_1.42.5-1.1+deb7u1_armhf.deb
  libdb5.3                         libdb5.3_5.3.15-2_armhf.deb
  libdbus-1-3                      libdbus-1-3_1.6.8-1+deb7u6_armhf.deb
  liblzma5                         liblzma5_5.1.1alpha+20120614-2_armhf.deb
  liblzo2-2                        liblzo2-2_2.06-1+deb7u1_armhf.deb
  libmount1                        libmount1_2.20.1-5.3_armhf.deb
  libncurses5                      libncurses5_5.9-10_armhf.deb
  libnl-3-200                      libnl-3-200_3.2.7-4+deb7u1_armhf.deb
  libnl-genl-3-200                 libnl-genl-3-200_3.2.7-4+deb7u1_armhf.deb
  libpam0g                         libpam0g_1.1.3-7.1_armhf.deb
  libpcsclite1                     libpcsclite1_1.8.4-1+deb7u2_armhf.deb
  libselinux1                      libselinux1_2.1.9-5_armhf.deb
  libslang2                        libslang2_2.2.4-15_armhf.deb
  libssl1.0.0                      libssl1.0.0_1.0.1t-1+deb7u2_armhf.deb
  libtinfo5                        libtinfo5_5.9-10_armhf.deb
  libuuid1                         libuuid1_2.20.1-5.3_armhf.deb
  lsb-base                         lsb-base_4.1+Debian8+rpi1+deb7u1_all.deb
  netbase                          netbase_5.0_all.deb
  ntpdate                          ntpdate_4.2.6.p5+dfsg-2+deb7u7_armhf.deb
  raspbian-archive-keyring         raspbian-archive-keyring_20120528.2_all.deb
  rng-tools                        rng-tools_2-unofficial-mt.14-1_armhf.deb
  tar                              tar_1.26+dfsg-0.1+deb7u1_armhf.deb
  util-linux                       util-linux_2.20.1-5.3_armhf.deb
  vim-common                       vim-common_7.3.547-7+deb7u4_armhf.deb
  wpasupplicant                    wpasupplicant_1.0-3+deb7u4_armhf.deb
  zlib1g                           zlib1g_1.2.7.dfsg-13_armhf.deb
ERROR: Unable to find all required packages in package list!
Missing packages: 'raspberrypi-kernel f2fs-tools iproute2 libaudit-common libaudit1 libsmartcols1'
```

thijstriemstra, 2017-09-05:
> I guess newer is better so I'll give 1.1 a try.

That worked with following patch:

```diff
diff --git a/update.sh b/update.sh
index 863d910..98efb0d 100755
--- a/update.sh
+++ b/update.sh
@@ -76,7 +76,7 @@ packages+=("libpcsclite1")
 packages+=("libselinux1")
 packages+=("libslang2")
 packages+=("libsmartcols1")
-packages+=("libssl1.0.0")
+packages+=("libssl1.1")
 packages+=("libtinfo5")
 packages+=("libuuid1")
 packages+=("zlib1g")
```

But now `build.sh` dies with the following error:

```
$ ./build.sh 
cp: cannot stat 'tmp/sbin/mkfs.btrfs': No such file or directory
```

Any idea @FooDeas?

FooDeas, 2017-09-05:
Using the current installer with stretch in the installer-config should already work and install Debian stretch as intended!
Porting the installer environment itself to stretch is a bit more work. I've tried that some time ago but hadn't enough time to finish. I'll give it another try very soon!

emaayan, 2017-09-10:
i'm not sure i follow, is it possible to use stretch from the image that comes with it? 

cause if this means using stretch that gets downloaded from the mirrors, that could be a problem cause those packages keep getting updated.

h3rald, 2017-09-10:
I specified release=stretch in the installer-configure.txt file and I got this error:

`Unknown Suite: stretch`

Any ideas? I would like to perform an unattended installation of a minimal raspbian stretch release... did anyone succeed in doing so?

FooDeas, 2017-09-10:
Yes - it seems that cdebootstrap is the problem. I took one our struggling to include curl to the installer environment.
With porting to stretch, I need a more efficient package resolve/copy solution. The current way to do that is horrible!

h3rald, 2017-09-11:
But as far as I can tell from reading online, the latest version of cdebootstrap doesn't have any problem... so is there any way to manually substitute the .deb package with a more recent one? Does the installer cache the downloaded packages somewhere?

Someone managed to do it with the raspbian-ua-installer, I wonder if it is possible with this installer as well -- especially knowing that you already sorted out curl installation.

malignus, 2017-09-19:
I've got it working now after resolving all the curl dependencies.

thijstriemstra, 2017-09-19:
@malignus, can you make a pull request..?

malignus, 2017-09-20:
@thijstriemstra, done

emaayan, 2017-09-26:
hi.. so what's the status of this issue? can this installer be used with a stretch image?

FooDeas, 2017-09-30:
90% is done - I'm still fixing some issues.

FooDeas, 2017-09-30:
I applied all needed changes from malignus' PR and got cdebootstrap working.
The only issue during the installation is setting the timezone. I'll check that later. More important:

The system reboots - loads firmware, kernel and - stops with "can't access tty; job control turned off".
The system partition is mounted (read only?) and readable. I can execute shell commands and load bash.
I need someone, who has a short look at it.

malignus, 2017-09-30:
There is a symlink missing from /sbin/init to systemd. Please have a look into my install.sh

FooDeas, 2017-09-30:
I already tried that - and switched the "ln -s" arguments. I just used the right order now.
Success. Thanks!

FooDeas, 2017-09-30:
@thijstriemstra @malignus and anyone who want's to help:
Please help testing the devel_stretch branch!

mpmc, 2017-10-04:
@FooDeas (and everyone else involved) thanks for getting raspberrypi-ua-netinst working with Stretch! It's a shame that the original netinst is lagging behind & I'm guessing there is no way to merge up? :sob:  

Anyway, I ran into some issues initially, but everything seems to be working as expected & the additional options are fantastic!

Some (minor) issues.

* If IPv6 is enabled on the router but not configured to route IPv6 the install doesn't appear to complete and eventually gives up. I had to configure openwrt to assign IPv6 addresses (I'd previously disabled DHCPv6 as it caused a few issues on the network last time I used it). While this isn't an issue with the installer, it would be nice if there was an option to ignore ipv6 during install.

* Text appears "chopped off" during install (mainly when retrieving/configuring packages), I say chopped off but what appears to happen is the text is only partially printed with the cursor blinking. It however eventually catches up & prints the rest (along with other output). I observed this on 2 Pi's (Pi B+ 512M / Pi 2 B 1GB). I have a capture of the install if anyone would like to see it.

I initially did a blind install on the Pi 2B 1G, but that failed due to the aforementioned IPv6 issue. So used my older Pi B+ 512M (which is hooked up to a capture box) to see what the problem was, finally twigged that it was an IPv6 issue, fixed the network & the install completed in about 25 minutes! The newer Pi2 (not wifi) took circa 10 mins if that. 

Next I'm doing an install on:

* Original 256M Pi (One of the first production batch)
* Another 3x Pi 2 1G
* And a Pi Zero W.

I will report back if I encounter any issues, but hopefully it's clear sailing from here.

FooDeas, 2017-10-04:
Thanks!
Can you attach a screenshot of the problematic openwrt IPv6 settings and the installer log?

mpmc, 2017-10-04:
@FooDeas Unfortunately I didn't keep the log, I don't think the installer actually gives up, it just feels like it does, as each HTTP request takes an age to finally fallback to IPv4. I suspect it would complete, but how long is anyone's guess, I waited up to 2 hours before deciding something was wrong & pulled the plug. I could break my network again & record another install if it's really needed.

This problem isn't unique to this installer as I remember other HTTP requests failing on my Orange Pi's when running apt update. So I think the combinations of disabling DHCPv6 & the DNS lookups failing.

As for the problematic IPv6 settings I'd just turned off "Router Advertisement-Service" and "DHCPv6-Service". I also think openwrt was still advertising the IPv6 DNS addresses (2001:4860:4860::8888) even though it wasn't possible to query those servers. In all honestly it was bad configuration on my end, I'm still getting my head around IPv6. It doesn't help that my ISP (BT) do some odd things - see [here](https://www2.warwick.ac.uk/fac/sci/csc/people/computingstaff/jaroslaw_zachwieja/bt_fttc_ipv6/).

FooDeas, 2017-10-04:
Please create an issue for disabling IPv4 at all. Would you test it if I implement this feature?

mpmc, 2017-10-04:
@FooDeas - absolutely I have no issue breaking my network again!

thijstriemstra, 2017-10-06:
Tested with an older rpi:

```
==================================================
raspberrypi-ua-netinst
==================================================
Revision v2.0.0_RC1 (19430c7)
Built on vr  6 okt 2017  2:04:25 CEST
Running on Raspberry Pi version B+
==================================================
https://github.com/FooDeas/raspberrypi-ua-netinst/
==================================================
```
Everything's fine till here:

```
Configuring apt:
  Configuring Raspbian repository... OK
  Checking Raspbian repository entry... OK
  Adding raspberrypi.org GPG key to apt-key... OK
  Configuring RaspberryPi repository... OK
  Configuring RaspberryPi preference... OK
```

The custom repo I tested this with warned about not being able to parse the pgp key:

```
  Copying './foo.apt.list' to /etc/apt/sources.list.d/... OK
  Adding key './foo.gpg.key' to apt.Warning: apt-key output should not be parsed (stdout is not a terminal)
OK
OK
  Copying './01proxy' to /etc/apt/apt.conf.d/... OK
```
And then it dies with errors related to old packages specified on my side:

```
Updating package lists... OK

Downloading packages...
  Reading package lists...
  Building dependency tree...
  Reading state information...
  E: Unable to locate package libinput5
```

which I'll fix.

So looks good except for the `apt-key` warning (error?).

FooDeas, 2017-10-09:
We have to ignore the apt-key warning because stdout tees to serial and log. But the output itself is ugly...

bengtfredh, 2017-10-12:
raspberrypi-ua-netinst v2.0.0 RC2:
root_ssh_pwlogin=1 has no effect.
root login ssh with password not enabled.

closed 2017-10-15 by FooDeas

thijstriemstra, 2017-10-15:
> We have to ignore the apt-key warning because stdout tees to serial and log. But the output itself is ugly...

Ok but was it properly executed?

FooDeas, 2017-10-16:
Its a warning only.

FooDeas, 2017-10-16:
More in detail:
Your return type was 0 - no fail here. So it was executed successfully.

thijstriemstra, 2017-10-16:
Thanks for the feedback. I suppose stretch version is more strict with apt-key.