FooDeas/raspberrypi-ua-netinst #91: support for QEMU

conversy, 2017-09-23:
Hello,

would it be possible to add support to using QEMU ? That would make testing easier to do...

I tried with inspiration from https://www.raspberrypi.org/forums/viewtopic.php?f=66&t=89693
but to no avail...
I tried also by using usb boot support that happens to be similar to how QEMU handles the raspberry (no /dev/mmcblk, but /dev/sda instead), again to no avail...

Best regards,

s.

FooDeas, 2017-09-24:
For USB boot, the script checks the hardware requirements. Especially the USB boot flag has to be set - which isn't possible in hypervisor. I could add a workaround for this.

In the meantime, you could comment out the "# MSD boot" section in "install.sh".

FooDeas, 2017-10-29:
Please report if this works.

conversy, 2017-11-03:
Hi, sorry for the late answer.

Trying with the 'devel' branch, using this command:
qemu-system-arm -kernel kernel-qemu-4.4.34-jessie -cpu arm1176 -m 256 -M versatilepb -serial stdio -append "root=/dev/sda1 panic=1 rootfstype=ext4 rw" -hda raspberrypi-ua-netinst-git-066f0cb.img

It fails with this message:
No filesystem could mount root, tried: ext4
Kernel panic - not syncing: VFS: unamble to mount root fs on unknown-block(8,1)

Mausy5043, 2017-11-04:
The default fstype is `f2fs`. Maybe QEMU doesn't support that?

conversy, 2017-11-04:
Note that there was a thread on raspberrypi.org forums about raspbian-ua-netinst:
https://www.raspberrypi.org/forums/viewtopic.php?f=66&t=89693

it pointed to this script:
https://pastebin.com/raw/sqtQ19Ci

I tried to imitate what this last script does to the .img, before the .img was created, with no success. I was thinking that raspberrypi-ua-netinst could include an option to directly build an img compatible with qemu...

FooDeas, 2017-11-04:
Thanks for your investigation!
It's on my ToDo list.

conversy, 2017-12-10:
FYI: according to this comment https://stackoverflow.com/a/45893057 @ SO (see the end of the comment), this does not seem to be ua-netinst specific.

conversy, 2020-05-20:
I gave it a quick try today, using https://github.com/dhruvvyas90/qemu-rpi-kernel/ and this command:
`qemu-system-arm \
  -M versatilepb \
  -cpu arm1176 \
  -m 256 \
  -hda raspberrypi-ua-netinst-git-caf7423.img \
  -net user,hostfwd=tcp::5022-:22 \
  -dtb qemu-rpi-kernel/versatile-pb-buster.dtb \
  -kernel qemu-rpi-kernel/kernel-qemu-4.19.50-buster \
  -append 'root=/dev/sda1 panic=1' \
  -no-reboot \
  -serial stdio`

There is a progress as it passes the `root` failure (thx to `root=/dev/sda1`), but it fails when looking for init:
`Kernel panic - not syncing: No working init found.`

I tried passing `init=/init` or `init=/bin/busybox` to no avail, I do not know what the init executable is when ua-netinst boots for the first time.