FooDeas/raspberrypi-ua-netinst #92: Feature: (Optionally) Disable IPv6/4

mpmc, 2017-10-04:
Pretty much as title - see #88 for more.

FooDeas, 2017-10-04:
Try this: [raspberrypi-ua-netinst-git-a5f4ee3.zip](https://github.com/FooDeas/raspberrypi-ua-netinst/releases/download/v2.0.0_RC1/raspberrypi-ua-netinst-git-a5f4ee3.zip)
Set `ip_ipv6=0` in the installer-config. IPv6 should be completely disabled during installation and in the installed system.

mpmc, 2017-10-04:
Getting an error :(

```

==================================================
raspberrypi-ua-netinst
==================================================
Revision a5f4ee3
Built on Thu Oct  5 00:17:05 CEST 2017
Running on Raspberry Pi version Zero W
==================================================
https://github.com/FooDeas/raspberrypi-ua-netinst/
==================================================
Starting HWRNG... OK
Mounting boot partition... OK
Copying boot files... OK
Executing installer-config.txt...
OK

Searching for deprecated variables...

Checking if config.txt needs to be modified before starting installation...
OK

Unmounting boot partition... OK

Network configuration:
  ifname = eth0
  ip_addr = dhcp
  online_config = 

Disabling IPv6... /opt/raspberrypi-ua-netinst/install.sh: line 801: /proc/sys/net/ipv6/conf/all/disable_ipv6: No such file or directory
OK
Waiting for eth0... 1.. 2.. 3.. 4.. 5.. 6.. 7.. 8.. 9.. FAILED

Error: The installation could not be completed!
```

FooDeas, 2017-10-06:
Thanks! I've to check the order.

FooDeas, 2017-10-08:
Please try v2.0.0 RC2.

mpmc, 2017-10-09:
I broke my network again & this issue is now resolved for me.

```
Installation finished at Mon Oct  9 23:19:39 UTC 2017 and took 23 min 20 sec (1400 seconds)
```
The only thing I noticed were these errors, but they don't appear to affect anything, but posting here anyway (JIC)

```
mark@rpidelta:~$ sudo cat /var/log/raspberrypi-ua-netinst.log | grep Failed
  Failed to set capabilities on file `/bin/ping6' (Invalid argument)
  Failed to set capabilities on file `/bin/ping' (Operation not supported)
  Failed to set capabilities on file `/usr/bin/systemd-detect-virt' (Operation not supported)
```

Thanks :)

FooDeas, 2017-10-10:
The warnings come from the raspberrypi-sys-mods module and can be fully ignored.
Nice to hear, that disabling IPv6 works now!

closed 2017-10-10 by FooDeas