FooDeas/raspberrypi-ua-netinst #99: release is set to "stretch" by default

thijstriemstra, 2017-10-15:
"release is set to "stretch" by default" is mentioned in the changelog but as far as i can see, `jessie` is no longer supported.

FooDeas, 2017-10-16:
I hope we didn't break anything with stretch. Jessie and wheezy _should_ still work...

thijstriemstra, 2017-10-16:
I saw some commits that *should* have broken it, let me find some.

FooDeas, 2017-10-16:
You're right. I don't think, building is still possible with jessie. But I see no reason why we need this.
The installer itself can install all three releases! It tried this one week ago without any problems.

thijstriemstra, 2017-10-16:
Ok I fully agree @FooDeas, but let's make a note of that (in a positive way ;)) in the release notes/changelog

FooDeas, 2017-10-16:
But that's wrong. If you want to use an older version of raspbian. Just set the release version in the installer-config.
The build.sh content can and should be left as it is.

thijstriemstra, 2017-10-16:
Fair enough.

closed 2017-10-16 by thijstriemstra

FooDeas, 2017-10-16:
I thought about it again.
The problem is: The build environment has to be always up to date with the highest release that is intended to be installed.
When I see the package dependency changes with the update to stretch and the workload, I come to the decision that the whole build process has to be rewritten. This will resolve many similar problems permanently.

Before then, the release version in the build process can't be changed - unfortunately this has always been the case.